fwupd/fwupd #914: build issue: help2man: can't get `--help' info from src/fwupdmgr

swidz, 2019-01-05:
Version 1.2.3 downloaded from Releases page
Linux Debian 9.6

The ninja build stops at generating perameter help/manual page:
I have tried to run this line and get the same output:

```
/usr/bin/help2man src/fwupdmgr --no-info --output src/fwupdmgr.1 --name fwupd --manual 'User$ Commands' --version-string 1.2.3
help2man: can't get `--help' info from src/fwupdmgr
Try `--no-discard-stderr' if option outputs to stderr

```
```
$meson build -Dplugin_dell=false

The Meson build system
Version: 0.47.2
Source dir: /home/sebastian/Downloads/fwupd-1.2.3
Build dir: /home/sebastian/Downloads/fwupd-1.2.3/build
Build type: native build
Project name: fwupd
Project version: 1.2.3
Native C compiler: cc (gcc 6.3.0 "cc (Debian 6.3.0-18+deb9u1) 6.3.0 20170516")
Build machine cpu family: x86_64
Build machine cpu: x86_64
Program git found: YES (/usr/bin/git)
Compiler for C supports arguments -fstack-protector-strong: YES
Compiler for C supports arguments -Waggregate-return: YES
Compiler for C supports arguments -Wunused: YES
Compiler for C supports arguments -Warray-bounds: YES
Compiler for C supports arguments -Wcast-align: YES
Compiler for C supports arguments -Wclobbered: YES
Compiler for C supports arguments -Wdeclaration-after-statement: YES
Compiler for C supports arguments -Wduplicated-branches: NO
Compiler for C supports arguments -Wduplicated-cond: YES
Compiler for C supports arguments -Wempty-body: YES
Compiler for C supports arguments -Wformat=2: YES
Compiler for C supports arguments -Wformat-nonliteral: YES
Compiler for C supports arguments -Wformat-security: YES
Compiler for C supports arguments -Wformat-signedness: YES
Compiler for C supports arguments -Wignored-qualifiers: YES
Compiler for C supports arguments -Wimplicit-function-declaration: YES
Compiler for C supports arguments -Wincompatible-pointer-types-discards-qualifiers: NO
Compiler for C supports arguments -Winit-self: YES
Compiler for C supports arguments -Wlogical-op: YES
Compiler for C supports arguments -Wmissing-declarations: YES
Compiler for C supports arguments -Wmissing-format-attribute: YES
Compiler for C supports arguments -Wmissing-include-dirs: YES
Compiler for C supports arguments -Wmissing-noreturn: YES
Compiler for C supports arguments -Wmissing-parameter-type: YES
Compiler for C supports arguments -Wmissing-prototypes: YES
Compiler for C supports arguments -Wnested-externs: YES
Compiler for C supports arguments -Wno-cast-function-type -Wcast-function-type: NO
Compiler for C supports arguments -Wno-error=cpp -Werror=cpp: YES
Compiler for C supports arguments -Wno-unknown-pragmas -Wunknown-pragmas: YES
Compiler for C supports arguments -Wno-discarded-qualifiers -Wdiscarded-qualifiers: YES
Compiler for C supports arguments -Wno-missing-field-initializers -Wmissing-field-initializers: YES
Compiler for C supports arguments -Wno-strict-aliasing -Wstrict-aliasing: YES
Compiler for C supports arguments -Wno-suggest-attribute=format -Wsuggest-attribute=format: YES
Compiler for C supports arguments -Wno-unused-parameter -Wunused-parameter: YES
Compiler for C supports arguments -Wnull-dereference: YES
Compiler for C supports arguments -Wold-style-definition: YES
Compiler for C supports arguments -Woverride-init: YES
Compiler for C supports arguments -Wpointer-arith: YES
Compiler for C supports arguments -Wredundant-decls: YES
Compiler for C supports arguments -Wreturn-type: YES
Compiler for C supports arguments -Wshadow: YES
Compiler for C supports arguments -Wsign-compare: YES
Compiler for C supports arguments -Wstrict-aliasing: YES
Compiler for C supports arguments -Wstrict-prototypes: YES
[... 161 lines cut ...]
libsoup-2.4.vapi:1955.3-1955.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:1958.3-1958.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2000.3-2000.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2003.3-2003.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2006.3-2006.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2076.3-2076.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2080.3-2080.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2083.3-2083.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2086.3-2086.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2089.3-2089.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2092.3-2092.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2095.3-2095.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2098.3-2098.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2101.3-2101.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2104.3-2104.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2107.3-2107.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
libsoup-2.4.vapi:2111.3-2111.12: warning: [Deprecated] is deprecated. Use [Version (deprecated = true, deprecated_since = "", replacement = "")]
[91/326] Generating fwupdmgr-man with a custom command.
FAILED: src/fwupdmgr.1 
/usr/bin/help2man src/fwupdmgr --no-info --output src/fwupdmgr.1 --name fwupd --manual 'User Commands' --version-string 1.2.3
help2man: can't get `--help' info from src/fwupdmgr
Try `--no-discard-stderr' if option outputs to stderr
[98/326] Compiling C object 'src/src@@fwupd@exe/fu-engine.c.o'.
ninja: build stopped: subcommand failed.
```

swidz, 2019-01-05:
After further investigation this is not help2man related. When I try to run 
`src/fwupdmgr`
I get the following error:

`fwupdmgr: error while loading shared libraries: libfwupd.so.2: cannot open shared object file: No such file or directory`

swidz, 2019-01-05:
I have added `/usr/local/lib/x86_64-linux-gnu ` to `/etc/ld.so.conf.d/local.conf` (created new file)
then run `ldconfig` which fixed the problem. 

```
$ ldd fwupdmgr
	linux-vdso.so.1 (0x00007fffc89e7000)
	libfwupd.so.2 => /home/sebastian/Downloads/fwupd-1.2.3/build/src/./../libfwupd/libfwupd.so.2 (0x00007f13468f8000)
	libxmlb.so.1 => /usr/local/lib/x86_64-linux-gnu/libxmlb.so.1 (0x00007f13466d6000)
	libgio-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0 (0x00007f1346340000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f13460ed000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f1345dd9000)
	libsoup-2.4.so.1 => /usr/lib/x86_64-linux-gnu/libsoup-2.4.so.1 (0x00007f1345ae9000)
	libsqlite3.so.0 => /usr/lib/x86_64-linux-gnu/libsqlite3.so.0 (0x00007f13457e1000)
	libarchive.so.13 => /usr/lib/x86_64-linux-gnu/libarchive.so.13 (0x00007f134552d000)
	libjson-glib-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libjson-glib-1.0.so.0 (0x00007f1345305000)
	libuuid.so.1 => /lib/x86_64-linux-gnu/libuuid.so.1 (0x00007f1345100000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f1344d61000)
	libgmodule-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0 (0x00007f1344b5d000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f1344943000)
	libselinux.so.1 => /lib/x86_64-linux-gnu/libselinux.so.1 (0x00007f134471b000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f1344504000)
	libmount.so.1 => /lib/x86_64-linux-gnu/libmount.so.1 (0x00007f13442b6000)
	libffi.so.6 => /usr/lib/x86_64-linux-gnu/libffi.so.6 (0x00007f13440ad000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f1343e3a000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f1343c1d000)
	libxml2.so.2 => /usr/lib/x86_64-linux-gnu/libxml2.so.2 (0x00007f1343862000)
	libgssapi_krb5.so.2 => /usr/lib/x86_64-linux-gnu/libgssapi_krb5.so.2 (0x00007f1343617000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f1343313000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f134310f000)
	libnettle.so.6 => /usr/lib/x86_64-linux-gnu/libnettle.so.6 (0x00007f1342ed8000)
	libacl.so.1 => /lib/x86_64-linux-gnu/libacl.so.1 (0x00007f1342ccf000)
	liblzo2.so.2 => /lib/x86_64-linux-gnu/liblzo2.so.2 (0x00007f1342aad000)
	liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f1342887000)
	liblz4.so.1 => /usr/lib/x86_64-linux-gnu/liblz4.so.1 (0x00007f1342675000)
	libbz2.so.1.0 => /lib/x86_64-linux-gnu/libbz2.so.1.0 (0x00007f1342465000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f1346d3e000)
	libblkid.so.1 => /lib/x86_64-linux-gnu/libblkid.so.1 (0x00007f134221f000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f1342017000)
	libicui18n.so.57 => /usr/lib/x86_64-linux-gnu/libicui18n.so.57 (0x00007f1341b9d000)
	libicuuc.so.57 => /usr/lib/x86_64-linux-gnu/libicuuc.so.57 (0x00007f13417f5000)
	libicudata.so.57 => /usr/lib/x86_64-linux-gnu/libicudata.so.57 (0x00007f133fd78000)
	libkrb5.so.3 => /usr/lib/x86_64-linux-gnu/libkrb5.so.3 (0x00007f133fa9e000)
	libk5crypto.so.3 => /usr/lib/x86_64-linux-gnu/libk5crypto.so.3 (0x00007f133f86b000)
	libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x00007f133f667000)
	libkrb5support.so.0 => /usr/lib/x86_64-linux-gnu/libkrb5support.so.0 (0x00007f133f45b000)
	libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007f133f257000)
	libattr.so.1 => /lib/x86_64-linux-gnu/libattr.so.1 (0x00007f133f052000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f133ecd0000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f133eab9000)
```

Now When I run fwupdmgr get-devices
I get the timeout but it seems a different problem

```
$ sudo fwupdmgr get-devices
Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
```

hughsie, 2019-01-06:
# fwupdmgr 

Except we shouldn't be running the installed fwupdmgr binary for help2man, we should be using the built version; does calling ./src/fwupdmgr before the ldconfig also fail? That should be able to resolve the ./libfwupd/*.so, although maybe we're missing some env vars for the help2man generation.

> have ninja install call ldconfig automatically

I've not seen any other projects do this. It would also need to be disabled by default, as rpm and deb already call ldconfig at the end of the transaction based on a file trigger, rather that for every package updated.

superm1, 2019-01-06:
Looking more closely at original issue It looks like ./libfwupd wasn't the problem. It was specifically libxmlb.

```
libfwupd.so.2 => /home/sebastian/Downloads/fwupd-1.2.3/build/src/./../libfwupd/libfwupd.so.2 (0x00007fa991b52000)
	libxmlb.so.1 => not found
```

Ldconfig really had to be called after installing libxmlb.

For me I didn't reproduce help2man issue since I was using the subproject version of libxmlb (since I used git checkout of fwupd not a tarball)

superm1, 2019-01-06:
> I've not seen any other projects do this. It would also need to be disabled by default, as rpm and deb already 

Is it common practice to just document needing to do this then? I do wonder if we should bring some discussion to upstream meson regarding this. It should be a common problem.

I wouldn't mind the default behavior to call `ldconfig` and then we disabled it for the packaged targets with some extra argument to `meson` or `ninja` also.

superm1, 2019-01-06:
In the case of a tarball install you won't get that option though for a static library to link. Another thought is to use `LD_RUN_PATH` during building.

superm1, 2019-01-07:
I filed an issue with meson to discuss calling `ldconfig` automatically.

I do think that documentation for libxmlb should be adjusted to avoid this issue, so I've added an issue for that as well.

LinuxOnTheDesktop, 2022-04-29:
I get the same problem when running the Flatpak.

```
$ sudo flatpak install flathub org.freedesktop.fwupd
[. . . .]
$ flatpak list
Name                                          Application ID                              Version        Branch        Installation
[. . . .]
fwupd                                         org.freedesktop.fwupd                       1.8.0          stable        system
[. . . .]
$ sudo -i flatpak run org.freedesktop.fwupd
/app/bin/fwupdtool: error while loading shared libraries: libfwupd.so.2: cannot open shared object file: No such file or directory
```

hughsie, 2022-05-02:
Huh, I get the same thing with the flatpack build; how odd.

hughsie, 2022-05-03:
Okay, works with todays flatpak. Phew.

closed 2022-05-03 by hughsie

LinuxOnTheDesktop, 2022-05-03:
It is not ideal to leave it to luck whether the flatpak version works!