fwupd/fwupd #915: fwupd 1.2.3 Debian : Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out

swidz, 2019-01-05:
```
$ fwupdmgr --version
client version:	1.2.3
compile-time dependency versions
	gusb:	0.2.9
	efivar:	34
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out
```

```
fwupdmgr get-devices
Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out
```

after deleting the `pending.db` file which in my case was not located in `/var/lib/fwupd` but in `/var/local/lib/fwupd` 

```
$ fwupdmgr --version
client version:	1.2.3
compile-time dependency versions
	gusb:	0.2.9
	efivar:	34
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
```

But the file `pending.db` is getting recreated in `/var/local/lib/fwupd `

Please answer the following questions:

    Operating system and version: Debian 9.6 
    How did you install fwupd: build with meson and ninja
    Have you tried rebooting? Yes
    Are you using an NVMe disk? Yes

[pending.db.zip](https://github.com/hughsie/fwupd/files/2729939/pending.db.zip)

hughsie, 2019-01-05:
Can you get the fwupd --verbose output please.

superm1, 2019-01-05:
My guess is the systemd unit isn't getting installed somewhere proper, or that the information needed for dbus activation isn't. 

For example maybe something in /usr/local isn't in default search path on Debian 9.

swidz, 2019-01-05:
~$ fwupd --verbose
bash: fwupd: command not found

```
$ systemctl list-unit-files --type=service
...
fwupd-offline-update.service               static   
fwupd.service                              static
...
```

superm1, 2019-01-06:
And do you have any fwupd related packages installed from Debian repos?

superm1, 2019-01-06:
Regarding this:
```
~$ fwupd --verbose
bash: fwupd: command not found
```
That's to be expected,  `/usr/local/libexec/fwupd/` is not in `PATH` by default.
You would have to run `/usr/local/libexec/fwupd/fwupd -v` to manually execute it.

I guess it would be good to see if that works.

swidz, 2019-01-06:
> And do you have any fwupd related packages installed from Debian repos?

I had fwupd installed from debian stretch repo version (version 0.7.4-2), but I had removed this. It was before a successful build, but after first attepts with the described issues.

I am attaching fwupd verbose ran with and without sudo

[fwupd-verbose.txt](https://github.com/hughsie/fwupd/files/2730575/fwupd-verbose.txt)
[sudo-fwupd-verbose.txt](https://github.com/hughsie/fwupd/files/2730576/sudo-fwupd-verbose.txt)

superm1, 2019-01-06:
Something seems wrong with the dbus setup.
```
15:01:16:0475 FuMain               FuMain: lost name: org.freedesktop.fwupd
```

I think it might be the stuff ending up in `/usr/local/etc/dbus-1` isn't being sourced by default.

swidz, 2019-01-06:
in `/usr/local/etc/dbus-1 ` there is a subdirectory system.d with a single file `org.freedesktop.fwupd.conf`
The file is dated as it was installed with the previous version, but folder created dates are more or less when I completed the build. Can it be that the file is from the older version?

The content of the file `org.freedesktop.fwupd.conf`:

```
<?xml version="1.0" encoding="UTF-8"?> <!-- -*- XML -*- -->

<!DOCTYPE busconfig PUBLIC
 "-//freedesktop//DTD D-BUS Bus Configuration 1.0//EN"
 "http://www.freedesktop.org/standards/dbus/1.0/busconfig.dtd">
<busconfig>

  <!-- This configuration file specifies the required security policies
       for the fwupd to work. -->

  <!-- Only user root can own the fwupd service -->
  <policy user="root">
    <allow own="org.freedesktop.fwupd"/>
  </policy>

 <!-- Allow anyone to call into the service - we'll reject callers using PolicyKit -->
  <policy context="default">
    <allow send_destination="org.freedesktop.fwupd"
           send_interface="org.freedesktop.fwupd"/>
    <allow send_destination="org.freedesktop.fwupd"
           send_interface="org.freedesktop.DBus.Properties"/>
    <allow send_destination="org.freedesktop.fwupd"
           send_interface="org.freedesktop.DBus.Introspectable"/>
    <allow send_destination="org.freedesktop.fwupd"
           send_interface="org.freedesktop.DBus.Peer"/>
  </policy>

</busconfig>   
```

swidz, 2019-01-09:
I have uninstalled the 1.2.3 version with `ninja -C build uninstall` and installed this version via snap.
It works. I was able to update the Unifying Receiver.

I am attaching the output of fwupd --verbose from snap

[snap-sudo-fwupd-verbose.txt](https://github.com/hughsie/fwupd/files/2741280/snap-sudo-fwupd-verbose.txt)

superm1, 2019-01-09:
I can reproduce this error on Ubuntu from source.  If the file `/usr/local/etc/dbus-1/system.d/org.freedesktop.fwupd.conf` is copied into `/etc/dbus-1/system.d` the error goes away.

Additionally the following steps make it go away:
1) Create `/etc/dbus-1/system-local.conf` with following contents
```
<busconfig>
    <includedir>/usr/local/etc/dbus-1/system.d</includedir>
</busconfig
```
2) Reload dbus daemon
```
systemctl reload dbus.service
```

dnoliver, 2021-05-28:
Scratched my head for a while with this error on Ubuntu 20.04:

The wiki say `/usr/local/etc/` but the path in 20.04 seems to be `/usr/local/share`

```shell
root@test-NUC8v5PNK:~/fwupd# cat /etc/dbus-1/system-local.conf
<busconfig>
    <includedir>/usr/local/share/dbus-1/system.d</includedir>
</busconfig>
```

:)

superm1, 2021-05-29:
That is odd to me, but go ahead and add it to the wiki page with a footnote! :)