fwupd/fwupd #920: Make policy kit optional?

superm1, 2019-01-07:
This issue was reported in the Google tracker.
https://bugs.chromium.org/p/chromium/issues/detail?id=862390

> mmm polkit doesn't belong on CrOS.  if fwupd needs it, you'll need to fix fwupd.

It's blocking on bringing fwupd into ChromeOS.

What are your thoughts in making policykit an optional item for use in ChromeOS?

hughsie, 2019-01-07:
> What are your thoughts in making policykit an optional item for use in ChromeOS?

I don't think we can make it optional. Either we'd allow all non-root
actions (causing a security problem) or we'd deny them (which means
nothing works). We have to use some kind of ACL framework, and without
knowing how Google want to use fwupd in ChromeOS we're simply
guessing.

superm1, 2019-01-07:
The current intention is that firmware upgrades will be run automatically and non-interactively.  I don't know the actual mechanism that will invoke them yet (login/logout script, or scheduler or what not).  I'll ask the folks over there to come comment in this thread to talk more about it.

superm1, 2019-01-14:
There is a commit in this branch that achieves this https://github.com/hughsie/fwupd/tree/wip/superm1/polkit but it's not yet confirmed to be necessary.  I think lets keep that branch hanging around and this issue open until decided if that approach is needed.

superm1, 2019-02-14:
To close this loop;
CrOS approach is currently to:
* Preload CAB files in filesystem, in a dynamically built metadata directory.
* launch `fwupdtool update` with a udev rule when devices detected

closed 2019-02-14 by superm1

jedahan, 2020-09-11:
was this merged in? I don't see a way to disable polkit in the docs...

superm1, 2020-09-11:
This was not merged in.  For now PK is mandatory to use the daemon and client solution.  If you want to go without PK then use the tool only.  There is some discussion in the chromium fork to make it a configurable item.
The code is here if you would like to see what they've done:
https://chromium-review.googlesource.com/c/chromiumos/third_party/fwupd/+/2354521

hughsie, 2020-09-11:
>  For now PK is mandatory to use the daemon and client solution

I think we always have to use some kind of authorization policy; letting everyone do everything without any kind of verification would be madness. That said, I'd be open for making this pluggable if there was another authorization daemon/library that was available.