fwupd/fwupd #923: Allow uploader to sign firmware report

hughsie, 2019-01-08:
If we sign the report, and the signer signature matches something in the LVFS vendor whitelist then we could automatically populate the device checksums for a given firmware release. This would allow a QA user to trivially test the firmware on actual hardware and set the attestation checksums as part of the first report.

superm1, 2019-01-08:
Sign it with what?  A GPG key?  I suppose then you would need to also match GPG keys to those at LVFS.

At least on LVFS side I guess you have an email address though - it could be possible to try to look at public keyservers for matching email addresses to get public keys to verify.

hughsie, 2019-01-08:
>  A GPG key

I was pondering just using PKCS-7.

> match GPG keys to those at LVFS

Yes, the LVFS `User` would have a public key set.

superm1, 2019-01-08:
When it comes to keys I guess as long as it's not a pain in the butt to generate the keys it doesn't really matter if it's GPG or PKCS-7.

GPG I think does have that big advantage of public keyring servers, and you can build up web of trust around particular signers.

Another thought that might simply things is that if the firmware is in embargo, imply trust.  Firmware in embargo would always be coming from some sort of QA entity.

hughsie, 2019-10-30:
We did this.

closed 2019-10-30 by hughsie