fwupd/fwupd #926: `fwupd` service crashing on Fedora Silverblue 29

miabbott, 2019-01-09:
```shell
$ fwupdmgr --version
client version: 1.1.4
compile-time dependency versions
        appstream-glib: 0.7.14
        gusb:   0.3.0
        efivar: 35

Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

```shell
$ fwupdmgr get-devices
Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

Please answer the following questions:

- Operating system and version:  Fedora Silverblue 29 (29.20190102.0)
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):  RPM
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes

The `fwupd` service fails to start repeatedly on this kind of host and dumps core:

```shell
$ systemctl status fwupd
● fwupd.service - Firmware update daemon
   Loaded: loaded (/usr/lib/systemd/system/fwupd.service; static; vendor preset: disabled)
   Active: failed (Result: core-dump) since Wed 2019-01-09 09:07:15 EST; 4min 5s ago
     Docs: https://fwupd.org/
  Process: 2020 ExecStart=/usr/libexec/fwupd/fwupd (code=dumped, signal=SEGV)
 Main PID: 2020 (code=dumped, signal=SEGV)

Jan 09 09:07:15 mastershake systemd[1]: Starting Firmware update daemon...
Jan 09 09:07:15 mastershake fwupd[2020]: g_unix_is_system_fs_type: assertion 'fs_type != NULL && *fs_type != '\0'' failed
Jan 09 09:07:15 mastershake fwupd[2020]: g_unix_is_system_device_path: assertion 'device_path != NULL && *device_path != '\0'' failed
Jan 09 09:07:15 mastershake systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Jan 09 09:07:15 mastershake systemd[1]: fwupd.service: Failed with result 'core-dump'.
Jan 09 09:07:15 mastershake systemd[1]: Failed to start Firmware update daemon.
```

```shell
$ journalctl -b -u fwupd
-- Logs begin at Mon 2018-04-30 16:12:58 EDT, end at Wed 2019-01-09 09:10:51 EST. --
Jan 04 16:52:07 mastershake systemd[1]: Starting Firmware update daemon...
Jan 04 16:52:12 mastershake fwupd[2682]: g_unix_is_system_fs_type: assertion 'fs_type != NULL && *fs_type != '\0'' failed
Jan 04 16:52:12 mastershake fwupd[2682]: g_unix_is_system_device_path: assertion 'device_path != NULL && *device_path != '\0'' failed
Jan 04 16:52:13 mastershake systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Jan 04 16:52:13 mastershake systemd[1]: fwupd.service: Failed with result 'core-dump'.
Jan 04 16:52:13 mastershake systemd[1]: Failed to start Firmware update daemon.
Jan 04 16:52:32 mastershake systemd[1]: Starting Firmware update daemon...
Jan 04 16:52:33 mastershake fwupd[3546]: g_unix_is_system_fs_type: assertion 'fs_type != NULL && *fs_type != '\0'' failed
Jan 04 16:52:33 mastershake fwupd[3546]: g_unix_is_system_device_path: assertion 'device_path != NULL && *device_path != '\0'' failed
Jan 04 16:52:33 mastershake systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Jan 04 16:52:33 mastershake systemd[1]: fwupd.service: Failed with result 'core-dump'.
Jan 04 16:52:33 mastershake systemd[1]: Failed to start Firmware update daemon.
Jan 04 16:53:08 mastershake systemd[1]: Starting Firmware update daemon...
Jan 04 16:53:08 mastershake fwupd[4169]: g_unix_is_system_fs_type: assertion 'fs_type != NULL && *fs_type != '\0'' failed
Jan 04 16:53:08 mastershake fwupd[4169]: g_unix_is_system_device_path: assertion 'device_path != NULL && *device_path != '\0'' failed
Jan 04 16:53:08 mastershake systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Jan 04 16:53:08 mastershake systemd[1]: fwupd.service: Failed with result 'core-dump'.
Jan 04 16:53:08 mastershake systemd[1]: Failed to start Firmware update daemon.
Jan 04 16:56:34 mastershake systemd[1]: Starting Firmware update daemon...
Jan 04 16:56:34 mastershake fwupd[4697]: g_unix_is_system_fs_type: assertion 'fs_type != NULL && *fs_type != '\0'' failed
Jan 04 16:56:34 mastershake fwupd[4697]: g_unix_is_system_device_path: assertion 'device_path != NULL && *device_path != '\0'' failed
Jan 04 16:56:34 mastershake systemd[1]: fwupd.service: Main process exited, code=dumped, status=11/SEGV
Jan 04 16:56:34 mastershake systemd[1]: fwupd.service: Failed with result 'core-dump'.
Jan 04 16:56:34 mastershake systemd[1]: Failed to start Firmware update daemon.
Jan 06 10:19:21 mastershake systemd[1]: Starting Firmware update daemon...
[... 61 lines cut ...]
                #0  0x00007f70dc5f0421 __poll (libc.so.6)
                #1  0x00007f70dbe47d5f n/a (libusb-1.0.so.0)
                #2  0x00007f70dc6cc58e start_thread (libpthread.so.0)
                #3  0x00007f70dc5fb6a3 __clone (libc.so.6)
                Stack trace of thread 5864:
                #0  0x00007f70dc5f0421 __poll (libc.so.6)
                #1  0x00007f70dbe41495 n/a (libusb-1.0.so.0)
                #2  0x00007f70dbe4266b libusb_handle_events_timeout_completed (libusb-1.0.so.0)
                #3  0x00007f70dbe426c3 libusb_handle_events (libusb-1.0.so.0)
                #4  0x00007f70dc8fc039 g_usb_context_event_thread_cb (libgusb.so.2)
                #5  0x00007f70dcb532aa n/a (libglib-2.0.so.0)
                #6  0x00007f70dc6cc58e start_thread (libpthread.so.0)
                #7  0x00007f70dc5fb6a3 __clone (libc.so.6)

                Stack trace of thread 5865:
                #0  0x00007f70dc5f0421 __poll (libc.so.6)
                #1  0x00007f70dcb2a3a6 n/a (libglib-2.0.so.0)
                #2  0x00007f70dcb2a762 g_main_loop_run (libglib-2.0.so.0)
                #3  0x00007f70dc9f110a n/a (libgio-2.0.so.0)
                #4  0x00007f70dcb532aa n/a (libglib-2.0.so.0)
                #5  0x00007f70dc6cc58e start_thread (libpthread.so.0)
                #6  0x00007f70dc5fb6a3 __clone (libc.so.6)
```

This was originally reported in the Silverblue forums - https://discussion.fedoraproject.org/t/fwupd-service-fails-to-start/922

hughsie, 2019-01-09:
@miabbott -- does `/sys/firmware/efi/efivars` exist? Does `/sys/firmware/efi`? I can't reproduce on this VM or my Lenovo, so information about your hardware would be good, thanks.

miabbott, 2019-01-09:
@hughsie I'm using a Lenovo T470s laptop with a Samsung NVME drive.

```shell
$ ls -la /sys/firmware/efi/efivars/     
total 0                                                                                      
drwxr-xr-x. 2 root root    0 Jan  4 16:50 .                                             
drwxr-xr-x. 5 root root    0 Jan  4 16:50 ..                                          
-rw-r--r--. 1 root root   26 Jan  4 16:50 AppName-1fd8b79f-0be2-4d57-b241-81c5e24e01a1                
-rw-r--r--. 1 root root   33 Jan  4 16:50 AppPlatform-1fd8b79f-0be2-4d57-b241-81c5e24e01a1         
-rw-r--r--. 1 root root    5 Jan  4 16:50 AuthVarKeyDatabase-aaf32c78-947b-439a-a180-2e144ec37792
-rw-r--r--. 1 root root  116 Jan  4 16:50 Boot0000-8be4df61-93ca-11d2-aa0d-00e098032b8c              
-rw-r--r--. 1 root root  200 Jan  4 16:50 Boot0001-8be4df61-93ca-11d2-aa0d-00e098032b8c       
-rw-r--r--. 1 root root   46 Jan  4 16:50 Boot0010-8be4df61-93ca-11d2-aa0d-00e098032b8c     
-rw-r--r--. 1 root root   54 Jan  4 16:50 Boot0011-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r--. 1 root root   84 Jan  4 16:50 Boot0012-8be4df61-93ca-11d2-aa0d-00e098032b8c        
-rw-r--r--. 1 root root   72 Jan  4 16:50 Boot0013-8be4df61-93ca-11d2-aa0d-00e098032b8c             
-rw-r--r--. 1 root root   80 Jan  4 16:50 Boot0014-8be4df61-93ca-11d2-aa0d-00e098032b8c 
-rw-r--r--. 1 root root   74 Jan  4 16:50 Boot0015-8be4df61-93ca-11d2-aa0d-00e098032b8c             
-rw-r--r--. 1 root root   60 Jan  4 16:50 Boot0016-8be4df61-93ca-11d2-aa0d-00e098032b8c    
-rw-r--r--. 1 root root   64 Jan  4 16:50 Boot0017-8be4df61-93ca-11d2-aa0d-00e098032b8c         
-rw-r--r--. 1 root root   66 Jan  4 16:50 Boot0018-8be4df61-93ca-11d2-aa0d-00e098032b8c            
-rw-r--r--. 1 root root   63 Jan  4 16:50 Boot0019-8be4df61-93ca-11d2-aa0d-00e098032b8c       
-rw-r--r--. 1 root root   69 Jan  4 16:50 Boot001A-8be4df61-93ca-11d2-aa0d-00e098032b8c 
-rw-r--r--. 1 root root   66 Jan  4 16:50 Boot001B-8be4df61-93ca-11d2-aa0d-00e098032b8c  
-rw-r--r--. 1 root root   66 Jan  4 16:50 Boot001C-8be4df61-93ca-11d2-aa0d-00e098032b8c        
-rw-r--r--. 1 root root   69 Jan  4 16:50 Boot001D-8be4df61-93ca-11d2-aa0d-00e098032b8c 
-rw-r--r--. 1 root root   71 Jan  4 16:50 Boot001E-8be4df61-93ca-11d2-aa0d-00e098032b8c   
-rw-r--r--. 1 root root   70 Jan  4 16:50 Boot001F-8be4df61-93ca-11d2-aa0d-00e098032b8c 
-rw-r--r--. 1 root root   72 Jan  4 16:50 Boot0020-8be4df61-93ca-11d2-aa0d-00e098032b8c        
-rw-r--r--. 1 root root   66 Jan  4 16:50 Boot0021-8be4df61-93ca-11d2-aa0d-00e098032b8c     
-rw-r--r--. 1 root root   68 Jan  4 16:50 Boot0022-8be4df61-93ca-11d2-aa0d-00e098032b8c        
-rw-r--r--. 1 root root   66 Jan  4 16:50 Boot0023-8be4df61-93ca-11d2-aa0d-00e098032b8c     
-rw-r--r--. 1 root root    6 Jan  4 16:50 BootCurrent-8be4df61-93ca-11d2-aa0d-00e098032b8c        
-rw-r--r--. 1 root root    8 Jan  4 16:50 BootOptionSupport-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r--. 1 root root   22 Jan  4 16:50 BootOrder-8be4df61-93ca-11d2-aa0d-00e098032b8c           
-rw-r--r--. 1 root root   20 Jan  4 16:50 BootOrderDefault-0b7646a4-6b44-4332-8588-c8998117f2ef         
-rw-r--r--. 1 root root    5 Jan  4 16:50 BootState-60b5e939-0fcf-4227-ba83-6bbed45bc0e3
-rw-r--r--. 1 root root   28 Jan  4 16:50 CapsuleLongModeBuffer-711c703f-c285-4b10-a3b0-36ecbd3c8be2
-rw-r--r--. 1 root root    8 Jan  4 16:50 certdb-d9bee56e-75dc-49d9-b4d7-b534210f637a      
-rw-r--r--. 1 root root  284 Jan  4 16:50 ConIn-8be4df61-93ca-11d2-aa0d-00e098032b8c               
-rw-r--r--. 1 root root  100 Jan  4 16:50 ConInDev-8be4df61-93ca-11d2-aa0d-00e098032b8c           
-rw-r--r--. 1 root root   34 Jan  4 16:50 ConOut-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r--. 1 root root   34 Jan  4 16:50 ConOutDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r--. 1 root root  522 Jan  4 16:50 CpuSetup-b08f97ff-e6e8-4193-a997-5e9e9b0adb32
-rw-r--r--. 1 root root   14 Jan  4 16:50 CpuSetupVolatileData-b08f97ff-e6e8-4193-a997-5e9e9b0adb32
-rw-r--r--. 1 root root 5084 Jan  4 16:50 db-d719b2cb-3d3a-4596-a3bc-dad00e67656f
-rw-r--r--. 1 root root 3728 Jan  4 16:50 dbx-d719b2cb-3d3a-4596-a3bc-dad00e67656f
-rw-r--r--. 1 root root    6 Jan  4 16:50 DIAGSPLSHSCRN-a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380
-rw-r--r--. 1 root root  144 Jan  4 16:50 EPCBIOS-c60aa7f6-e8d6-4956-8ba1-fe26298f5e87
-rw-r--r--. 1 root root   34 Jan  4 16:50 ErrOutDev-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r--. 1 root root   44 Jan  4 16:50 ESRTPLATFORMENTRY-e2acd8e4-a376-47bb-a316-4f1e62a8ca1f
-rw-r--r--. 1 root root   68 Jan  4 16:50 EventLog-57a34c69-4d62-0b7e-86e4-939c5c9b7c93
-rw-r--r--. 1 root root    5 Jan  4 16:50 InitSetupVariable-ec87d643-eba4-4bb5-a1e5-3f3e36b20da9
-rw-r--r--. 1 root root 2549 Jan  4 16:50 KEK-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r--. 1 root root  122 Jan  4 16:50 Key0000-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r--. 1 root root  128 Jan  4 16:50 Key0001-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r--. 1 root root  164 Jan  4 16:50 Key0002-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r--. 1 root root  146 Jan  4 16:50 Key0003-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r--. 1 root root  164 Jan  4 16:50 Key0004-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r--. 1 root root  152 Jan  4 16:50 Key0005-8be4df61-93ca-11d2-aa0d-00e098032b8c
-rw-r--r--. 1 root root  166 Jan  4 16:50 Key0006-8be4df61-93ca-11d2-aa0d-00e098032b8c
[... cut ...]
overlay on /var/lib/containers/storage/overlay/8f5b8664d21e6b1e624bfb3b055704c34e9de534edebf380847c9e3b5c7622ea/merged type overlay (rw,relatime,context="system_u:object_r:container_file_t:s0:c62,c346",lowerdir=
/var/lib/containers/storage/overlay/l/DCSJZKILXRYMGDRK4RO4TJZDVL:/var/lib/containers/storage/overlay/l/W4LPOYQQH43WK5Q3HNSPKC2RAK,upperdir=/var/lib/containers/storage/overlay/8f5b8664d21e6b1e624bfb3b055704c34e9d
e534edebf380847c9e3b5c7622ea/diff,workdir=/var/lib/containers/storage/overlay/8f5b8664d21e6b1e624bfb3b055704c34e9de534edebf380847c9e3b5c7622ea/work)
```

hughsie, 2019-01-09:
Hmm. Are you able to generate a backtrace with debug symbols installed? I've tried to find the problem but I can't see how we could be passing NULL to g_unix_mount_at(). Thanks.

miabbott, 2019-01-09:
@hughsie I'll give that a try today.

miabbott, 2019-01-09:
@hughsie I did this from in a container using a more recent corefile...not sure how useful it is:

```shell
$ gdb /usr/libexec/fwupd/fwupd core.fwupd.0.89bbba03bc9f45d2870278d2e9eeaa14.2020.1547042835000000
GNU gdb (GDB) Fedora 8.2-5.fc29                                       
Copyright (C) 2018 Free Software Foundation, Inc.                                                                                                                                                                  License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>       
This is free software: you are free to change and redistribute it.    
There is NO WARRANTY, to the extent permitted by law.                 
Type "show copying" and "show warranty" for details.                  
This GDB was configured as "x86_64-redhat-linux-gnu".                 
Type "show configuration" for configuration details.                  
For bug reporting instructions, please see:                           
<http://www.gnu.org/software/gdb/bugs/>.                              
Find the GDB manual and other documentation resources online at:      
    <http://www.gnu.org/software/gdb/documentation/>.                                                                                                                                                             
                                                                                  
For help, type "help".                                                                    
Type "apropos word" to search for commands related to "word"...                                
Reading symbols from /usr/libexec/fwupd/fwupd...Reading symbols from /usr/lib/debug/usr/libexec/fwupd/fwupd-1.1.4-1.fc29.x86_64.debug...done.
done.                                                                             
[New LWP 2020]                                                        
[New LWP 2023]                                                        
[New LWP 2022]                                                                                                                                                                                                    
[New LWP 2021]                                                                                                                                                                                                    
[New LWP 2024]                                                                                     
                                                                                    
warning: .dynamic section for "/lib64/libglib-2.0.so.0" is not at the expected address (wrong library or version mismatch?)
                                                                      
warning: .dynamic section for "/lib64/libgio-2.0.so.0" is not at the expected address (wrong library or version mismatch?)
                                                                                                                                                                                                                  
warning: Could not load shared library symbols for 2 libraries, e.g. /usr/lib64/gio/modules/libgioremote-volume-monitor.so.                                                                                       
Use the "info sharedlibrary" command to see the complete listing.                                                                                                                                                 
Do you need "set solib-search-path" or "set sysroot"?                                                                                                                                                             
                                                                                 
warning: Loadable section ".note.gnu.property" outside of ELF segments                                                                                                                                            
                                                                                               
warning: Loadable section ".note.gnu.property" outside of ELF segments              
                                                                                  
warning: Loadable section ".note.gnu.property" outside of ELF segments
                                                                      
warning: Loadable section ".note.gnu.property" outside of ELF segments                     
                                                                      
warning: Loadable section ".note.gnu.property" outside of ELF segments
                                                                      
warning: Loadable section ".note.gnu.property" outside of ELF segments

warning: Loadable section ".note.gnu.property" outside of ELF segments

warning: Loadable section ".note.gnu.property" outside of ELF segments

warning: Loadable section ".note.gnu.property" outside of ELF segments

warning: Loadable section ".note.gnu.property" outside of ELF segments

warning: Loadable section ".note.gnu.property" outside of ELF segments

warning: Loadable section ".note.gnu.property" outside of ELF segments
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib64/libthread_db.so.1".

warning: Loadable section ".note.gnu.property" outside of ELF segments

[... cut ...]
#6  0x00007fbc908362aa in g_test_get_dir (file_type=(unknown: 1946159904)) at gtestutils.c:3679
#7  0x00007fbc903af58e in start_thread (arg=<optimized out>) at pthread_create.c:486
#8  0x00007fbc902de6a3 in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95

Thread 1 (Thread 0x7fbc8efc7280 (LWP 2020)):
#0  0x00007fbc9033df42 in __strcmp_avx2 () at ../sysdeps/x86_64/multiarch/strcmp-avx2.S:102
#1  0x00007fbc906fe3e5 in  () at gunixmounts.c:2462
#2  0x00007fbc9074973b in  () at /lib64/libgio-2.0.so.0
#3  0x0000000000000000 in  ()
```

superm1, 2019-01-09:
If you unmounted your container overlays does it keep happening?

Those would be most suspicious to me to causing the crash.

Could you also install debugging symbols for the glib package corresponding to that crashing library?
libgio-2.0.so.0

If you step into that stack frame you may be able to see what variable/value it's trying to reference.

kirbyfan64, 2019-01-10:
I did some research into this. The code is crashing during `g_unix_mount_at ("/sys/firmware/efi/efivars", NULL)`, though both the call (tried it via gjs) and fwupd work outside of systemd.

hughsie, 2019-01-10:
@kirbyfan64 do you have a method to reproduce this? Are you using fedora-toolbox for example?

beelzetron, 2019-01-10:
I am hitting this very same issue on a Lenovo T450 with Fedora Silverblue 29.

kirbyfan64, 2019-01-10:
@hughsie So far I've managed to reproduce the core issue via:

```
 ryan@fedora  ~  systemctl cat fwupd-test   
# /etc/systemd/system/fwupd-test.service
[Unit]
Description=Firmware update daemon

[Service]
Type=simple
ExecStart=/usr/bin/python3 -c 'from gi.repository import Gio; print(Gio.unix_mount_at("/sys/firmware/efi/efivars"))'
ProtectSystem=full
ReadWritePaths=/var/lib/fwupd /etc/fwupd/remotes.d -/boot/efi -/efi/EFI -/boot/EFI
 ryan@fedora  ~  sudo systemctl start fwupd-test
 ryan@fedora  ~  systemctl status fwupd-test    
● fwupd-test.service - Firmware update daemon
   Loaded: loaded (/etc/systemd/system/fwupd-test.service; static; vendor preset: disabled)
   Active: failed (Result: core-dump) since Thu 2019-01-10 09:30:47 CST; 7s ago
  Process: 8932 ExecStart=/usr/bin/python3 -c from gi.repository import Gio; print(Gio.unix_mount_at("/sy>
 Main PID: 8932 (code=dumped, signal=SEGV)

Jan 10 09:30:47 fedora.refi64.com systemd[1]: Started Firmware update daemon.
Jan 10 09:30:47 fedora.refi64.com python3[8932]: g_unix_is_system_fs_type: assertion 'fs_type != NULL && >
Jan 10 09:30:47 fedora.refi64.com python3[8932]: g_unix_is_system_device_path: assertion 'device_path != >
Jan 10 09:30:47 fedora.refi64.com systemd[1]: fwupd-test.service: Main process exited, code=dumped, statu>
Jan 10 09:30:47 fedora.refi64.com systemd[1]: fwupd-test.service: Failed with result 'core-dump'.
 ✘ ryan@fedora  ~  
```

Dropping `ProtectSystem=full` fixes it.

kirbyfan64, 2019-01-10:
Based on the errors, it seems [this code](https://gitlab.gnome.org/GNOME/glib/blob/master/gio/gunixmounts.c#L524) ends up passing a bunch of NULL values, though I have to figure out why still.

superm1, 2019-01-10:
Presumably this is caused by the jump from systemd 239 to 240 right?  I glanced the `NEWS`  file but nothing jumped out relevant to this to me.
You can downgrade systemd to 239 and confirm though.

kirbyfan64, 2019-01-10:
@superm1 I'm on F29 with systemd 239.

superm1, 2019-01-10:
Can you try adding `PrivateMounts=false` to the systemd unit?

hughsie, 2019-01-10:
The root cause seems to be this: https://gitlab.gnome.org/GNOME/glib/issues/1645

gabx, 2019-01-14:
Very similar errors with lenovo Yoga 20 and Fedora Silverblue 29

superm1, 2019-01-14:
Seems https://gitlab.gnome.org/GNOME/glib/commit/e695ca9f310c393d8f39694f77471dbcb06daa9e needs to be backported to glib to fix it.

Since not a fwupd bug, closing issue.

closed 2019-01-14 by superm1