fwupd/fwupd #927: Jabra SPEAK 510 USB wont update

joakim-tjernlund, 2019-01-10:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.2.3
compile-time dependency versions
	gusb:	0.2.11
	efivar:	37
daemon version:	1.2.3

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
hinkpad T480s Thunderbolt Controller
  DeviceId:             91d92644a1b1b73a0cd6836bf4185d271db90566
  Guid:                 061f2847-061b-5cfb-9c39-3dbfe2b49d69
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  Icon:                 computer
  Created:              2019-01-10

20L7001UMX System Firmware
  DeviceId:             e7cc234e82d08b6a14e246beb6b5f8142091e570
  Guid:                 ebfe8df8-dee7-4692-a721-cbcf5095c5cf
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 2b79e0a0-0801-5fc9-a7a5-90374e924a9b
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.27
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2019-01-10

UEFI Device Firmware
  DeviceId:             cf3ad1e5e6838b09cba64bb5ccb1142dd44995d2
  Guid:                 676af093-2a5c-4238-9c29-db8063a33532
  Guid:                 039bd398-f61a-5c46-b02c-b660d1cf4926
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.55.3510
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-01-10

UEFI Device Firmware
  DeviceId:             323bda90831520867f8c02544efb9711dfca9fb6
  Guid:                 4d254d6e-cd67-477b-97d5-bc3048af45c4
  Guid:                 cc9cf2fe-ba41-532b-bc58-17ca9cc3c63d
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.10
  VersionLowest:        0.1.10
  Icon:                 audio-card
  Created:              2019-01-10

SAMSUNG MZVLW512HMJP-000L7
  DeviceId:             c82980cd3fe66f7d0a8bb83b353cbc271177efd4
  Guid:                 5b3df2da-f745-5fd0-81de-5dafd7f0bf8c
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Serial:               S359NX0J314215
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              7L7QCXY7
  Icon:                 drive-harddisk
  Created:              2019-01-10

Jabra SPEAK 510 USB
  DeviceId:             71413822e87deae707da948bd4b7e537af463c19
  Guid:                 95e2dec1-d070-5257-825c-90cef66bf680
  Guid:                 6695abef-6c60-5cbb-b8c1-2228b2c389e5
  Guid:                 443b9b32-7603-5c3a-bb30-291a7d8d6dbd
  Guid:                 93412def-aba7-50ee-9e2e-cf6925047014
  Guid:                 0f40ad4b-38e9-5a5c-8f42-122b46649eda
  Guid:                 b2a98bf6-dca5-5c9f-b5e8-f0aae82f7c90
  Guid:                 cc878932-9d8b-53f6-8760-2708ef53c61f
  Guid:                 870d03da-0624-50d0-8689-7230e4da212c
  Guid:                 1c123739-37d1-5e32-9b6a-bdbed964851f
  Guid:                 afa5518d-7c55-51cb-8e1c-bf3b11b67dfa
  Guid:                 cee9c5bc-aa12-51eb-a220-d0ef588d7cb2
  Guid:                 571229d5-caed-5cbc-91c1-5c216601a194
  Serial:               50C971FD4051020700
  Plugin:               dfu
  Flags:                updatable|supported|registered
  VendorId:             USB:0x0B0E
  Version:              2.7
  Icon:                 drive-harddisk-usb
  Created:              2019-01-10
  Modified:             2019-01-10
  UpdateState:          failed
  UpdateError:          failed to update_detach using dfu: cannot get device state: USB error on device 0b0e:0420 : Input/output error [-1]

________________________________________________

Devices that were not updated correctly:

 • Jabra SPEAK 510 USB (2.7 → 2.10)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.

```

```shell
$ efibootmgr -v

**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version:
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Gentoo
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Ys
- Is secure boot enabled (only for the UEFI plugin)? No

fwupdmgr -v update
No upgrades for 20L7001UMX System Firmware, current is 0.1.27: 0.1.25=older, 0.1.26=older, 0.1.27=same
No upgrades for UEFI Device Firmware, current is 184.55.3510: 184.55.3510=same
Downloading 2.10 for Jabra SPEAK 510 USB...
(fwupdmgr:17772): FuCommon-DEBUG: 13:37:35.610: creating path /root/.cache/fwupdmgr
(fwupdmgr:17772): FuMain-DEBUG: 13:37:35.617: skpping download as file already exists
(fwupdmgr:17772): Fwupd-DEBUG: 13:37:35.618: Emitting ::status-changed() [decompressing]
Decompressing…           [-                                      ](fwupdmgr:17772): Fwupd-DEBUG: 13:37:35.630: Emitting ::status-changed() [idle]
Decompressing…           [***************************************]
(fwupdmgr:17772): Fwupd-DEBUG: 13:37:35.653: Emitting ::status-changed() [waiting-for-auth]
Authenticating…          [ -                                     ](fwupdmgr:17772): Fwupd-DEBUG: 13:37:35.655: Emitting ::status-changed() [idle]
Authenticating…          [***************************************]
failed to update_detach using dfu: cannot get device state: USB error on device 0b0e:0420 : Input/output error [-1]

I managed to update it once before, to 2.7

joakim-tjernlund, 2019-01-14:
Finally I  unpacked the CAB file and ran dfu-tool write and that worked.
Then I found that the FW on Jabra support site had version 2.25 (vs the 2.10 here) so
I used the dfu-tool again and that worked fine too.

I think Jabra is lagging a bit w.r.t LVFS updates.

hughsie, 2019-01-16:
I can reproduce on my 510 too. Looking into it now.

hughsie, 2019-01-16:
> I think Jabra is lagging a bit w.r.t LVFS updates.

I've pinged my contact there. Fingers crossed.

joakim-tjernlund, 2019-01-16:
Very nice ! Thanks.

joakim-tjernlund, 2019-01-16:
"hughsie closed this in e58fa31 22 minutes ago"
How come is is so hard(not possible?) to extract a patch from github web GUI?

hughsie, 2019-01-16:
@joakim-tjernlund it's probably easier to just build from master. Either that or just `git show commit-id > foo.patch`

joakim-tjernlund, 2019-01-16:
Got it ! :
https://github.com/hughsie/fwupd/commit/e58fa31887c842986953f2eb0a3584e516bd5372.patch

Easier for me to just throw the patch into portage and rebuild 1.2.3 in Gentoo's pkg DB

superm1, 2019-01-16:
@hughsie should this patch come back to other stable branches too at all?

joakim-tjernlund, 2019-01-16:
You migth want look into:
```
Building documentation for libfwupd
Unescaped left brace in regex is deprecated here (and will be fatal in Perl 5.30), passed through in regex; marked by <-- HERE in m/(.*?){ <-- HERE / at /usr/bin/gtkdoc-scan line 735.

hughsie, 2019-01-16:
> should this patch come back to other stable branches too

Yes, but I want to test master on all the devices in my drawer-of-hardware first.

joakim-tjernlund, 2019-01-16:
OT: I notice Dell has added FW for their Dock.
Is Lenovo working on that too? We sure could use that over here.

joakim-tjernlund, 2019-02-13:
> > I think Jabra is lagging a bit w.r.t LVFS updates.

> I've pinged my contact there. Fingers crossed.

I am afraid that didn't work out, there is no JABRA FW update available.