fwupd/fwupd #928: Failed to activate service 'org.desktop.fwupd' timed out - Gnome Software

yerbestpal, 2019-01-10:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.2.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.3

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
client version:	1.2.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.3
[rossmclean@dobby ~]$ fwupdmgr get-devices
81EU System Firmware
  DeviceId:             fb53e738be78b2c5965524c771636da6a0a3546d
  Guid:                 b1b2e56f-15ad-406a-ab1a-f27ff3cb70e3
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 78373260-27eb-51d8-b90e-83c68789035e
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              87.9.4184
  VersionLowest:        82.67.0
  Icon:                 computer
  Created:              2019-01-10

UEFI Device Firmware
  DeviceId:             afcf33fac0bd4d085dad003c37963c75374d9920
  Guid:                 865d322c-6ac7-4734-b43e-55db5a557d63
  Guid:                 eb8b1db3-ac03-5eb3-a638-6bea7ae7b272
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.0.3510
  VersionLowest:        0.0.1000
  Icon:                 audio-card
  Created:              2019-01-10

UEFI Device Firmware
  DeviceId:             ed1ceee8586c676c7f20f76d9644d4529cd13885
  Guid:                 978b4cea-7c60-41f8-b5dd-93797fd3f314
  Guid:                 024b190b-f2b3-5049-a7a2-1bc709b620e0
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.0.1000
  VersionLowest:        0.0.1000
  Icon:                 audio-card
  Created:              2019-01-10

UEFI Device Firmware
  DeviceId:             79afaa9dde013e8e6c9ce03f9f5c5c71640cf980
  Guid:                 c725e332-8c08-450b-97ba-57cccfa0da35
  Guid:                 463979a5-d952-57b5-8bfd-b6d165d1e358
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.0.1000
  VersionLowest:        0.0.1000
  Icon:                 audio-card
  Created:              2019-01-10

UEFI Device Firmware
  DeviceId:             c34a06357b449bfd952f5ebc015d94d4cacd9d52
  Guid:                 234124e9-40b7-43ef-9b5e-97470808d440
  Guid:                 ac60ef3d-2aec-5598-a384-eb71e3dc27ad
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.0.4
  Icon:                 audio-card
  Created:              2019-01-10

HFS256GD9TNG-62A0A
  DeviceId:             03281da317dccd2b18de2bd1cc70a782df40ed7e
  Guid:                 2260c743-963b-566e-93f1-8223c3e6b87b
  Guid:                 8c1c105b-8aad-5ad2-9d67-68d143164128
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               SK hynix
  VendorId:             NVME:0x1C5C
  Version:              80001E00
  Icon:                 drive-harddisk
  Created:              2019-01-10

```

```shell
$ efibootmgr -v
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Fedora 29
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): It was pre-installed.
- Have you tried rebooting? Yes, many times
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? I don't know if I am using a UEFI plugin, but secure boot _is_ enabled.

superm1, 2019-01-10:
duplicate of https://github.com/hughsie/fwupd/issues/928

closed 2019-01-10 by superm1

yerbestpal, 2019-01-10:
@superm1 #928 is this issue.

Taijian, 2019-01-11:
@superm1 Yes, please reopen, because I'm having this exact issue as well. On Arch.

superm1, 2019-01-25:
Sorry please see https://github.com/hughsie/fwupd/issues/926 for the correct suplict