fwupd/fwupd #931: ModemManager plugin: Device locker removes inhibition too soon

aleksander0m, 2019-01-13:
I'm testing the wip/hughsie/libmm branch, and looking at the logs in fwupd and the behavior in MM, to me it seems that the device inhibition is not working completely as expected

From the logs below, my impression is that the device inhibition is via the device locker object lasts for as long as fwupd uses the AT port to send the AT^FASTBOOT command, and once the command is sent, the inhibition is removed right away. Is this expected? For a correct sync with MM, the inhibition should be locked for as long as the firmware update operation is performed, including the whole fastboot operation.

```
13:55:29:0901 FuHistory            remove device DW5821e Snapdragon X20 LTE [8e3d1037d3917b1bd1ab56258a596b0ae68f3162]
13:55:29:0914 FuHistory            add device DW5821e Snapdragon X20 LTE [8e3d1037d3917b1bd1ab56258a596b0ae68f3162]
13:55:29:0922 FuPlugin             performing update_detach() on modem_manager
13:55:29:0922 FuPluginMm           inhibit /sys/devices/pci0000:00/0000:00:1c.4/0000:02:00.0/usb3/3-2/3-2.3
13:55:29:0988 FuPluginMm           uninhibit /sys/devices/pci0000:00/0000:00:1c.4/0000:02:00.0/usb3/3-2/3-2.3
13:55:30:0031 FuDeviceList         plugin modem_manager did not specify a remove delay for 8e3d1037d3917b1bd1ab56258a596b0ae68f3162, so guessing we should wait 10000ms for replug
13:55:30:0031 FuPluginMm           removed modem: /org/freedesktop/ModemManager1/Modem/2
13:55:30:0031 FuPlugin             emit removed from modem_manager: 8e3d1037d3917b1bd1ab56258a596b0ae68f3162
13:55:30:0031 FuDeviceList         ::removed 8e3d1037d3917b1bd1ab56258a596b0ae68f3162
13:55:30:0031 FuPlugin             performing device_removed() on dell_dock
13:55:30:0031 FuPlugin             performing device_removed() on synapticsmst
13:55:31:0819 FuDevice             using 390e8c98-f738-597d-8611-d20839844933 for USB\VID_413C&PID_81D6&REV_0100
13:55:31:0819 FuDevice             using a5d74db8-71c5-501d-92e2-d41046e8b3f3 for USB\VID_413C&PID_81D6
13:55:31:0819 FuDevice             using 795e079d-093b-5503-aa59-35b832480e95 for USB\VID_413C
13:55:31:0819 FuDevice             using e49bd1ea-2f0e-5709-a60c-63e72f607ba2 for USB\CLASS_FF&SUBCLASS_42&PROT_03
13:55:31:0819 FuDevice             using 54f7aa0f-6450-5d0c-ac85-0302809a7f6b for USB\CLASS_FF&SUBCLASS_42
13:55:31:0819 FuDevice             using 4bd986bc-4aaf-5baf-8bc2-9f7928182387 for USB\CLASS_FF
13:55:31:0819 FuPlugin             performing usb_device_added() on fastboot
13:55:31:0819 FuDevice             using 85a1a606abccb8d4875817613963d7e058780c19 for usb:00:02:03
13:55:31:0840 FuPluginFastboot     fastboot version=0.5
```
What I see in the MM logs is that  device is inhibited, but then uninhibited just a bit later on, so MM starts to re-probe all ports right away, and suddenly the ports go away (as the modem was requested to enter fastboot mode)

```
ModemManager[19714]: <info>  [1547387729.956284] Device inhibition setup for uid '/sys/devices/pci0000:00/0000:00:1c.4/0000:02:00.0/usb3/3-2/3-2.3'
...
ModemManager[19714]: <debug> [1547387729.982375] Modem (Dell) '/sys/devices/pci0000:00/0000:00:1c.4/0000:02:00.0/usb3/3-2/3-2.3' completely disposed
ModemManager[19714]: <info>  [1547387730.002184] Device inhibition teardown for uid '/sys/devices/pci0000:00/0000:00:1c.4/0000:02:00.0/usb3/3-2/3-2.3'
ModemManager[19714]: <info>  [1547387730.004631] [device /sys/devices/pci0000:00/0000:00:1c.4/0000:02:00.0/usb3/3-2/3-2.3] creating modem with plugin 'Dell' and '6' ports
...
ModemManager[19714]: <info>  [1547387730.217236] (net/wwan0): released by device '/sys/devices/pci0000:00/0000:00:1c.4/0000:02:00.0/usb3/3-2/3-2.3'
ModemManager[19714]: [/dev/cdc-wdm3] unexpected port hangup!
ModemManager[19714]: [/dev/cdc-wdm3] channel destroyed
ModemManager[19714]: [/dev/cdc-wdm3] unexpected port hangup!
ModemManager[19714]: [/dev/cdc-wdm3] channel destroyed
ModemManager[19714]: <info>  [1547387730.248151] (usbmisc/cdc-wdm3): released by device '/sys/devices/pci0000:00/0000:00:1c.4/0000:02:00.0/usb3/3-2/3-2.3'
ModemManager[19714]: <info>  [1547387730.254056] (tty/ttyUSB0): released by device '/sys/devices/pci0000:00/0000:00:1c.4/0000:02:00.0/usb3/3-2/3-2.3'
ModemManager[19714]: <info>  [1547387730.258025] (tty/ttyUSB1): released by device '/sys/devices/pci0000:00/0000:00:1c.4/0000:02:00.0/usb3/3-2/3-2.3'
ModemManager[19714]: <info>  [1547387730.261825] (tty/ttyUSB2): released by device '/sys/devices/pci0000:00/0000:00:1c.4/0000:02:00.0/usb3/3-2/3-2.3'
ModemManager[19714]: <info>  [1547387730.263650] (tty/ttyUSB3): released by device '/sys/devices/pci0000:00/0000:00:1c.4/0000:02:00.0/usb3/3-2/3-2.3'
ModemManager[19714]: <debug> [1547387730.264203] Removing empty device '/sys/devices/pci0000:00/0000:00:1c.4/0000:02:00.0/usb3/3-2/3-2.3'
```

aleksander0m, 2019-01-13:
@hughsie yeah, so from what I can read in the code, the FuDeviceLocker is setup as an autoptr, and so the inhibition will only be held during fu_mm_device_detach(); i.e. only for the step to send the AT command. I guess we can store the FuDeviceLocker in the private info while the update is in progress, right?

hughsie, 2019-01-13:
> I guess we can store the FuDeviceLocker in the private info while the update is in progress, right?

We can, but maybe the FuMmDevice might get destroyed when it enters fastboot mode? Maybe it's still alive... One to test certainly. The other thing we could do is just hold the inhibit until we notice the tty's disappear, or even just wait a couple of seconds before releasing the inhibit.

aleksander0m, 2019-01-13:
I'm also thinking in the QMI PDC steps after fastboot; wondering if we could hold the inhibition not only until the modem goes away, but also while the modem comes back after fastboot, until we finish the QMI PDC procedure.

aleksander0m, 2019-02-28:
This can be closed, the latest changes in the MM plugin made sure the inhibition is hold for as long as the upgrade procedure is working.

closed 2019-02-28 by aleksander0m