fwupd/fwupd #933: Adding canary capabilities to fwupd

timevortex, 2019-01-14:
Hi Richard,

Opening an issue to discuss the possibilities and options for adding canary capabilities to fwupdmgr.

**Use case**
Assume we have a fleet of 100 linux desktops of same manufacturer & model whose BIOS we would like to update using fwupd on a regular/semi-regular basis. 

We could setup a cron job to simply call "fwupdmgr update" every day on every machine. But if the manufacturer releases a bad update for any reason then the update would be scheduled on the whole fleet at once. So ideally we would want to do a controlled roll-out. Maybe 10% of the fleet runs the commands then 7 days later another 10 or 20% and so on and on according to some schedule until we reach 100%.

Currently the only way to do this would be to write some kind of a separate script which runs everyday and which decides whether to call fwupdmgr on that machine or not. This seems less than ideal so I'm creating this issue to discuss whether it would possible to build some kind canary capabilities into fwupd directly?

**One possibility**
Since this is a managed-fleet use case, we could assume that instead of downloading updates from fwupd.org, the organisation might decide to spin up it's own server and change DownloadURI in /etc/fwupd.conf to point to their own server. So it might be possible to relay the canary information via this. i.e. the XML downloaded from server could indicate rollout percentage and then the machine can self-select whether it belongs in the rollout bucket specified. This will be fuzzy and we won't get perfect distributions each time (i.e. 10% rollout on a fleet of 100 might actually end up including 8 machines or 12 machines depending on how the self-selection is done). However this could be acceptable without having to build a complex server where clients check-in and the server needs tracks individual machines.

There are probably many others ways to do this as well :)

Looking forward to your thoughts.

Regards,
Parth

timevortex, 2019-01-14:
FYI to @pkern who's also interested in this.

superm1, 2019-01-14:
Currently the infrastructure that exists to support something like this is "server" side.  It's possible for an OEM to set something up in their firmware on fwupd.org that will control how many downloads are offered each day.  Since no clients check in automatically this just controls how quickly the whole fleet may "potentially" brick.  Also this is more of a "general" solution than a fleet specific roll out as it is the world's download count, not just your fleet.

> change DownloadURI in /etc/fwupd.conf to point to their own server.

Just FYI you mean to say "create a new fwupd remote in `/etc/fwupd/remotes.d` that points to their own server".

> However this could be acceptable without having to build a complex server where clients check-in and the server needs tracks individual machines.

As a very "simple" solution I might recommend:
1) To spin up your own LVFS instance locally.
2) Set up a signing server to sign the firmware with.
3) Upload firmware to this LVFS instance.  Configure the maximum number of downloads per day on the criteria you deem necessary for staged rollout.
4) On the client side enroll the new signing key and configure the internal LVFS instance remote.
5) On the client side script something that would do the following actions:
`# fwupdmgr update --assume-yes`

That will update all firmware and also send any pending reports for success.  Your fleet will slowly upgrade at the speed you deem necessary.  At some point you can decide that everything has been successful enough and remove the max downloads/day criteria.

Another improvement can be to add support to LVFS to automatically remove that after X downloads with X successful reports.

hughsie, 2019-01-14:
> "simple"

Indeed. On the whole I wanted to keep "fleet management" out of fwupd/LVFS and leave it to tools like Satellite and Fleet Commander. fwupdmgr was designed to be "scriptable" and so usable from Puppet and Ansible. I agree with Mario, in that if you're not watching the failures for your deployment, then there is not much point staging the deployment, and so the standalone LVFS looks tempting, but it's not exactly trivial to set up.

Perhaps we could do something server side for this; e.g. assign the limits for pools, so that anything checking in with the same `pool_id` set in the header gets a limit checked, although that does kinda mean the LVFS has to keep track of the pool IDs, which isn't also awesome from a privacy point of view.

I guess my recommendation here would be Ansible and SSH.

pkern, 2019-01-14:
So right now we have scripted our update process around fwupdate - i.e. push down the cab file and apply it. And then we use the telemetry from that (actual failures from fwupdate and BIOS version post-update) to see if the update worked. Unfortunately the telemetry check is still ad-hoc checking by an engineer.

I think what we are looking for is actually serving the LVFS data ourselves (for various reasons, privacy as well as double checking the data we send). But the question is if we could drop our ad-hoc policy-based updater at least for firmware update (it'd still do settings management for now) by using fwupd with a proper server side implementation. We know what the size of the fleet is, so if fwupd supports downloads-based limiting that'd be fine, too.  I see tracks in the source code, is the downloads-based limiting already implemented and if so, how? By regenerating the firmware list and dropping the update again when the count is reached? (In which case we could probably generate this on the fly per client.)

Regarding telemetry currently in LVFS it looks like it counts successes, failures, "issues" (which I suppose are described [here](https://blogs.gnome.org/hughsie/2018/02/13/fwupd-now-tells-you-about-known-issues/)) and downloads right now. That's a good start, especially when you are privacy conscious. I suppose in an enterprise context it'd be worth looking at the individual clients and see if they came back with a check-in of the right version after it had been handed to them. At the same time I agree that this could also be higher level instrumentation. Reporting back the actual redacted failure messages would be useful, though.

Ansible and SSH is unfortunately not an option with a five digit deployment. Local scripts (like masterless Puppet or even local Ansible playbooks) would obviously work, but if it just calls `fwupdmgr update` then we are still left with the canarying problem and would need to modify the input into the process. For raw `fwupdate` the problem is that we need to get the right files to the right hosts and it feels preferable to make LVFS and fwupd work for that rather than rolling our own. (Although it is still TBD if it is possible for us to reuse LVFS for technical and security reasons - hence us focussing more on the client-side. We'd run a mirror of the public LVFS anyway, so it's unclear if that would even work with the current model. That being said, thank you very much for actually publishing what's behind that web site.)

hughsie, 2019-01-15:
> then we use the telemetry from that (actual failures from fwupdate

Okay, that's awesome. It sounds like you've been doing it before fwupd learned that trick.

> is the downloads-based limiting already implemented

Yes, there's a server-side counter that allows that many clients to download the update per day. See:

[image]

> see if they came back with a check-in of the right version after it had been handed to them

We don't (deliberately) send the installed version back to the LVFS, we only ever request the shared metadata file and then the firmwares if required. Obviously we include this information for a success/failure "Report" but only of the user manually agrees to this. But see below...

> just calls fwupdmgr update then we are still left with the canarying problem

I don't think calling fwupdmgr update in a cron job or playbook in a deployment of this size makes sense.

> it feels preferable to make LVFS and fwupd work for that rather than rolling our own

Yes, I agree. At the moment we've focused almost exclusively on the "consumer" case where the user is essentially their own admin. I've been talking to other companies/agencies that want a more enterprise-y fleet-management-style firmware update experience, much like the vendor-specific tools provide. With the latest attestation work we're doing, with the metadata in the LVFS and the local information in the client we've actually got all the information we need to "enroll" a client into a "pool" and then do domain-wide policy and verification for thousands of hosts. I don't know whether that means working with satellite, cockpit, openstack or any of the other existing solutions.

Unfortunately, although I think such a thing would be a good idea, and possible to incubate in the lvfs-website project I don't personally have any more resources to do something like this. I think you could use the existing LVFS website as a whole-domain internal scheduling agent for firmware, but I'd much rather see someone working on a model where clients can enroll and we can do whole-domain firmware upgrades and attestation. If you have spare resources, I'd certainly be willing to help mentor someone to do this, but for me personally I have to draw the line somewhere for the sake of my sanity.

superm1, 2019-01-15:
> Obviously we include this information for a success/failure "Report" but only of the user manually agrees to this. 

Well if you cron `# fwupdmgr update --assume-yes` then I think you'll be getting that result actually.  The reports should be getting sent in between update checks.

> I'd much rather see someone working on a model where clients can enroll and we can do whole-domain firmware upgrades and attestation

Agreed.  Furthermore, I'd really like to see it done in an open manner that sets an example for how other organizations can and should do it with you guys taking the lead if possible.

hughsie, 2020-08-18:
I think fwupdagent gets us halfway there. As time goes on it'll be easier and easier to deploy things like this.

closed 2020-08-18 by hughsie