fwupd/fwupd #935: The UX capsule is not appended to firmware capsule

Steven5857, 2019-01-15:
```shell
$ fwupdmgr update
After running "fwupdmgr update" for online BIOS firmware update or "fwupdmgr install firmware.cab" for offline BIOS firmware update. The graphic output as Windows WUFU did not show during the capsule update. 

I copied the final firmware file (BIOS.Cap) which was sent to EFI UpdateCapsule() files of Windows WUFU and Linux fwupd and then compared them. I found the Windows WUFU would create a UX capsule and append it after the firmware capsule automatically. But Linux fwupd did not do the same thing, the final firmware capsule image was still the same as what I created.

Did the Linux fwupd will create the UX capsule before calling UpdateCapsule()? If yes, what is the mechanism and where the UX capsule is stored?

Thanks a lot!

Steven

superm1, 2019-01-15:
A few things that come to mind with regards to UX capsule.
1) Which fwupd version are you using?  It needs to be 1.1.x or newer to use integrated UX capsule
2) Are you booted with legacy option ROM enabled?  I believe that may cause BGRT table to not populate properly which might make UX capsule not work.
3) Can you check the debugging logs from when you execute before the reboot?  Make sure it was turned on in those logs.  You can launch daemon with `--daemon-verbose=FuPluginUefi` option to see these logs.

Steven5857, 2019-01-15:
TThanks superm1's commets.

1. Which fwupd version are you using? It needs to be 1.1.x or newer to use integrated UX capsule
   Ans> I used Fedora 29-1.2 + fwupd 1.2.3
2. Are you booted with legacy option ROM enabled? I believe that may cause BGRT table to not 
   populate properly which might make UX capsule not work.
   Ans> No, just UEFI boot was enabled, and legacy boot is disabled.
3. Can you check the debugging logs from when you execute before the reboot? Make sure it was 
   turned on in those logs. You can launch daemon with --daemon-verbose=FuPluginUefi option to see 
   these logs.
   Ans> Do you mean the data which is stored in FWUPDATE_DEBUG_LOG variable? Which reboot? After flash update or before? How to launch daemon with --daemon-verbose=FuPluginUefi option? I directly dumpped the data in FWUPDATE_DEBUG_LOG variable after flash update and before rebooting & booting to Fedora as following.

FWUPDATE_DEBUG_LOG../plugins/uefi/efi/fwupdate.c:462:find_updates(): Found update fwupd-e20bafd3-9914-4f4f-9537-3129e090eb3c-0
../plugins/uefi/efi/fwupdate.c:1232:efi_main(): Adding new capsule
../plugins/uefi/efi/fwupdate.c:641:search_file(): File \EFI\fedora\fw\fwupd-e20bafd3-9914-4f4f-9537-3129e090eb3c.cap searched
../plugins/uefi/efi/fwupdate.c:702:open_file(): Found device
../plugins/uefi/efi/fwupdate.c:709:open_file(): Found volume
../plugins/uefi/efi/fwupdate.c:717:open_file(): Found file
../plugins/uefi/efi/fwupdate.c:1043:add_capsule(): Read file; 8828448 bytes
../plugins/uefi/efi/fwupdate.c:1044:add_capsule(): updates guid: E20BAFD3-9914-4F4F-9537-3129E090EB3C
../plugins/uefi/efi/fwupdate.c:1045:add_capsule(): File guid: E20BAFD3-9914-4F4F-9537-3129E090EB3C
../plugins/uefi/efi/fwupdate.c:1246:efi_main(): n_updates: 1
../plugins/uefi/efi/fwupdate.c:1096:apply_capsules(): QueryCapsuleCapabilities: Success max: 181641216 reset:1
../plugins/uefi/efi/fwupdate.c:1097:apply_capsules(): Capsules: 1
../plugins/uefi/efi/fwupdate.c:1272:efi_main(): fwupdate: Reset System

Steven5857, 2019-01-15:
Sorry, I misunderstood your meaning. I got the message below after I run "sudo /usr/libexec/fwupd/fwupd --daemon-verbose=FuPluginUefi"

10:02:39:0335 FuPluginUefi         no TPM device found
10:02:39:0335 FuPluginUefi         detected BootNext, not cleaning up
10:02:39:0335 FuPluginUefi         ESP mountpoint set as /boot/efi
10:02:39:0336 FuPluginUefi         SecureBoot is: Disabled
10:02:39:0336 FuPluginUefi         BGRT setup failed: BGRT status was 0
10:02:39:0336 FuPluginUefi         UX Capsule support : Disabled

hughsie, 2019-01-15:
> 10:02:39:0336 FuPluginUefi BGRT setup failed: BGRT status was 0

According to https://wiki.osdev.org/Broken_UEFI_implementations status needs to be `1` if the firmware displays a logo on boot. Have we misunderstood the meaning of the BGRT status for the capsule update?

Steven5857, 2019-01-15:
According to Microsoft description about the Status as "The Status field of the BGRT describes whether the image is currently displayed on the screen. This is not applicable to the firmware update display capsule." It seems the Status field only needs to be set when the image is currently displayed on the screen. Let me consult some experts of our company tomorrow and comment it again.

https://docs.microsoft.com/en-us/windows-hardware/drivers/bringup/boot-screen-components

Thanks!

Steven5857, 2019-01-16:
After discussed it with some experts internally, we thought that you might misunderstand the meaning of the Valid bit of BGRT Status field. It does not mean the BGRT images are valid, and it means the current displaying image is a BGRT image and still valid, hence Windows does not need to clear screen and re-display the BGRT image when Windows starts up. The Microsoft's description may is better is clear.

 "The Status field of the BGRT describes whether the image is currently displayed on the screen. This is not applicable to the firmware update display capsule."

BIOS will show a splash logo at POST, and Windows will show a BGRT logo at start up. If the splash logo is equal to BGRT logo, then the BGRT Status can set to 1. OEM/ODM can decide to use same or different logos. Besides, the Status may set to 0 by some BIOS settings in Setup, for example, enable Fast Boot, change CSM status, etc.. Just like Microsoft said "This is not applicable to the firmware update display capsule.", whether the UX capsule is displayed should not rely on the BGRT Status value. You just need to check the BGRT existence in ACPI tables. Windows WUFU can show the UX capsule properly no matter the BGRT Status value is 1 or 0.

Thanks!

hughsie, 2019-01-16:
@Steven5857 thanks for doing that research. In the case where status=0x0, can you still read the other `/sys/firmware/acpi/bgrt/*` values? I think the only thing we use the bgrt value for is to set the yoffset for the "_Updating the system_" localized text we set as the display capsule. The idea was that we'd use the vendor splash, and then under it we'd show the localised updating text. I suppose in the absence of the BGRT we should just put the text a few pixels above the bottom of the screen.

Steven5857, 2019-01-16:
**hughsie:**  In the case where status=0x0, can you still read the other /sys/firmware/acpi/bgrt/* values? 
Yes, I still can read the status, type, version, xoffset, yoffset values, and image size is 294774.

I think the only thing we use the bgrt value for is to set the yoffset for the "Updating the system" localized text we set as the display capsule.
Steven> Yes, I also created a similar bitmap message in the UX capsule during my past tests.

Thanks!

closed 2019-01-16 by hughsie