fwupd/fwupd #941: I could send you an error message of Gnome software about fwupd, but it is impossible to copy and paste that error message. Why?? please make your error messages copy-able.

tuxedotje, 2019-01-16:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
**INSERT OUTPUT HERE**
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
**INSERT OUTPUT HERE**
```

```shell
$ efibootmgr -v
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version:
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
- Have you tried rebooting?
- Are you using an NVMe disk?
- Is secure boot enabled (only for the UEFI plugin)?

tuxedotje, 2019-01-16:
cannot edit message here 
version client version:	1.2.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.3

hughsie, 2019-01-16:
There is no information here. Please reopen if you can type out the error message.

closed 2019-01-16 by hughsie