fwupd/fwupd #946: New plugin that wraps hdparm?

superm1, 2019-01-16:
In looking at the options for SATA disk upgrades, I did see that hdparm actually has support for a variety of download modes.

Taken from hdparm 9.48:
```
 --fwdownload            Download firmware file to drive (EXTREMELY DANGEROUS)
 --fwdownload-mode3      Download firmware using min-size segments (EXTREMELY DANGEROUS)
 --fwdownload-mode3-max  Download firmware using max-size segments (EXTREMELY DANGEROUS)
 --fwdownload-mode7      Download firmware using a single segment (EXTREMELY DANGEROUS)
 --fwdownload-modee      Download firmware using mode E (min-size segments) (EXTREMELY DANGEROUS)
 --fwdownload-modee-max  Download firmware using mode E (max-size segments) (EXTREMELY DANGEROUS)
```

However it also requires the (quite reassuring) `--yes-i-know-what-i-am-doing --please-destroy-my-drive`

So that makes me wonder if it's worth trying to produce a plugin that wraps it.  I haven't talked to any of the storage vendors about this yet though.

hughsie, 2019-01-17:
If you have a storage vendor that's going to provide both the redistributable firmware and also the guidance on what mode to use (it seems much more complex and risky than NVMe...) then I think it's a good idea. My gut instinct is we shouldn't try to wrap the hdparm tool, and like the `nvme` plugin only implement in C the ioctls we need. That way we have full control over the error reporting and progress changed notifications without screen scraping.

hughsie, 2019-01-28:
There's a storage vendor interested in this. I've started on a new plugin for this.