fwupd/fwupd #964: fwupd does not load under kernel 4.20 in Fedora 28

alvarorece, 2019-01-25:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.2.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
It doesn't work but it used to appear (before updating from kernel 4.18) Thunderbolt and Dell Precision 7520 firmware
```
Please answer the following questions:

- Operating system and version: Fedora 28, kernel 4.20
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): default install
- Have you tried rebooting?yes
- Are you using an NVMe disk?no
- Is secure boot enabled (only for the UEFI plugin)?no

The service doesn't load, even if I try to do it manually. It also marks my kernel as tainted. 
dmesg | grep -i 'taint' output:
[  392.193180] CPU: 6 PID: 2460 Comm: fwupd Not tainted 4.20.3-200.fc29.x86_64 #1
[  392.193453] CPU: 6 PID: 2460 Comm: fwupd Tainted: G        W         4.20.3-200.fc29.x86_64 #1

hughsie, 2019-01-25:
> Failed to connect to daemon: Error calling StartServiceByName

If you load it using `/usr/libexec/fwupd/fwupd --verbose` can you upload the log here please.

> It also marks my kernel as tainted

I don't understand that at all. fwupd doesn't ship any kernel modules, although it's possible when it runs it probes some hardware which causes an already-installed nonfree module. That might be related to the crash. Can you find out which module is loaded (i.e. reboot, run `lsmod` before you run fwupd, then `lsmod` after, then generate a diff) and tell us in this bug please? Thanks.

alvarorece, 2019-01-25:
Before running, when loading with verbose and after running it
[before.txt](https://github.com/hughsie/fwupd/files/2796055/before.txt)
[fwupd.txt](https://github.com/hughsie/fwupd/files/2796058/fwupd.txt)
[after.txt](https://github.com/hughsie/fwupd/files/2796060/after.txt)

hughsie, 2019-01-25:
So, it's tainted as you're using the nonfree amdgpu module. If you run fwupd locally, does it crash? If it does, can you run it using gdb and then get a backtrace please? Thanks.

alvarorece, 2019-01-25:
It seems that it doesn't crash, it just stops like in the log i sent you

superm1, 2019-01-25:
Maybe https://github.com/hughsie/fwupd/issues/928

hughsie, 2019-01-25:
@alvarorece do you have glib2 <= 2.58.3?

alvarorece, 2019-01-25:
@hughsie yep, glib2-2.58.2-1

hughsie, 2019-01-25:
Okay, then it's a dupe of #928 -- good lateral thinking @superm1! If you update glib2 the problem will magically go away.

closed 2019-01-25 by hughsie