fwupd/fwupd #966: fwupd service fails to load with kernel 4.20 and 4.19 but it works with 4.18 #964

alvarorece, 2019-01-26:
I will write the output of this from 4.18 kernel because I already did it with 4.20 in the previous issue. Updating glib2 didn't fix the problem, but booting 4.18 worked. Also, I don't know if it is related but my laptop can't shutdown in 4.20.

```shell
$ fwupdmgr --version
client version:	1.2.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.3
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
Precision 7520 System Firmware
  DeviceId:             5607a0e2c8e72eb0eb5fe70a50193b679d91f2a3
  Guid:                 fe771285-6926-4d47-a0bc-0f6e2a71e3ad
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 cbe07384-6566-52a4-9131-8edbe7f99108
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.13.0
  VersionLowest:        0.1.13.0
  Icon:                 computer
  Created:              2019-01-25

Precision 7520 Thunderbolt Controller
  DeviceId:             e3ead746ef02c58aa90db699110e2a82a3d596d6
  Guid:                 28245c9e-29ad-5d00-ad04-1eb80e746c49
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              33.00
  Icon:                 computer
  Created:              2019-01-25
```

```shell
$ efibootmgr -v
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001
Boot0000* ubuntu	HD(1,GPT,f6f08f6c-858c-448d-b547-82c311cc3465,0x800,0x190000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Fedora	HD(1,GPT,b769a0fb-a102-46a7-8acd-6acc55db7605,0x800,0x64000)/File(\EFI\fedora\shimx64.efi)
Boot0003* USB Storage Device	BBS(USB,USB Storage Device,0x0)..BO
Boot0004* CD/DVD/CD-RW Drive	BBS(CDROM,CD/DVD/CD-RW Drive,0x0)..BO
Boot0005* Onboard NIC	BBS(Network,IBA CL Slot 00FE v0109,0x0)..BO
Boot0006* Windows Boot Manager	HD(1,GPT,f6f08f6c-858c-448d-b547-82c311cc3465,0x800,0x190000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...o................
**This is only required if you use the UEFI plugin**
```

```shell
$ efivar -l | grep fw
**This is only required if you use the UEFI plugin**
```
This has no output

```shell
$ tree /boot
/boot
├── config-4.18.16-300.fc29.x86_64
├── config-4.20.3-200.fc29.x86_64
├── efi
│   ├── EFI
│   │   ├── BOOT
│   │   │   ├── BOOTIA32.EFI
│   │   │   ├── BOOTX64.EFI
│   │   │   ├── fbia32.efi
│   │   │   └── fbx64.efi
│   │   └── fedora
│   │       ├── BOOTIA32.CSV
│   │       ├── BOOTX64.CSV
│   │       ├── fonts
│   │       │   └── unicode.pf2
│   │       ├── gcdia32.efi
│   │       ├── gcdx64.efi
│   │       ├── grub.cfg
│   │       ├── grubenv
│   │       ├── grubia32.efi
│   │       ├── grubx64.efi
│   │       ├── mmia32.efi
│   │       ├── mmx64.efi
│   │       ├── shim.efi
│   │       ├── shimia32.efi
│   │       ├── shimia32-fedora.efi
│   │       ├── shimx64.efi
│   │       ├── shimx64-fedora.efi
│   │       └── themes
│   ├── mach_kernel
│   └── System
│       └── Library
│           └── CoreServices
│               └── SystemVersion.plist
├── elf-memtest86+-5.01
├── extlinux
│   ├── cat.c32
│   ├── chain.c32
│   ├── cmd.c32
│   ├── cmenu.c32
│   ├── config.c32
│   ├── cptime.c32
│   ├── cpu.c32
│   ├── cpuid.c32
│   ├── cpuidtest.c32
│   ├── debug.c32
│   ├── dhcp.c32
│   ├── dir.c32
│   ├── disk.c32
│   ├── dmi.c32
│   ├── dmitest.c32
│   ├── elf.c32
│   ├── ethersel.c32
│   ├── gfxboot.c32
│   ├── gpxecmd.c32
│   ├── hdt.c32
│   ├── hexdump.c32
│   ├── host.c32
│   ├── ifcpu64.c32
│   ├── ifcpu.c32
│   ├── ifmemdsk.c32
│   ├── ifplop.c32
│   ├── kbdmap.c32
│   ├── kontron_wdt.c32
│   ├── ldlinux.c32
│   ├── lfs.c32
│   ├── libcom32.c32
│   ├── libgpl.c32
│   ├── liblua.c32
│   ├── libmenu.c32
│   ├── libutil.c32
│   ├── linux.c32
│   ├── ls.c32
│   ├── lua.c32
│   ├── mboot.c32
│   ├── memdisk
│   ├── meminfo.c32
│   ├── menu.c32
│   ├── pci.c32
│   ├── pcitest.c32
│   ├── pmload.c32
│   ├── poweroff.c32
│   ├── prdhcp.c32
│   ├── pwd.c32
│   ├── pxechn.c32
│   ├── reboot.c32
│   ├── rosh.c32
│   ├── sanboot.c32
│   ├── sdi.c32
│   ├── sysdump.c32
│   ├── syslinux.c32
│   ├── vesa.c32
│   ├── vesainfo.c32
│   ├── vesamenu.c32
│   ├── vpdtest.c32
│   ├── whichsys.c32
│   └── zzjson.c32
├── grub2
│   ├── grubenv -> ../efi/EFI/fedora/grubenv
│   └── themes
│       └── system
│           ├── background.png
│           └── fireworks.png
├── initramfs-0-rescue-e2d9b78b16a244f3bfdd07af82abe0e3.img
├── initramfs-4.18.16-300.fc29.x86_64.img
├── initramfs-4.20.3-200.fc29.x86_64.img
├── loader
│   └── entries
├── lost+found
├── memtest86+-5.01
├── System.map-4.18.16-300.fc29.x86_64
├── System.map-4.20.3-200.fc29.x86_64
├── vmlinuz-0-rescue-e2d9b78b16a244f3bfdd07af82abe0e3
├── vmlinuz-4.18.16-300.fc29.x86_64
└── vmlinuz-4.20.3-200.fc29.x86_64

16 directories, 98 files

**This is only required if you use the UEFI plugin**
**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:

- Operating system and version: Fedora 29
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): default install
- Have you tried rebooting?Yes
- Are you using an NVMe disk?No
- Is secure boot enabled (only for the UEFI plugin)?I thought it wasn't, but apparently fwupd says that it is.

superm1, 2019-02-04:
Have you tried a few times and you're absolutely sure it's 4.20 that is keeping the service from loading?

alvarorece, 2019-02-24:
Yes, I have even tried with 4.20.10-200 but still does not work. It also marks my kernel as tainted.

superm1, 2019-02-25:
OK so I think what we need is to see the journal output from `fwupd.service` to see what it says about this failure.

If it's not showing anything about the failure, please modify the unit and add `--verbose` to the `ExecStart` and reload units to try again.

xenithorb, 2019-03-04:
This is related to `amdgpu` terribly sorry that I did not capture the kdump or log, but I did `blacklist amdgpu` (not needing the discrete unit) and then it started to work again. 

Perhaps if I have more time at the end of the day I will undo things and capture logs

superm1, 2019-03-04:
Sounds good.  Re-open if you've got more information about the failure scenario.

closed 2019-03-04 by superm1