fwupd/fwupd #967: failed to download https:// cdn.fwupd.org/downloads/firmware.xml.gz.asc

FlailAway, 2019-01-27:
Mint 19, dual boot with windows 8.1 and about 10 minutes after booting this morning I got the error message.

> _Unable to download firmware updates from "fwupd":
[*/cabinet/*/source/fwupd/*] failed to download https://
cdn.fwupd.org/downloads/firmware.xml.gz.asc: Cannot
resolve hostname_

To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.0.9
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
This treturned nothing
```

```shell
$ efibootmgr -v
**This is only required if you use the UEFI plugin**
****I don't know if I do or not so here it is anyway
BootCurrent: 0002
Timeout: 0 seconds
BootOrder: 0002,0001,0003
Boot0001* HDD: 	HD(1,GPT,b77a4115-a747-4318-99f9-de4eebbd5542,0x800,0x100000)/File(\EFI\ubuntu\grubx64.efi)RC
Boot0002* ubuntu	HD(1,GPT,b77a4115-a747-4318-99f9-de4eebbd5542,0x800,0x100000)/File(\EFI\ubuntu\shimx64.efi)
Boot0003* Windows Boot Manager	HD(1,GPT,b77a4115-a747-4318-99f9-de4eebbd5542,0x800,0x100000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...i................
```

```shell
$ efivar -l | grep fw
**This is only required if you use the UEFI plugin**
****I don't know if I do or not so here it is anyway
Command 'efivar' not found, but can be installed with:

sudo apt install efivar
```

```shell
$ tree /boot
**This is only required if you use the UEFI plugin**
**** I don't know if I do or not so here it is anyway
Command 'tree' not found, but can be installed with:

sudo apt install tree

**We're looking for any `.cap` files and the location of `fwupx64.efi`**
```

Please answer the following questions:
- Operating system and version:
RELEASE=19.1
CODENAME=tessa
EDITION="Cinnamon" 4.0.9
DESCRIPTION="Linux Mint 19.1 Tessa"
DESKTOP=Gnome
TOOLKIT=GTK

- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
**** I have no idea. The error message popped up this morning

- Have you tried rebooting?
**** Yes and no error message this time
- Are you using an NVMe disk?
**** I have no idea. How would I check?
- Is secure boot enabled (only for the UEFI plugin)?
**** I have no idea. How would I check?

hughsie, 2019-01-27:
Can you download https://cdn.fwupd.org/downloads/firmware.xml.gz.asc in your browser?

tstoeckler, 2019-02-20:
I'm having this problem as well on Fedora 29. And yes, I can download https://cdn.fwupd.org/downloads/firmware.xml.gz.asc in the browser and in the terminal with `wget`, as well.

If you need any other information, please let me know.

Here's the relevant info for my system:
```
$ fwupdmgr --version
client version:	1.2.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.3

$ fwupdmgr get-devices
Carp 1.0 Thunderbolt Controller
  DeviceId:             82c9bfc3e8faee37d7a57fe579545945820f03d4
  Guid:                 15105a7c-9b49-5e4b-a8ed-3a2084d61805
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               HP
  VendorId:             TBT:0x00F0
  Version:              20.00
  Icon:                 computer
  Created:              2019-02-18

HP Spectre x360 Convertible 13-ae0xx System Firmware
  DeviceId:             b6d3640817c3334f2f36890539014b6156262f18
  Guid:                 5582f387-7110-4075-ad36-a19523e1c89f
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 918bc068-a413-530e-946a-f551a20ca1cb
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              15.34.0
  VersionLowest:        15.34.0
  Icon:                 computer
  Created:              2019-02-18

THNSN51T02DUK TOSHIBA
  DeviceId:             f954c7acdf5fab61aeaca1cd71d29ea5ade6992f
  Guid:                 bfe8d16e-b612-5f26-9526-0123dd679e7d
  Guid:                 a7e9412b-a3cf-5819-9029-105efad314f1
  Guid:                 ad9fe8f7-cdc4-52c9-9fea-31b6f4988ffa
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Toshiba America Info Systems
  VendorId:             NVME:0x1179
  Version:              5KHA4102
  Icon:                 drive-harddisk
  Created:              2019-02-18

$ efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0002,0004,9999
Boot0000* Fedora	HD(1,GPT,e096d15d-851b-4f3c-9eff-c4c69405b4cd,0x800,0x82000)/File(\EFI\fedora\shimx64.efi)
Boot0002* Windows Boot Manager	HD(1,GPT,e096d15d-851b-4f3c-9eff-c4c69405b4cd,0x800,0x82000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}...+................
Boot0004* Solid State Disk	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-08-0D-02-00-1E-CD-CC)/HD(1,GPT,e096d15d-851b-4f3c-9eff-c4c69405b4cd,0x800,0x82000)..BO
Boot9999* USB Drive (UEFI)	PciRoot(0x0)/Pci(0x1d,0x0)/USB(16,0)..BO

$ efivar -l | grep fw
# No output here

$ tree /boot | grep -E '(.cap|fwup64.efi)'
# No output here
```
* Operating system and version:
  RELEASE=29
  CODENAME=Fedora
  EDITION=Workstation
  DESCRIPTION=Fedora 29 (Workstation Edition)
  DESKTOP=Gnome
  TOOLKIT=GTK

* How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
Was installed out-of-the-box, but I guess "via `dnf`" would be the correct answer?!

* Have you tried rebooting?
Yes, error still appears. It doesn't come everytime Software is opened but it does reliably come "at some point", regardless of any reboots or anything.

* Are you using an NVMe disk?
Yes (see above).

* Is secure boot enabled (only for the UEFI plugin)?
Yes.

TheRoarkster, 2019-03-07:
Same issue here, also on Fedora 29.  Same answers as @tstoeckler, except not using a NVMe disk and secure boot is disabled.

vicosoft, 2019-03-09:
I have the same problem. I can access the URL by Firefox and wget, but not from fwupd. I have Ubuntu 18.10

ghisvail, 2019-04-01:
Although, it might be caused by something else (GNOME Software ?), since I also get a `failed to download https://odrs.gnome.org/1.0/reviews/api/ratings: Cannot resolve hostname`.

ghisvail, 2019-04-01:
I don't think so. I am using the standard network setup from the Debian installer.

RolandHeath, 2019-04-02:
Also having this issue, on Fedora.
As requested:
fwupdmgr refresh --verbose
Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)

superm1, 2019-04-02:
Can you share your fwupd journal?

RolandHeath, 2019-04-02:
Definitely disregard my last comment. 
Filesystem and some kernel modules were totally borked, and after fixing them I'm not getting the same issue at all. 
Will let you know if I run into it again with more info.

xcorvis, 2019-04-17:
I was having this issue for some time (Ubuntu 18.10) but after running `sudo fwupdmgr refresh --verbose` the error went away and I can't reproduce it. The output of the command listed nothing notable - it just downloaded the file and worked like it should. Later I thought to check the .cache/fwupd permissions. Some files were owned by root. I ended up blowing the whole dir away and re-running without sudo, everything still seems fine.

tstoeckler, 2019-05-01:
Here is the output of the command:
```
$ fwupdmgr refresh --verbose
(fwupdmgr:5311): FuCommon-DEBUG: 21:42:08.280: creating path /home/tobias/.cache/fwupd
(fwupdmgr:5311): Fwupd-DEBUG: 21:42:08.281: looking for os-release at /etc/os-release
(fwupdmgr:5311): FuMain-DEBUG: 21:42:08.282: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz to /home/tobias/.cache/fwupd/lvfs-metadata.xml.gz
Metadaten werden abgerufen https://cdn.fwupd.org/downloads/firmware.xml.gz
(fwupdmgr:5311): dconf-DEBUG: 21:42:08.296: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:5311): dconf-DEBUG: 21:42:08.297: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:5311): dconf-DEBUG: 21:42:08.297: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:5311): dconf-DEBUG: 21:42:08.297: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:5311): dconf-DEBUG: 21:42:08.297: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:5311): dconf-DEBUG: 21:42:08.304: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:5311): dconf-DEBUG: 21:42:08.305: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:5311): dconf-DEBUG: 21:42:08.305: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:5311): dconf-DEBUG: 21:42:08.306: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:5311): dconf-DEBUG: 21:42:08.306: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:5311): FuMain-DEBUG: 21:42:08.489: progress: 1%
Herunterladen …          [                                       ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.489: progress: 2%
Herunterladen …          [                                       ] Weniger als eine Minute verbleiben…(fwupdmgr:5311): FuMain-DEBUG: 21:42:08.489: progress: 4%
Herunterladen …          [*                                      ] Weniger als eine Minute verbleiben…(fwupdmgr:5311): FuMain-DEBUG: 21:42:08.490: progress: 6%
Herunterladen …          [**                                     ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.490: progress: 8%
Herunterladen …          [***                                    ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.504: progress: 9%
Herunterladen …          [***                                    ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.504: progress: 11%
Herunterladen …          [****                                   ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.504: progress: 13%
Herunterladen …          [*****                                  ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.505: progress: 15%
Herunterladen …          [*****                                  ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.505: progress: 17%
Herunterladen …          [******                                 ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.506: progress: 18%
Herunterladen …          [*******                                ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.506: progress: 20%
Herunterladen …          [*******                                ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.518: progress: 25%
Herunterladen …          [*********                              ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.518: progress: 31%
Herunterladen …          [************                           ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.519: progress: 36%
Herunterladen …          [**************                         ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.519: progress: 42%
Herunterladen …          [****************                       ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.522: progress: 47%
Herunterladen …          [******************                     ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.523: progress: 52%
Herunterladen …          [********************                   ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.533: progress: 58%
Herunterladen …          [**********************                 ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.533: progress: 63%
Herunterladen …          [************************               ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.534: progress: 68%
Herunterladen …          [**************************             ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.534: progress: 74%
Herunterladen …          [****************************           ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.537: progress: 79%
Herunterladen …          [******************************         ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.537: progress: 84%
Herunterladen …          [********************************       ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.539: progress: 90%
Herunterladen …          [***********************************    ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.540: progress: 95%
Herunterladen …          [*************************************  ](fwupdmgr:5311): FuMain-DEBUG: 21:42:08.540: progress: 100%
Herunterladen …          [***************************************]
(fwupdmgr:5311): FuMain-DEBUG: 21:42:08.542: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz.asc to /home/tobias/.cache/fwupd/lvfs-metadata.xml.gz.asc
Signatur wird abgerufen https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
(fwupdmgr:5311): FuMain-DEBUG: 21:42:08.555: progress: 100%
```

choeffer, 2019-05-16:
Got the same error message in GNOME Software this morning with Debian 10. But

```
fwupdmgr refresh --verbose
(fwupdmgr:7228): FuCommon-DEBUG: 06:24:23.155: creating path /home/pink/.cache/fwupd
(fwupdmgr:7228): Fwupd-DEBUG: 06:24:23.155: looking for os-release at /etc/os-release
(fwupdmgr:7228): FuMain-DEBUG: 06:24:23.155: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz to /home/pink/.cache/fwupd/lvfs-metadata.xml.gz
Metadaten werden abgerufen https://cdn.fwupd.org/downloads/firmware.xml.gz
(fwupdmgr:7228): dconf-DEBUG: 06:24:23.157: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:7228): dconf-DEBUG: 06:24:23.157: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:7228): dconf-DEBUG: 06:24:23.157: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:7228): dconf-DEBUG: 06:24:23.157: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:7228): dconf-DEBUG: 06:24:23.157: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:7228): dconf-DEBUG: 06:24:23.158: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:7228): dconf-DEBUG: 06:24:23.158: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:7228): dconf-DEBUG: 06:24:23.158: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:7228): dconf-DEBUG: 06:24:23.158: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:7228): dconf-DEBUG: 06:24:23.158: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:7228): FuMain-DEBUG: 06:24:23.238: progress: 1%
Herunterladen …          [                                       ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.239: progress: 2%
Herunterladen …          [                                       ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.239: progress: 4%
Herunterladen …          [*                                      ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.248: progress: 6%
Herunterladen …          [**                                     ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.250: progress: 7%
Herunterladen …          [**                                     ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.250: progress: 9%
Herunterladen …          [***                                    ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.251: progress: 11%
Herunterladen …          [****                                   ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.252: progress: 12%
Herunterladen …          [****                                   ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.254: progress: 14%
Herunterladen …          [*****                                  ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.255: progress: 16%
Herunterladen …          [******                                 ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.256: progress: 17%
Herunterladen …          [******                                 ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.259: progress: 19%
Herunterladen …          [*******                                ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.260: progress: 24%
Herunterladen …          [*********                              ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.261: progress: 29%
Herunterladen …          [***********                            ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.262: progress: 34%
Herunterladen …          [*************                          ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.263: progress: 39%
Herunterladen …          [***************                        ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.268: progress: 44%
Herunterladen …          [*****************                      ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.269: progress: 49%
Herunterladen …          [*******************                    ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.270: progress: 54%
Herunterladen …          [*********************                  ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.271: progress: 59%
Herunterladen …          [***********************                ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.272: progress: 64%
Herunterladen …          [************************               ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.273: progress: 69%
Herunterladen …          [**************************             ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.274: progress: 74%
Herunterladen …          [****************************           ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.275: progress: 79%
Herunterladen …          [******************************         ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.276: progress: 84%
Herunterladen …          [********************************       ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.277: progress: 89%
Herunterladen …          [**********************************     ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.278: progress: 94%
Herunterladen …          [************************************   ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.279: progress: 99%
Herunterladen …          [************************************** ](fwupdmgr:7228): FuMain-DEBUG: 06:24:23.280: progress: 100%
Herunterladen …          [***************************************]
(fwupdmgr:7228): FuMain-DEBUG: 06:24:23.282: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz.asc to /home/pink/.cache/fwupd/lvfs-metadata.xml.gz.asc
Signatur wird abgerufen https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
(fwupdmgr:7228): FuMain-DEBUG: 06:24:23.292: progress: 100%
```
Was working fine.

tarach, 2019-05-17:
Hi. I'm having the same issue just now. Getting 503 when I'm trying to connect directly ( fresh ubuntu 19.04 installation ) 2 days ago it worked out of the box. I don't know whether I messed up my configuration or the service just doesn't work right now.

```
$ sudo fwupdmgr refresh --verbose
(fwupdmgr:2760): FuCommon-DEBUG: 16:02:01.687: creating path /home/sebastian/.cache/fwupd
(fwupdmgr:2760): Fwupd-DEBUG: 16:02:01.687: looking for os-release at /etc/os-release
(fwupdmgr:2760): FuMain-DEBUG: 16:02:01.688: downloading https://cdn.fwupd.org/downloads/firmware.xml.gz to /home/sebastian/.cache/fwupd/lvfs-metadata.xml.gz
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
(fwupdmgr:2760): GLib-GIO-DEBUG: 16:02:01.695: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.695: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.695: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.695: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.695: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.695: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.695: unwatch_fast: "/system/proxy/" (active: 0, establishing: 1)
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.695: unwatch_fast: "/system/proxy/http/" (active: 0, establishing: 1)
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.695: unwatch_fast: "/system/proxy/https/" (active: 0, establishing: 1)
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.695: unwatch_fast: "/system/proxy/ftp/" (active: 0, establishing: 1)
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.695: unwatch_fast: "/system/proxy/socks/" (active: 0, establishing: 1)
(fwupdmgr:2760): GLib-GIO-DEBUG: 16:02:01.697: _g_io_module_get_default: Found default implementation libproxy (GLibproxyResolver) for ‘gio-proxy-resolver’
(fwupdmgr:2760): GLib-DEBUG: 16:02:01.698: posix_spawn avoided (fd close requested) 
(fwupdmgr:2760): GLib-GIO-DEBUG: 16:02:01.700: _g_io_module_get_default: Found default implementation gnutls (GTlsBackendGnutls) for ‘gio-tls-backend’
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.715: watch_established: "/system/proxy/" (establishing: 0)
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.715: watch_established: "/system/proxy/http/" (establishing: 0)
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.715: watch_established: "/system/proxy/https/" (establishing: 0)
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.715: watch_established: "/system/proxy/ftp/" (establishing: 0)
(fwupdmgr:2760): dconf-DEBUG: 16:02:01.715: watch_established: "/system/proxy/socks/" (establishing: 0)
(fwupdmgr:2760): FuMain-DEBUG: 16:02:03.044: ignoring status code 503 (Backend unavailable, connection timeout)

Failed to download https://cdn.fwupd.org/downloads/firmware.xml.gz: Service Unavailable
```

hughsie, 2019-05-17:
@tarach nope, it's everyone. The AMS datacenter is down: https://status.scaleway.com/

aplatypus, 2019-07-13:
Is there no way to work around this?

hughsie, 2019-07-15:
@aplatypus depends on the failure you're seeing. The datacenter wasn't down since May...

aplatypus, 2019-07-16:
@hughsie  well from the GUI i just get told I'm unable to update.  That's about as non-specific as it gets as far as I'm concerned.  

When that message shows in the app store / software updates area, you may not click on the message or the URL shown.  There is no hit or path to a log file to check.  It is a UX _cul de sac_ _imho_.

However ... I used the command example given earlier:

    fwupdmgr refresh --verbose

and it failed with "Permission denied."  A clue I hear shouted from the crowd ... Perhaps.

    sudo fwupdmgr refresh --verbose

100% done / successful I'm guessing.

So here's a thing, does the software manager need to run as admin?

ndias, 2019-07-26:
I have the same problem and I can download via browser.
I'm using Fedora 30

# fwupdmgr refresh 
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz

(fwupdmgr:3074): GLib-GIO-CRITICAL **: 11:38:54.041: g_file_new_for_path: assertion 'path != NULL' failed

Failed to download https://cdn.fwupd.org/downloads/firmware.xml.gz: SSL handshake failed

hughsie, 2019-07-29:
> GLib-GIO-CRITICAL **: 11:38:54.041: g_file_new_for_path

Can you do this please:

    sudo dnf debuginfo-install fwupd
    export G_DEBUG=fatal-criticals
    gdb --args /usr/bin/fwupdmgr refresh

then wait for the "crash", then do:

    t a a bt

then paste the backtrace. Thanks.

ndias, 2019-07-29:
Sorry, I've done that but nothing happens!!

$ export G_DEBUG=fatal-criticals
$ gdb --args /usr/bin/fwupdmgr refresh
GNU gdb (GDB) Fedora 8.3-3.fc30
Copyright (C) 2019 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-redhat-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/fwupdmgr...
Reading symbols from /usr/lib/debug/usr/bin/fwupdmgr-1.2.10-1.fc30.i386.debug...
(gdb) t a a bt
(gdb)

hughsie, 2019-07-29:
Gah, sorry, I forgot a step, sorry. Type `r` and enter before the `bt` command.

ndias, 2019-07-29:
Sorry about this, but I found the problem, for some reason the version installed was the 32 bit version instead of the 64 bit version, after remove and install the right version everything works as expected.

closed 2019-09-30 by stale[bot]

mbetrifork, 2021-03-23:
It's unfortunate that this issue is closed. The problem persists to this day, on a fresh 20.10 install from 2021. Why has this not been fixed ages ago?

superm1, 2021-03-23:
> It's unfortunate that this issue is closed. The problem persists to this day, on a fresh 20.10 install from 2021. Why has this not been fixed ages ago?

Can you please open a new issue with your situation?