fwupd/fwupd #989: Got this error "Could not apply capsule update: Not Found"

sudoyang, 2019-02-01:
Found update fwupd-798ffd60-f10e-4ac4-8939-c8beabfe55b4-0
File \EFI\ubuntu\fw\fwupd-798ffd60-f10e-4ac4-8939-c8beabfe55b4-0 searched
Could not apply capsule update: Not Found
fwupdate: Could not apply capsules: Not Found
start_image() returned Not Found

My machine is X1 carbon 5gen

```shell
$ fwupdmgr --version

client version:	1.1.2
daemon version:	1.1.2
compile-time dependency versions
	appstream-glib:	0.7.12
	gusb:	0.2.11
	efivar:	34

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices

20HRCTO1WW System Firmware
  DeviceId:             1cbe298fc17877b6883e85560778ac0812f0a385
  Guid:                 798ffd60-f10e-4ac4-8939-c8beabfe55b4
  Guid:                 8f58d6d2-9753-5def-8169-1ef0971b1edc
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.35
  VersionLowest:        0.1.14
  Icon:                 computer
  Created:              2019-02-01
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             4af4b60cae789891a862f14e995f9b0bcb75c2e2
  Guid:                 1c7df930-c51b-43f2-af4e-01239686ff2d
  Guid:                 4b1468ba-bb93-51e0-952b-76b21589040b
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.55.3510
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-02-01

UEFI Device Firmware
  DeviceId:             6b3fcb33b92ea5cdd94954b276ad0b63ed14f9bb
  Guid:                 74997a6b-1adf-4b12-b994-401f06ea8c72
  Guid:                 1fe412bc-cb71-5d49-8812-8198a006fa58
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.19
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-02-01

ThinkPad X1 Carbon Thunderbolt Controller
  DeviceId:             8dad23fff29e05c18d0fd24ee216a115e052b146
  Guid:                 89d9d1e6-9e4c-5f07-b5c1-603da3d61835
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
[... 392 lines cut ...]
- Are you using an NVMe disk?
No
- Is secure boot enabled (only for the UEFI plugin)?
I tried both. But either enable or not give me same error messages. 

I was able to use it update bios successfully. I remember one time I delete my EFI partition by mistake and ask my friend help me to use some command to restore the EFI partition. Maybe the restore causes the issue here?

Below it should be a set of commands used to restore EFI partition. It might be wrong since I do not remember it clearly.

 9607  sudo grub-install /dev/sda1
 9608  cat /etc/fastb
 9609  cat /etc/fstab
 9610  ls -l /
 9611  ls -l /boot
 9612  ls -l /boot/efi
 9613  sudo vi /etc/fstab
 9614  sudo mount /boot/efi
 9615  monut
 9616  mount
 9617  sudo grub-mkconfig
 9618  sudo grub-mkconfig --help
 9619  sudo grub-install --help
 9620  sudo grub-install /dev/sda
 9621  ls -l efi
 9622  ls -l efi/EFI

superm1, 2019-02-04:
Can you guys retry using the [fwupd snap](https://github.com/hughsie/fwupd/wiki/fwupd-snap)?

sudoyang, 2019-02-05:
@superm1  I tried but it failed. Please see the picture.

[image]

@ilya40umov  You can see there is a EFI label.

[image]

superm1, 2019-02-25:
decided Duplicate of #1037

closed 2019-02-25 by superm1

n-peugnet, 2019-12-10:
> What helped in my case was a full re-install where I let the installer create all necessary partitions for UEFI boot.
> I suspect this issue might have something to do with some properties of `/boot/efi` partition which the installer (unlike me) would set properly.
> Can you run GParted and check if EFI partition has the following flags: `boot, esp`?

Hi @ilya40umov, I am almost in the exact same case (except that I added that partition afterwards as I first installed linux for a legacy BIOS)  and I would like not to reinstall the system. Could you please give me more infos on the specificity of your working partition ?

Here is mine, I added the `esp` flag because of you comment but it is still not working:
[image]

By the way, I don't really think that it's the same issue as the one pointed as duplicate.

megatog615, 2023-07-12:
For anyone arrive here from Google(or what have you), I have solved this problem by ensuring the ESP partition is labelled "ESP"(no quotes) and the esp and boot flags are enabled. You can enable the esp flag in parted, which automatically added the boot flag as well. Previously, my ESP partition was labelled "ESD" and had only the msftdata flag enabled. Keep in mind enabling the esp flag will disable msftdata for some reason.

In any case, I was able to install the firmware by doing this.

sliedes, 2025-01-01:
I saw this happen (on a ThinkPad E14 Gen 6) when I had a USB stick with an ESP partition (that I had used to do other firmware updates) inserted. My best guess is that fwupdx64.efi or whatever tried to load the capsules from that instead of the EFI partition that I booted from. Unplugging the USB stick allowed the update to succeed.