fwupd/fwupd #994: 8bitdo: support for "Retro recievers"

pixel-stuck, 2019-02-03:
```shell
$ fwupdmgr --version
client version:	1.2.4
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.4
```

Please answer the following questions:

- Operating system and version: Arch Linux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting?: yes
- Are you using an NVMe disk?: no
- Is secure boot enabled (only for the UEFI plugin)?: N/A

More of an enhancement, but the console-side receivers are not supported. From what I can gather, they seem to be similar to the actual controllers themselves; the receivers seem to have gotten an update at some point and, so far, looking at the update files they seem similar format-wise and the USB ID for bootloader mode is only a nybble different (0483:5760 rather than 0483:5750). Any chance of getting support for these?

hughsie, 2019-02-03:
If you add the required VID/PID in the ebitdo.quirk file (and maybe the plugin?) does the firmware update work?

superm1, 2019-02-04:
To make the whole thing easier, you can prototype the update without the CAB by using `fwupdtool` and the argument 'install-blob' instead of `fwupdmgr`.

superm1, 2019-02-04:
Once you add the VID/PID to the quirks (and possibly the plugin) you should be able to call `/usr/libexec/fwupd/fwupdtool install-blob $FILE` and choose your device in the prompt.  If it works to update it, then that's all there is to it to prove flashing worked.

pixel-stuck, 2019-02-04:
Alright; looks like the update did work! It now registers as version 1.31 where it used to register as version 1.30.

There was one unexpected thing:
```
Writing…                 [***************************************] Less than one minute remaining…
failed to get ACK for mark firmware as successful: unexpected device response
```
(possibly due to it expecting it to re-enumerate as the normal VID/PID?)
But this didn't appear to be a problem. The receiver functions as normal, with the expected "more reliable initial connection" that the update mentions in the changelog.

superm1, 2019-02-04:
Well that's good to hear that it was mostly successful.
Could you submit a PR with the changes you had to make for the VID/PID to recognize the new device?

> failed to get ACK for mark firmware as successful

That makes me think that
```
	if (!fu_ebitdo_device_send (self,
				 FU_EBITDO_PKT_TYPE_USER_CMD,
				 FU_EBITDO_PKT_CMD_UPDATE_FIRMWARE_DATA,
				 FU_EBITDO_PKT_CMD_FW_UPDATE_OK,
				 NULL, 0,
```

Caused the device to reboot or needed a delay between commands or something.  Would you be able to experiment with what was happening?  Some thoughts:
1) look at `lsusb` throughout the update to see which VID/PID present
2) Set `--plugin-verbose=ebitdo` on command line to turn on ebitdo debugging (`FWUPD_EBITDO_VERBOSE`)
3) Try to add a delay etc between send and receive

You should be able to reinstall the same update using `install-blob` again.

pixel-stuck, 2019-02-04:
I noticed that the receiver immediately went back into it's normal mode (searching for my controller while plugged into the PC) could that be the issue? It registers as an Xbox controller when it's in that mode

pixel-stuck, 2019-02-05:
A question: the controller FW won't work for the receiver (I assume anyway) and vice versa; how will fwupd know the difference between the FWs?

superm1, 2019-02-05:
Normally fwupd creates GUIDs that describe the device.  These GUIDs are put into the metadata (XML) that maps the CAB file to the correct GUID.

It's entirely possible for extra checking to happen on fwupd level (or device level) but that is not guaranteed.  Hence `fwupdtool install-blob` is a very dangerous command and shouldn't generally be used for anything other than adding support for a device or debugging.

pixel-stuck, 2019-02-05:
So the different PID should end up mapping to a different GUID and then it'd be separate from the controllers? Neat. Anyway, I'll PR my changes soon :)

vanfanel, 2019-02-19:
Hello guys! So, would it be possible to update USB the Receiver (2.4Gz, red brick model) using current git version of fwupd??

hughsie, 2019-02-24:
What's the U00xxxx number laser etched into the metal?

Steltek, 2019-07-26:
Used this on the USB wireless adapter: http://www.8bitdo.com/wireless-usb-adapter/

by adding these lines:

```
DeviceInstanceId=USB\VID_0483&PID_5760]
Plugin = ebitdo
Flags = is-bootloader
```

It seems to successfully have updated to 1.26 (from 1.22), but with the same error about not getting the ACK.

```
Version:             1.26
Destination Address: 8003400
Destination Length:  65536
Installing on 8Bitdo  …  [  -                                    ]
Writing…                 [  -                                    ]Cancelled
Writing…                 [***************************************]failed to get ACK for mark firmware as successful: unexpected device response
```

zeus86, 2019-08-28:
> [...]
> and running
> `/usr/lib/fwupd/fwupdtool install-blob 8Bitdo_USB_RR_Firmware_V1.26.dat`
> 
> It seems to successfully have updated to 1.26 (from 1.22), but with the same error about not getting the ACK.
> 
> ```
> Version:             1.26
> Destination Address: 8003400
> Destination Length:  65536
> Installing on 8Bitdo  …  [  -                                    ]
> Writing…                 [  -                                    ]Cancelled
> Writing…                 [***************************************]failed to get ACK for mark firmware as successful: unexpected device response
> ```

I can confirm, that Steltek's method works also for the NES Retro Receiver (the one for the Real NES, not the NES Mini), but i got the same error after flashing the receivers, because they boot up in normal mode after flashing. In normal Mode the Receivers are not recognized by fwupdmgr, but in flashing mode they are. I successfully updated 2 NES Retro Receivers (this ones: https://www.8bitdo.com/retro-receiver-nes/ ) from V1.31 to V1.34 with this method, but the error of the failing ACK is still present.

Version used: 
``` 
# fwupdmgr --version
client version: 1.2.10
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.2.10
``` 

Distro/Kernel:
```
Arch x86_64 / 5.2.9-arch1-1-ARCH
```

Device Details (flashing mode): 
```
fwupdmgr get-devices 
8Bitdo  
  DeviceId:             71413822e87deae707da948bd4b7e537af463c19
  Guid:                 6c1f10fc-1878-500a-939f-7db26c827953
  Guid:                 bc1739b5-313d-5a82-a123-95e5dc4a5702
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|registered|is-bootloader
  Vendor:               8Bitdo
  VendorId:             USB:0x0483
  Version:              1.34
  VersionFormat:        pair
  Icon:                 input-gaming
  InstallDuration:      120
  Created:              2019-08-28
```
dmesg (flashing mode):
```
[ 9620.934034] hid-generic 0003:0483:5760.0006: hiddev3,hidraw0: USB HID v1.10 Device [8BitdoJoy 8Bitdo  ] on usb-0000:00:14.0-2/input0
[ 9636.144645] usb 2-2: USB disconnect, device number 26
[ 9958.351733] usb 2-2: new full-speed USB device number 27 using xhci_hcd
[ 9958.493344] usb 2-2: New USB device found, idVendor=0483, idProduct=5760, bcdDevice= 2.00
[ 9958.493351] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 9958.493355] usb 2-2: Product: 8Bitdo  
[ 9958.493358] usb 2-2: Manufacturer: 8BitdoJoy
[ 9958.493361] usb 2-2: SerialNumber: BootMod
```
(SerialNumber is "Receiver" in normal mode)

...unfortunately there seem to be none auto-updateable fw-versions for this reveiver by LVFS.

hughsie, 2019-08-28:
> but the error of the failing ACK is still present

Maybe we can add a quirk for these devices, so they ignore this problem and report success?

superm1, 2019-10-18:
I think that this PR should sort out these devices: https://github.com/fwupd/fwupd/pull/1487

I don't actually have one myself, so I can't test it though.

PR #1487 (2019-10-18): ebitdo: if the device has the will disappear flag, don't wait for ack…
… (Fixes: #994)

Some 8bitdo receivers seem to automatically reboot when done writing,
so just stop if we have that flag and notice this behavior.

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit cbbde24e (2019-10-18): ebitdo: if the device has the will disappear flag, don't wait for ack (Fixes: #994)

Some 8bitdo receivers seem to automatically reboot when done writing,
so just stop if we have that flag and notice this behavior.
closed 2019-10-18 by superm1