fwupd/fwupd #1007: dell-dock stopped enumerating

superm1, 2019-02-07:
I noticed with master today that a dock supported by dell-dock stopped enumerating.

I bisected it to figure out what's going on.  Here's the bisect log:
```
$ git bisect log
git bisect start
# bad: [387bda4667b8c4b8a2474fab086fc85b27804aeb] fu-tool: Correct a crash caused by calling fwupdtool update
git bisect bad 387bda4667b8c4b8a2474fab086fc85b27804aeb
# good: [8ece8874b5588f74805ba1ee283af295d2a276d6] Release fwupd 1.2.4
git bisect good 8ece8874b5588f74805ba1ee283af295d2a276d6
# good: [5d7aa4094e0a4a6a8fa973ea5756b7dbef7bb20e] trivial: fu-tool: Don't show root warning when redirected into logging
git bisect good 5d7aa4094e0a4a6a8fa973ea5756b7dbef7bb20e
# bad: [90d3075a5d1de211a35457a1ddd7d2073d52ffeb] wacom-usb: The touch address is big endian
git bisect bad 90d3075a5d1de211a35457a1ddd7d2073d52ffeb
# bad: [040ac626b5bef0bfcd008df4f7faba992189d6bc] dfu: Fill holes when reading SREC files
git bisect bad 040ac626b5bef0bfcd008df4f7faba992189d6bc
# bad: [1bc432316a4ae49cc90d431b602bdd330b55336f] Defer the InstanceID->GUID hashing until after setup
git bisect bad 1bc432316a4ae49cc90d431b602bdd330b55336f
# good: [2472e45ac69f856fc033f4ddc5a787fc69258be6] ata: Check for USB enclosures as well
git bisect good 2472e45ac69f856fc033f4ddc5a787fc69258be6
# first bad commit: [1bc432316a4ae49cc90d431b602bdd330b55336f] Defer the InstanceID->GUID hashing until after setup
```

Here's the relevant snippet from the daemon log.

```
16:28:34:0813 FuEngine             Emitting PropertyChanged('Status'='loading')
Loading…                 [-                                      ]16:28:34:0813 FuEngine             using plugins: dell_dock
16:28:34:0813 FuEngine             no plugin specified for USB device 0bda:8153
16:28:34:0813 FuEngine             no plugin specified for USB device 0bda:0413
16:28:34:0813 FuEngine             no plugin specified for USB device 0bda:0487
16:28:34:0813 FuEngine             no plugin specified for USB device 1d6b:0003
16:28:34:0814 FuEngine             no plugin specified for USB device 413c:b06e
16:28:34:0814 FuEngine             no plugin specified for USB device 413c:b06f
16:28:34:0814 FuEngine             no plugin specified for USB device 0bda:402e
16:28:34:0814 FuEngine             no plugin specified for USB device 0bda:5413
16:28:34:0814 FuEngine             no plugin specified for USB device 0bda:5487
16:28:34:0814 FuEngine             no plugin specified for USB device 1d6b:0002
16:28:34:0814 FuEngine             no plugin specified for USB device 1d6b:0003
16:28:34:0814 FuEngine             no plugin specified for USB device 0489:e0a2
16:28:34:0814 FuEngine             no plugin specified for USB device 0bda:58f4
16:28:34:0814 FuEngine             no plugin specified for USB device 1d6b:0002
```

superm1, 2019-02-07:
The two VID/PID that were supposed to enumerate were these:
```
[DeviceInstanceId=USB\VID_413C&PID_B06F]
Name = Unprobed Dell accessory endpoint
Plugin = dell_dock
[DeviceInstanceId=USB\VID_413C&PID_B06E]
Name = Unprobed Dell accessory endpoint
Plugin = dell_dock
```

hughsie, 2019-02-07:
I was worried 1bc432316a4ae49cc90d431b602bdd330b55336f would break something like this. I guess we need to actually check GUIDs for `Guid=` and check Instance IDs for `DeviceInstanceId=` rather than treating them both like GUIDs?

closed 2019-02-07 by hughsie