fwupd/fwupd #1018: Logitech Unifying Receiver of K400+ Keyboard Bricked After Firmware Update

Starkillua, 2019-02-11:
Hello,
I've updated two unifying receivers, one linked to a K750 keyboard and the second to a K400+ keyboard. Since the update, the K400+ receiver does not work anymore. 

```shell
$ sudo fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc
```

```shell
$ sudo fwupdmgr update
No upgrades for Unifying Receiver, current is RQR12.07_B0029: RQR12.07_B0029=same
No upgrades for Unifying Receiver, current is 24.000000: RQR24.05_B0029=older
```

```shell
$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.2.4
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

```shell
$ fwupdmgr get-devices
CT500MX500SSD1
  DeviceId:             4869f6ccf087a28c26998f57bcfa7bea8dfa5056
  Guid:                 b77c6ec6-d187-5029-9d3c-e3196465fbee
  Guid:                 4a382f63-5319-5261-b05d-4c598d8bcd8b
  Guid:                 ea728598-6b78-58ff-ad8a-e82eb571af3b
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              M3CR010
  Icon:                 drive-harddisk
  Created:              2019-02-11

ST4000DM004-2CV104
  DeviceId:             68c94c6fb21a2e851da70bded878b1cc7f4b10a8
  Guid:                 cbbc9f2c-ceae-52da-8c30-5873a0229fdd
  Guid:                 a0d04afd-14d4-58d3-8f22-3a82777f1dad
  Guid:                 260e3820-44e9-5a3f-8d20-82acec2aa0a1
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0001
  Icon:                 drive-harddisk
  Created:              2019-02-11

ST4000DM004-2CV104
  DeviceId:             15c5c40528ff138602d2cd91f75fe3afd77e8a2c
  Guid:                 cbbc9f2c-ceae-52da-8c30-5873a0229fdd
  Guid:                 a0d04afd-14d4-58d3-8f22-3a82777f1dad
  Guid:                 260e3820-44e9-5a3f-8d20-82acec2aa0a1
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0001
  Icon:                 drive-harddisk
  Created:              2019-02-11

Unifying Receiver
  DeviceId:             9c8fcae272cc4f7146f11ab2d01ac5fc94212a6a
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.07_B0029
  VersionBootloader:    BOT01.02_B0014
  Icon:                 preferences-desktop-keyboard
  Created:              2019-02-11

Unifying Receiver
  DeviceId:             5bafeb7fb4712b6a3acb28fae20bc1a6aa3547b7
  Guid:                 de81631f-2117-51f8-bcb1-4f874a293b29
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 941f4133-39e5-56a0-9df6-902547436c5c
  Guid:                 afa5518d-7c55-51cb-8e1c-bf3b11b67dfa
  Guid:                 cee9c5bc-aa12-51eb-a220-d0ef588d7cb2
  Guid:                 571229d5-caed-5cbc-91c1-5c216601a194
  Summary:              A miniaturised USB wireless receiver (bootloader)
  Plugin:               unifying
  Flags:                updatable|supported|registered|is-bootloader
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              24.000000
  VersionBootloader:    BOT03.01_B0008
  Icon:                 preferences-desktop-keyboard
  Created:              2019-02-11
```
I run : 

Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.1 LTS
Release:	18.04
Codename:	bionic
Linux jelaet 4.20.6-042006-generic #201901310331 SMP Thu Jan 31 08:33:33 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

I also tried the snap version : 

sudo /snap/bin/fwupdmgr --version
client version:	1.2.4
checkout info:	1.2.4
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.4

sudo /snap/bin/fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading?             [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

sudo /snap/bin/fwupdmgr update
No upgrades for Unifying Receiver, current is RQR12.07_B0029: RQR12.07_B0029=same
No upgrades for Unifying Receiver, current is 24.000000: RQR24.05_B0029=older

(Multiple reboots have been performed, no NVMe and no secure boot)

This is very similar issue to https://github.com/hughsie/fwupd/issues/655
I tried the recommendations given in this closed issue without success.

Thanks in advance.

hughsie, 2019-02-11:
Do you have two unifying receivers attached to the same computer? The 2nd receiver seems in bootloader mode (not bricked) and is recoverable from fwupd -- but you might need to detach the other device first.

Starkillua, 2019-02-11:
Thanks for your reply.
I've not a wired keyboard close at hand, that's why I have 2 receiver including a functional one for another keyboard.
However I've put the commands in a bash file, with a 30s sleep allowing me to detach this device in the meantime.

#!/bin/bash
sleep 30
/snap/bin/fwupdmgr get-devices --verbose
/snap/bin/fwupdmgr refresh --verbose
/snap/bin/fwupdmgr update --verbose
[log.txt](https://github.com/hughsie/fwupd/files/2852634/log.txt)

I still have the same output :+1: 
No upgrades for Unifying Receiver, current is 24.000000: RQR24.05_B0029=older

Thanks

hughsie, 2019-02-11:
I'll take a look in the morning. R

hughsie, 2019-02-12:
I deliberately failed a unifying flash (ctrl-c the daemon when writing) and that put my device in bootloader mode. Which that patch in the PR I was able to just `fwupdmgr update` back to working hardware. I can do you a Fedora rpm of git master if that helps, although a git master snap is a bit beyond my capability.

Starkillua, 2019-02-12:
Thanks for having a look and taking time to fix that issue.
It would be great if you can provide an rpm, then I should be able to fix my receiver from a live version of Fedora.
I'll take you updated if that works.

Thanks

closed 2019-02-12 by hughsie

hughsie, 2019-02-12:
@Starkillua unsigned RPMs available here: https://people.freedesktop.org/~hughsient/fedora/

superm1, 2019-02-12:
There should be a git master snap automatically generated on the edge channel you can use.

Starkillua, 2019-02-12:
Thank you. I'm not familiar of snap but 
```shell
sudo snap info fwupd --verbose
name:      fwupd
summary:   Firmware updates for Linux
publisher: Richard Hughes
contact:   https://github.com/hughsie/fwupd
license:   unset
description: |
  The full fwupd firmware updating stack for Linux.  Can be used to perform updates on all supported
  devices via https://fwupd.org
commands:
  - fwupd.dfu-tool
  - fwupd.fwupdmgr
  - fwupd.fwupdtool
services:
  fwupd:             simple, enabled, active
notes:               
  private:           false
  confinement:       classic
  devmode:           false
  jailmode:          false
  trymode:           false
  enabled:           true
  broken:            false
  ignore-validation: false
snap-id:      HpOj37PuyuaMUZY0NQhtwnp7oS5P8u5R
tracking:     stable
refresh-date: 11 days ago, at 18:05 CET
channels:
  stable:    1.2.4             2019-02-01 (543) 21MB classic
  candidate: 1.2.4             2019-02-01 (543) 21MB classic
  beta:      1.2.4             2019-02-01 (543) 21MB classic
  edge:      1.2.4-28-g0b0fab8 2019-02-06 (551) 21MB classic
installed:   1.2.4                        (543) 21MB 
```
I've tried to install the rpm from a live Fedora, and also tried to convert rpm to deb with alien.
In both cases I had the following issue : 

fwupdmgr: error while loading shared libraries: libxmlb.so.1: cannot open shared object file: No such file or directory

Thanks for your help.

superm1, 2019-02-12:
To [switch snap channels](https://docs.snapcraft.io/channels/551):

```
# sudo snap refresh --channel=edge fwupd
```

Although currently the snap build is failing (see https://github.com/hughsie/fwupd/issues/1023).  So it's not updated to include hughsie's fix.  Once that bug is fixed you can switch snap channels and grab the one from edge.

Starkillua, 2019-02-12:
The unifying receiver has been correctly updated thanks to your fixes.
Thank you again for your help.

3r1k0, 2023-08-31:
After trying many versions, to revive my logitech k400+ keyboard, I found this link https://support.logi.com/hc/en-us/articles/360035037273 and run FirmwareUpdateTool_3.2.276959_x64.exe, with that it was able to complete the flashed DFU and revived.