fwupd/fwupd #1023: snap build is failing

superm1, 2019-02-12:
Noticed it's failing with this error:

```
FAILED: cc -Ilibfwupd/libfwupd@@fwupd@sha -Ilibfwupd -I../libfwupd -Ilibfwupd/.. -I../libfwupd/.. -I/usr/include/gio-unix-2.0/ -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/libsoup-2.4 -I/usr/include/libxml2 -I/usr/include/json-glib-1.0 -I/build/fwupd/stage/include -I/build/fwupd/stage/usr/include -fdiagnostics-color=always -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c99 -O0 -g -fstack-protector-strong -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-unknown-pragmas -Wno-discarded-qualifiers -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wvla -Wwrite-strings -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -D_BSD_SOURCE -D_XOPEN_SOURCE=700 -fPIC -pthread '-DG_LOG_DOMAIN="Fwupd"' '-DLOCALSTATEDIR="/var"'  -MD -MQ 'libfwupd/libfwupd@@fwupd@sha/fwupd-common.c.o' -MF 'libfwupd/libfwupd@@fwupd@sha/fwupd-common.c.o.d' -o 'libfwupd/libfwupd@@fwupd@sha/fwupd-common.c.o' -c ../libfwupd/fwupd-common.c
../libfwupd/fwupd-common.c: In function ‘g_ascii_string_to_unsigned’:
../libfwupd/fwupd-common.c:606:2: error: ‘errno’ undeclared (first use in this function)
  errno = 0;
  ^
../libfwupd/fwupd-common.c:606:2: note: each undeclared identifier is reported only once for each function it appears in
../libfwupd/fwupd-common.c:612:42: error: ‘ERANGE’ undeclared (first use in this function)
      (saved_errno != 0 && saved_errno != ERANGE) ||
                                          ^
ninja: build stopped: subcommand failed.
```

I'm fairly surprised that no other builds failed this, which makes me wonder if we need to try to get the snap build back into CI again.  Maybe using another CI service like Circle CI so we don't slow down the travis builds harder?

hughsie, 2019-02-12:
> error: ‘errno’ undeclared

Can you try with this commit pls^^?

> Maybe using another CI service like Circle CI

Is CircleCI faster than Travis?

superm1, 2019-02-12:
> Is CircleCI faster than Travis?

Well i was thinking run it "in addition" to travis so they run in parallel.

hughsie, 2019-02-12:
If your volunteering to set it up, be my guest :) I've had my fill of setting up CI this week...

superm1, 2019-02-12:
> Can you try with this commit pls^^?

Confirmed it fixed the problem.

> If your volunteering to set it up, be my guest :) I've had my fill of setting up CI this week...

It at least [looks](https://circleci.com/docs/2.0/build-publish-snap-packages/) straight forward.
Can you turn it on for the project, and then we can try to run it on the branch I just created (`wip/superm1/circleci-snap`)

closed 2019-02-12 by superm1