fwupd/fwupd #1035: Update fails on Dell XPS 9560: fwup_set_update_statuses()... Invalid Parameter

hedgepigdaniel, 2019-02-18:
[image]

```shell
$ fwupdmgr --version
client version:	1.2.4
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.4
```

```shell
$ fwupdmgr get-devices
XPS 15 9560 Thunderbolt Controller
  DeviceId:             534f59e5f2c8d7b76d23e43cf62c8653ca5c449d
  Guid:                 72533768-6a6c-5c06-994a-367374336810
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              26.01
  Icon:                 computer
  Created:              2019-02-18

XPS 15 9560 System Firmware
  DeviceId:             f49999b6e3af5b4bf8ad1f2ed2922b76df457271
  Guid:                 34578c72-11dc-4378-bc7f-b643866f598c
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 18a8f546-f58a-556b-ae29-00fc15163b7d
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.11.0
  VersionLowest:        0.1.11.0
  Icon:                 computer
  Created:              2019-02-18
  UpdateState:          success

XPS 15 9560 TPM 2.0
  DeviceId:             cc2b222929056ab3e9a208df9f86b7d9a74ae514
  Guid:                 bc891321-25aa-54f6-86e7-633d065b8d5d
  Summary:              Platform TPM device
  Plugin:               uefi
  Flags:                internal|require-ac|supported|registered
  Vendor:               Dell Inc.
  Version:              1.3.1.0
  Icon:                 computer
  Created:              2019-02-18

PTN5002
  DeviceId:             672c087de09848d9e7ee32aa1dea2fbeb8b81e6b
  Guid:                 d5ec8dcb-f9b1-59c2-959b-f9bd882d0927
  Guid:                 a01d9cb7-dc1c-52dc-88ad-ba94f473681a
  Guid:                 f3f4f68c-a613-57d2-ac51-0bbcf03ae9c5
  Guid:                 073430ea-f120-5ba5-ab70-cc4fe2fe1fd9
  Guid:                 ab182806-032b-5b7b-994f-b0713031d2b6
  Guid:                 c418fefd-0db6-5152-a4da-7a7527183cd1
  Plugin:               dfu
  Flags:                updatable|registered
  Vendor:               NXP
  VendorId:             USB:0x1FC9
  Version:              1.0
  Icon:                 drive-harddisk-usb
  Created:              2019-02-18

THNSN5512GPUK NVMe TOSHIBA 512GB
  DeviceId:             04e17fcf7d3de91da49a163ffe4907855c3648be
  Guid:                 bfe8d16e-b612-5f26-9526-0123dd679e7d
  Guid:                 a7e9412b-a3cf-5819-9029-105efad314f1
  Guid:                 ad9fe8f7-cdc4-52c9-9fea-31b6f4988ffa
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Toshiba America Info Systems
  VendorId:             NVME:0x1179
  Version:              5KDA4103
  Icon:                 drive-harddisk
  Created:              2019-02-18

```

```shell
$ efibootmgr -v
BootNext: 0003
BootCurrent: 0001
Timeout: 0 seconds
BootOrder: 0001,0000,0002,0003
Boot0000* Windows Boot Manager	HD(1,GPT,3bcf4097-b8d2-4615-bd73-c65a37810fd2,0x800,0x12c000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* Linux Boot Manager	HD(1,GPT,3bcf4097-b8d2-4615-bd73-c65a37810fd2,0x800,0x12c000)/File(\EFI\systemd\systemd-bootx64.efi)
Boot0002* EFI Shell	PciRoot(0x0)/Pci(0x14,0x0)/USB(0,0)/HD(1,MBR,0xbb117e08,0x800,0x3ea000)/File(\BOOT\bootx64.efi)
Boot0003* Linux Firmware Updater	HD(1,GPT,3bcf4097-b8d2-4615-bd73-c65a37810fd2,0x800,0x12c000)/File(\EFI\arch\fwupdx64.efi)
```

```shell
$ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-34578c72-11dc-4378-bc7f-b643866f598c-0
```

```shell
$ tree /boot
**INSERT OUTPUT HERE**
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-34578c72-11dc-4378-bc7f-b643866f598c-0
```

Please answer the following questions:

- Operating system and version: Arch Linux, 4.20.8-arch1-1-ARCH
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? many times
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2019-02-18:
Believe this was just fixed in master.

https://github.com/hughsie/fwupd/commit/49448a99d4af74dc2def2b9e7b50eddfa18859c8

Can you please try that?

hedgepigdaniel, 2019-02-18:
Yes that does fix it, thanks :)

hughsie, 2019-02-18:
Looks like I need to tag a release when I get back from PTO. Assume there's going to be a new release in ~7 days time.

closed 2019-02-19 by superm1