fwupd/fwupd #1036: 8bitdo SF30 Pro/ SN30 Pro incorrect firmware from LVFS repo!

mainmachine, 2019-02-18:
8bitdo SF30 Pro controller has an incompatible firmware file in the LVFS repo.

Big tipoff is the firmware version jump - 1.29 to 6.10

Look like they made a mistake in submitting the firmware from the SF30 - the SF30 *Pro* is a different controller.

Checking their support page shows that the latest firmware for the SF30 Pro/SN30 Pro is indeed 1.29, and the latest for the SF30/SN30 is 6.10.

I think the 6.10 for the SF30 Pro/SN30 Pro needs to be pulled immediately. I thought I had bricked my controller, luckily I was able to downgrade, but I had to use the snap version of fwupd to do so, as the deb version in the Bionic repos won't allow a downgrade for some reason.

I was able to recover, but we should save others from that pointless struggle. :)

```shell
$ fwupdmgr --version
client version:	1.2.4
checkout info:	1.2.4
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.4
```

```shell
$ fwupdmgr get-devices
8Bitdo SF30 Pro
  DeviceId:             ef5791f75bea792344af5e07f52d4bb54563221c
  Guid:                 e95bfb47-11f9-5afd-856e-fbc45730456d
  Guid:                 269b3121-097b-50d8-b9ba-d1f64f9cd241
  Guid:                 5bc60e62-f461-58bd-91b6-7de3392f8bd6
  Guid:                 afa5518d-7c55-51cb-8e1c-bf3b11b67dfa
  Guid:                 cee9c5bc-aa12-51eb-a220-d0ef588d7cb2
  Guid:                 571229d5-caed-5cbc-91c1-5c216601a194
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
  Summary:              A redesigned classic game controller
  Plugin:               ebitdo
  Flags:                updatable|supported|needs-bootloader|registered
  Vendor:               8Bitdo
  VendorId:             USB:0x2DC8
  Version:              1.29
  Icon:                 input-gaming
  Created:              2019-02-18
```

OS: Ubuntu 18.04.2 LTS
Package source: Initially via APT (fwupd 1.0.9-0ubuntu2), then uninstalled that version in favor of the newer snap package

hughsie, 2019-02-18:
I'm just on holiday, but have untagged the firmware from stable. I'll find out what's wrong when I get back. So far there are 2 success and 3 failure reports for the firmware, so it's not quite as clear cut as "wrong GUID" I think.

mainmachine, 2019-02-18:
Appreciate the prompt response! It's possible those reports are all from me, as it looks like the firmware was uploaded very recently (the 15th), and I've run the upgrade a couple of times to confirm the problem. 

The upgrade reports as a success, the problem comes about when the controller is restarted - it just starts vibrating at full intensity and is undetectable! It's lucky the boot-loader mode is unaffected, as otherwise I'd expect this would have been an RMA situation.

hughsie, 2019-02-27:
So, I've looked at this a bit today. It's more complicated that it looks! "N30 Pro 2" is indeed a different model to "N30 Pro", the latter also being called a "SN30 Pro" at some stage. The former has "N30 Pro" silk screened on the front, so it's really confusing.

@superm1 -- using your "N30 Pro 2", can you get me the USB VID:PIDs of the hardware in each mode (e.g. xinput, android, mac etc) and either you or me can update the metainfo file, and resubmit the update on the LVFS. Thanks!

superm1, 2019-02-27:
Start +Y (default - switch mode):
```
Bus 001 Device 007: ID 057e:2009 Nintendo Co., Ltd 
```

Start + X (X input mode)
```
Bus 001 Device 011: ID 045e:028e Microsoft Corp. Xbox360 Controller
```

Start +B (Dinput mode)
```
Bus 001 Device 014: ID 2dc8:6000  
```

Start +A (Mac mode)
```
Bus 001 Device 015: ID 054c:05c4 Sony Corp. DualShock 4
```

superm1, 2019-02-27:
But mine is an "SF30 pro" according to the back of the controller

superm1, 2019-02-27:
And in Start +B (Dinput mode)

```
8Bitdo SF30 Pro
  DeviceId:             672c087de09848d9e7ee32aa1dea2fbeb8b81e6b
  Guid:                 043c7673-bd45-5e2e-b8b9-c22df2dfec2a
  Guid:                 b02f1319-c1c4-54f4-b7f2-5bb3552f6a9e
  Guid:                 e95bfb47-11f9-5afd-856e-fbc45730456d <- USB\VID_2DC8&PID_6000&REV_0001
  Guid:                 269b3121-097b-50d8-b9ba-d1f64f9cd241 <- USB\VID_2DC8&PID_6000
  Guid:                 5bc60e62-f461-58bd-91b6-7de3392f8bd6 <- USB\VID_2DC8
  Guid:                 afa5518d-7c55-51cb-8e1c-bf3b11b67dfa <- USB\CLASS_03&SUBCLASS_00&PROT_00
  Guid:                 cee9c5bc-aa12-51eb-a220-d0ef588d7cb2 <- USB\CLASS_03&SUBCLASS_00
  Guid:                 571229d5-caed-5cbc-91c1-5c216601a194 <- USB\CLASS_03
```

superm1, 2019-02-27:
I was able to update from `1.26` to `1.29` on it just now using master.
Of course there is still the problem that it doesn't "come back" after update that still needs to be sorted out.  That's because the controller boots up in switch mode after the update and you need to be in Dinput mode for using fwupd.

hughsie, 2019-02-28:
@superm1 so I think you do have a `SF30 pro` rather than a `N30 Pro 2` -- I need to find somebody with the latter.

superm1, 2019-02-28:
Yep that's what I thought it was I had too.