fwupd/fwupd #1037: Lenovo X1C5 update 1.36 is not applied

klingtnet, 2019-02-18:
As the title says, the `1.36` update is not applied, instead only a message is printed for a glimpse of a second after reboot then the machine restart's itself.

```shell
$ fwupdmgr --version
client version:	1.2.4
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.4
```

```shell
$ fwupdmgr get-devices
ThinkPad X1 Carbon Thunderbolt Controller
  DeviceId:             0489035fa9e9cd1c7b4b19b221e52c859ffffd62
  Guid:                 89d9d1e6-9e4c-5f07-b5c1-603da3d61835
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              15.00
  Icon:                 computer
  Created:              2019-02-18

20HRCTO1WW System Firmware
  DeviceId:             1cbe298fc17877b6883e85560778ac0812f0a385
  Guid:                 798ffd60-f10e-4ac4-8939-c8beabfe55b4
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 8f58d6d2-9753-5def-8169-1ef0971b1edc
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.35
  VersionLowest:        0.1.14
  Icon:                 computer
  Created:              2019-02-18
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             6461040534f2259a0439361986adf73979fd836e
  Guid:                 c35736d2-9e47-4578-93e9-68d5b04ea77e
  Guid:                 66ee5bec-e907-5049-a11d-ae78a18e01d9
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.55.3510
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-02-18

UEFI Device Firmware
  DeviceId:             6b3fcb33b92ea5cdd94954b276ad0b63ed14f9bb
  Guid:                 74997a6b-1adf-4b12-b994-401f06ea8c72
  Guid:                 1fe412bc-cb71-5d49-8812-8198a006fa58
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.19
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-02-18

LENSE20512GMSP34MEAT2TA
  DeviceId:             310f45f1f223064b5c16bf6dff31146755a64480
  Guid:                 b0cd8903-04e1-51f7-a69a-f08b77e4a6ad
  Guid:                 0696debf-90a2-5c62-8c96-10857c206b91
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Lenovo
  VendorId:             NVME:0x17AA
  Version:              2.7.8341
  Icon:                 drive-harddisk
  Created:              2019-02-18
```

```shell
$ efibootmgr -v
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0017,0018,0019,001A,001B,001C,001D,001E,0000
Boot0000* Linux-Firmware-Updater	HD(1,GPT,a3f54b20-ecb8-41af-b22d-2f266505a08f,0x800,0x200000)/File(\EFI\arch\fwupdx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001E  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot001F* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot0020* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot0021* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0022* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
```

```shell
$ efivar -l | grep fw
```

```shell
$ tree /boot
/boot
├── EFI
│   ├── arch
│   │   ├── fw
│   │   └── fwupdx64.efi
│   ├── boot
│   │   └── bootx64.efi
│   └── systemd
│       └── systemd-bootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── intel-ucode.img
├── loader
│   ├── entries
│   │   ├── arch.conf
│   │   ├── arch-defaults.conf
│   │   └── arch-lts.conf
│   └── loader.conf
├── vmlinuz-linux
└── vmlinuz-linux-lts

7 directories, 14 files
```

Please answer the following questions:

- Operating system and version: Linux m604 4.19.23-1-lts #1 SMP Fri Feb 15 11:18:29 CET 2019 x86_64 GNU/Linux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): `pacman -S fwupd`
- Have you tried rebooting? Yes.
- Are you using an NVMe disk? Yes.
- Is secure boot enabled (only for the UEFI plugin)? No.

superm1, 2019-02-18:
Duplicate of https://github.com/hughsie/fwupd/issues/1035 maybe

klingtnet, 2019-02-19:
[image]

It's pretty unreadable but the best I could get.

klingtnet, 2019-02-19:
> Duplicate of #1035 maybe

Using the latest version on master did not fix the problem.

superm1, 2019-02-19:
From master:
1. Would you please use `/usr/libexec/fwupd/fwupdate` to turn on the verbose debugging for UEFI?  It's one of the options in that tool.
2. Re-run the update process with verbose debugging enabled.
3. Come back to Linux OS and use that tool to view the logs captured from the UEFI run (it's another one of the options).

superm1, 2019-02-19:
Looking at your potato picture it seems like it might be this: #989.  In any case, verbose logging from master may be able to help identify what's going on.

ilinas, 2019-02-22:
I think I got the same message on Lenovo T480s:

```
Found update fwupd-ebfe8df8-dee7-4692-a721-cbcf5095c5cf-0
Adding new capsule
WARNING: Could not update variable status for '': Not Found
WARNING: Could not set update status: Not Found
start_image() returned Not Found
```

System details:
```
# fwupdmgr get-updates 
20L8002WMD System Firmware has firmware updates:
GUID:                    ebfe8df8-dee7-4692-a721-cbcf5095c5cf
GUID:                    230c8b18-8d9b-53ec-838b-6cfc0383493a
GUID:                    2b79e0a0-0801-5fc9-a7a5-90374e924a9b
ID:                      com.lenovo.ThinkPadN22ET.firmware
Update Version:          0.1.29
Update Name:             ThinkPad T480s
Update Summary:          Lenovo ThinkPad T480s System Firmware
Update Remote ID:        lvfs
Update Checksum:         SHA1(646d495a92291b4832225183ace8b88dd8e9ae88)
Update Location:         https://fwupd.org/downloads/7d2e5ce0403a04d9cdb73e5607f94d6b51e4f1e7-Lenovo-ThinkPad-T480s-SystemFirmware-1.29.cab
Update Description:      Lenovo ThinkPad T480s System Firmware Version 1.29
                         
                         This update also adds the following features:
                         
No upgrades for UEFI Device Firmware, current is 184.60.3561: 184.55.3510=older
```

```
# fwupdmgr --version
client version:	1.2.4
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.4
```

```
# fwupdmgr get-devices
Thinkpad T480s Thunderbolt Controller
  DeviceId:             78e65184f7a503a3a9a8ba66121e7dd358569b89
  Guid:                 061f2847-061b-5cfb-9c39-3dbfe2b49d69
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  Icon:                 computer
  Created:              2019-02-22

ThinkPad Thunderbolt 3 Dock
  DeviceId:             400be60c67e78ca6ae3936461232977d1d39bb6f
  Guid:                 df675f10-53ac-59ca-bddd-0a86ee492920
  Plugin:               thunderbolt
  Flags:                updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0108
  Version:              15.00
  Icon:                 audio-card
  Created:              2019-02-22

20L8002WMD System Firmware
  DeviceId:             e7cc234e82d08b6a14e246beb6b5f8142091e570
  Guid:                 ebfe8df8-dee7-4692-a721-cbcf5095c5cf
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 2b79e0a0-0801-5fc9-a7a5-90374e924a9b
[... 59 lines cut ...]
Boot0005  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0006  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0007  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0008  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot0009  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot000A* USBR BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot000B* USBR BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot000C* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot000D* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot000E* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot000F  PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0010* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot0011* debian	HD(1,GPT,505b0439-fbee-4d59-a77e-4fef4d73745d,0x800,0x82000)/File(\EFI\debian\shimx64.efi)
Boot0012* Linux-Firmware-Updater	HD(1,GPT,505b0439-fbee-4d59-a77e-4fef4d73745d,0x800,0x82000)/File(\EFI\debian\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0024  USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0025  USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0026* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot0027  ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot0028* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
```

```
# efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-ebfe8df8-dee7-4692-a721-cbcf5095c5cf-0
```

superm1, 2019-02-22:
Can you please retry with master and turn on verbose debugging?

klingtnet, 2019-02-25:
Can confirm that the update works with this version:

```sh
$ fwupdmgr --version
client version:	1.2.6
checkout info:	1.2.5-4-g36ed8a06
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.6
```

superm1, 2019-02-25:
OK thanks!
So it's fixed in 1.2.5 release.

closed 2019-02-25 by superm1

marvin-ad-martianum, 2020-01-08:
I have this error again. Anyone has an idea? 

I am running an X1 Yoga, Lenovo 1st gen and after running
```
sudo fwupdmgr update 
Upgrade available for System Firmware from 0.1.15 to 0.1.42
20FRS2M600 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 0.1.42 for System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating System Firmware…[  -                                    ]
Scheduling…              [***************************************]
Successfully installed firmware
Upgrade available for UEFI Device Firmware from 176.12.1001 to 184.70.3626
20FRS2M600 must remain plugged into a power source for the duration of the update to avoid damage. Continue with update? [Y|n]: y
Downloading 184.70.3626 for UEFI Device Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating UEFI Device Firmware…   \                               ]
Scheduling…              [***************************************]
Successfully installed firmware

An update requires a reboot to complete. Restart now? [y|N]: 
```

```
$ fwupdmgr --version
client version:	1.3.6
compile-time dependency versions
	gusb:	0.2.11
	efivar:	34
daemon version:	1.3.6
```

When running this it does not show the fwupdx64.efi, however when i run it manualy from the efi boot menu it gives me the above error
 "Could not apply capsule update: Not Found" 

```
$ efibootmgr -v
BootNext: 0001
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0000,0010,0011,0012,0013,001B,0017,0018,0019,001A,001C,0021,0001
Boot0000* ubuntu	HD(4,MBR,0x26364,0x1b2c8000,0x64000)/File(\EFI\ubuntu\shimx64.efi)
Boot0001* Linux-Firmware-Updater	HD(4,MBR,0x26364,0x1b2c8000,0x64000)/File(\EFI\ubuntu\shimx64.efi)\.f.w.u.p.d.x.6.4...e.f.i...
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001B* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001C* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001D* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,1,0)
Boot001E* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x16,0x2)/Ata(0,0,0)
Boot001F* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0020* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0021* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```