fwupd/fwupd #1042: Lenovo T470 firmware upgrade does not get applied by UEFI BIOS

agners, 2019-02-25:
Trying to apply the latest Lenovo UEFI BIOS update seems to succeed:

```
$ fwupdmgr update
Downloading 0.1.55 for 20HDCTO1WW System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating 20HDCTO1WW System Firmware…                             ]
Scheduling…              [***************************************]
Downloading 184.55.3510 for UEFI Device Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating UEFI Device Firmware…\                                  ]
Scheduling…              [***************************************]
No upgrades for Unifying Receiver, current is RQR12.07_B0029: RQR12.07_B0029=same

An update requires a reboot to complete. Restart now? [Y|n]:
```

However, on reboot, after the Lenovo Logo just before the boot loader this text appears very quickly:

```
Found update fwupd-4e06f4ae-ab2e-4804-9a23-888e596a31e7-0
Found update fwupd-bcff5fd8-0a99-48ad-b630-40e8eadb190e-0
Adding new capsule
Adding new capsule
WARNING: Could not update variable status for '': Not found
WARNING: Could not set update status: Not found
```

Also tried using the `efi_no_storage_paranoia` kernel parameter. This happens with Linux 4.20.12 as well as 4.19.25.

```shell
$ fwupdmgr --version
client version:	1.2.4
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.4
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
T470 Thunderbolt Controller
  DeviceId:             8ba0cfbe33abdaa7ad0fce23b4c3eddf4c319dbf
  Guid:                 6c2c78b7-81fa-566e-b0ff-e727dbb297a5
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               LENOVO
  VendorId:             TBT:0x0109
  Version:              04.00
  Icon:                 computer
  Created:              2019-02-25

20HDCTO1WW System Firmware
  DeviceId:             888f97106c5be770592b21740d18c1d29aa8e1b9
  Guid:                 4e06f4ae-ab2e-4804-9a23-888e596a31e7
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Guid:                 872eede9-59ef-5f7e-b8ef-baa0e4375c54
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.53
  VersionLowest:        0.1.23
  Icon:                 computer
  Created:              2019-02-25
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             4718f2fc055b1c4de66633017326683a72592b4c
  Guid:                 bcff5fd8-0a99-48ad-b630-40e8eadb190e
  Guid:                 44dc069e-84f6-52ec-8bc6-f50c9899d667
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              182.10.1196
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-02-25
  UpdateState:          success

UEFI Device Firmware
  DeviceId:             efc88c8524a5d0ada629ba38b89ac722dc3d5efb
  Guid:                 4b37c365-ecb2-4d89-8e15-e93c27c6ce12
  Guid:                 897f3781-238c-56a0-95fa-e2612102db0e
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.29
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-02-25

Samsung SSD 960 EVO 1TB
  DeviceId:             c82980cd3fe66f7d0a8bb83b353cbc271177efd4
  Guid:                 5b3df2da-f745-5fd0-81de-5dafd7f0bf8c
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              1B7QCXE7
  Icon:                 drive-harddisk
  Created:              2019-02-25

Unifying Receiver
  DeviceId:             a04d2149eebdb3eda67a37a2575ec263a8389cef
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.07_B0029
  VersionBootloader:    BOT01.02_B0015
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-02-25
```

```shell
$ sudo efibootmgr -v
BootCurrent: 0019
Timeout: 0 seconds
BootOrder: 0017,0018,0019,001A,001B,001C,001D,001E,001F,0000
Boot0000* Linux-Firmware-Updater	HD(1,GPT,c2754abe-515e-400c-9752-74a4b3ed7e8c,0x800,0x64000)/File(\EFI\arch\fwupdx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001B* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001C* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0024* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
```

```shell
$ efivar -l | grep fw
```

```shell
$ tree /boot
/boot
├── EFI
│   ├── Boot
│   │   └── BOOTX64.EFI
│   ├── EFI
│   │   └── arch
│   │       ├── fw
│   │       └── fwupdx64.efi
│   ├── arch
│   │   ├── fw
│   │   │   ├── fwupd-3b8c8162-188c-46a4-aec9-be43f1d65697.cap
│   │   │   ├── fwupd-4e06f4ae-ab2e-4804-9a23-888e596a31e7.cap
│   │   │   └── fwupd-bcff5fd8-0a99-48ad-b630-40e8eadb190e.cap
│   │   └── fwupdx64.efi
│   └── gummiboot
│       └── gummibootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── initramfs-linux.img
├── loader
│   ├── entries
│   │   ├── arch-lts.conf
│   │   └── arch.conf
│   └── loader.conf
├── syslinux
│   └── syslinux.cfg
├── vmlinuz-linux
└── vmlinuz-linux-lts

<after reboot>
$ tree /boot
/boot
├── EFI
│   ├── Boot
│   │   └── BOOTX64.EFI
│   ├── EFI
│   │   └── arch
│   │       ├── fw
│   │       └── fwupdx64.efi
│   ├── arch
│   │   ├── fw
│   │   └── fwupdx64.efi
│   └── gummiboot
│       └── gummibootx64.efi
├── initramfs-linux-fallback.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── initramfs-linux.img
├── loader
│   ├── entries
│   │   ├── arch-lts.conf
│   │   └── arch.conf
│   └── loader.conf
├── syslinux
│   └── syslinux.cfg
├── vmlinuz-linux
└── vmlinuz-linux-lts
```

Please answer the following questions:

- Operating system and version: Arch Linux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): pacman
- Have you tried rebooting? Yes, multiple times
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

hughsie, 2019-02-25:
Can you try with the recently released 1.2.5 please? That's got a fix for the issue I think you're hitting.

superm1, 2019-02-25:
It was confirmed in #1037 that 1.2.5 fixed it, so I'm thinking probably yes.

agners, 2019-02-26:
@hughsie @superm1 tonight ArchLinux updated to 1.2.5, and indeed, this morning the UEFI BIOS update worked just fine! Thx!

closed 2019-02-26 by agners