fwupd/fwupd #1054: can't run fwupd on Fedora Silverblue

andreasn, 2019-03-01:
Every time I launch GNOME Software I get this error:
[image]

If I run `/usr/libexec/fwupd/fwupd --immediate-exit --verbose` from a terminal. I get:
`Failed to load engine: Failed to load AppStream data: Error opening file “/var/cache/fwupd/metadata.xmlb”: Permission denied`

```shell
$ fwupdmgr --version
client version:	1.2.3
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.3
```

* Fedora Silverblue version 29.20190226.0 (2019-02-26T00:57:33Z)
* fwupd was installed by default in the base image
* rebooting didn't take effect
* I have no idea what a NVMe disk is
* secure boot is enabled

andreasn, 2019-03-01:
Just updated Fedora Silverblue to version: 29.20190301.0 that includes fwupd 1.2.5 and the issue persist.

hughsie, 2019-03-01:
> If I run `/usr/libexec/fwupd/fwupd --immediate-exit --verbose` from a terminal

Was that as root? Also, does it work with selinux turned off my any chance?

andreasn, 2019-03-01:
That was as myself (@kalev sent me the string, and maybe he expected me to run as root). If I run it with `sudo` I don't get the error.

andreasn, 2019-03-01:
And yes, if I set `setenforce 0` and run the command without `sudo` it seems to work. So seems like an selinux error indeed.

hughsie, 2019-03-01:
I think you're hitting https://github.com/hughsie/fwupd/commit/b768e4d9248cf206139e17363bc4ba513d54fcf3 -- kinda crappy the command just hangs for a selinux failure tho.

superm1, 2019-03-01:
Trying to get down to the bottom of that - I guess SELinux policy for `tpm2-tools` needs adjusting in Fedora Silverblue.  Reading TPM PCR values is something I would expect to be able to do SELinux or not.
Can you file a bug there so that can at least get fixed?

superm1, 2019-03-01:
And to confirm the hypothesis above you can try to manually run `tpm2_pcrlist` and probably see it hang too.

superm1, 2019-03-01:
Oh actually looking at https://bugzilla.redhat.com/show_bug.cgi?id=1665701 there was a proposed fix to tpm2-abrmd and a confirmation that fixed it too.

superm1, 2019-03-01:
So I'll close this for fwupd, track the issue in Fedora there.

closed 2019-03-01 by superm1