fwupd/fwupd #1058: UEFI upgrade might cause loosing bootloader

Porkepix, 2019-03-01:
Today I just used fwupd to upgrade the UEFI of my ThinkPad T470. There was a warning on ArchLinux's wiki about bootloader that might disappear so I was only half-surprised and it can be fixed later on. However I thought it was worth it to report the issue in case bootloader could be preserved, which would lead to way smoother upgrades.

```shell
[clement@T470-clement ~]$ fwupdmgr --version
client version:	1.2.5
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.5
```

```shell
[clement@T470-clement ~]$ fwupdmgr get-devices
T470 Thunderbolt Controller
  DeviceId:             foo
  Guid:                 bar
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               LENOVO
  VendorId:             TBT:0x0109
  Version:              04.00
  Icon:                 computer
  Created:              2019-02-28

20JNS3FG00 System Firmware
  DeviceId:             foo
  Guid:                 bar
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.55
  VersionLowest:        0.1.23
  Icon:                 computer
  Created:              2019-02-28

UEFI Device Firmware
  DeviceId:             foo
  Guid:                 bar
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.55.3510
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-02-28

UEFI Device Firmware
  DeviceId:             foo
  Guid:                 bar
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.33
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-02-28

SAMSUNG MZVLW256HEHP-000L7
  DeviceId:             foo
  Guid:                 bar
  Guid:                 bar
  Guid:                 bar
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              4L7QCXB7
  Icon:                 drive-harddisk
  Created:              2019-02-28
```
I just got rid of Ids, probably meaningless here, and maybe not the best to throw serial numbers out on the Internet.

```shell
[clement@T470-clement ~]$ efibootmgr -v
BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0017,0018,0019,001A,001B,001C,001D,001E,001F
Boot0000* ArchLinux	HD(1,GPT,a0195634-db7a-4175-91b6-4a4f05959e7a,0x800,0x100000)/File(\EFI\ArchLinux\grubx64.efi)
Boot0010  Setup	FvFile(721c8b66-426c-4e86-8e99-3457c46ab0b9)
Boot0011  Boot Menu	FvFile(126a762d-5758-4fca-8531-201a7f57f850)
Boot0012  Diagnostic Splash Screen	FvFile(a7d8d9a6-6ab0-4aeb-ad9d-163e59a7a380)
Boot0013  Lenovo Diagnostics	FvFile(3f7e615b-0d45-4f80-88dc-26b234958560)
Boot0014  Startup Interrupt Menu	FvFile(f46ee6f4-4785-43a3-923d-7f786c3c8479)
Boot0015  Rescue and Recovery	FvFile(665d3f60-ad3e-4cad-8e26-db46eee9f1b5)
Boot0016  MEBx Hot Key	FvFile(ac6fd56a-3d41-4efd-a1b9-870293811a28)
Boot0017* USB CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,86701296aa5a7848b66cd49dd3ba6a55)
Boot0018* USB FDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,6ff015a28830b543a8b8641009461e49)
Boot0019* NVMe0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,001c199932d94c4eae9aa0b6e98eb8a400)
Boot001A* ATA HDD1	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f601)
Boot001B* ATA HDD0	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f600)
Boot001C* USB HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,33e821aaaf33bc4789bd419f88c50803)
Boot001D* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)
Boot001E  Other CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a35406)
Boot001F  Other HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f606)
Boot0020* IDER BOOT CDROM	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,1)
Boot0021* IDER BOOT Floppy	PciRoot(0x0)/Pci(0x14,0x0)/USB(11,0)
Boot0022* ATA HDD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,91af625956449f41a7b91f4f892ab0f6)
Boot0023* ATAPI CD	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,aea2090adfde214e8b3a5e471856a354)
Boot0024* PCI LAN	VenMsg(bc7838d2-0f82-4d60-8316-c068ee79d25b,78a84aaf2b2afc4ea79cf5cc8f3d3803)

```

```shell
$ efivar -l | grep fw
# Empty result, no match. grep -i only find a result related to the TPM.
**This is only required if you use the UEFI plugin**
```

```shell
[clement@T470-clement ~]$ sudo exa -T /boot
/boot
├── efi
│  └── EFI
│     ├── arch
│     │  ├── fw
│     │  └── fwupdx64.efi
│     └── ArchLinux
│        └── grubx64.efi
```
I removed `/boot/grub` part of the tree which was very long and didn't looked like what was needed here. I can readd it later on if needed.

Please answer the following questions:

- Operating system and version: ArchLinux. Rolling release, up-to-date with stable version of softwares
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): yay (pacman wrapper)
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? No

superm1, 2019-03-01:
Sounds like maybe a duplicate of #820

Porkepix, 2019-03-01:
Well, no idea as titel of #820 suggests UEFI upgrade failed. Mine did works, it just had the nasty side-effect of removing my bootloader.