fwupd/fwupd #1060: Wrong summary strings reported by "fwupdmgr get-releases"

aleksander0m, 2019-03-03:
Testing with fwupd at commit 7120667ccd2d1341e15a53cc16edab286846f0aa (March 1st 2019). I have a CAB file uploaded to fwupd that contains multiple metainfo.xml files. Each metainfo.xml file reports a different version number and is associated to a different GUID, and so each metainfo.xml file also has a different Summary text. Worth noting that all metainfo.xml files reference **the same firmware.zip** file inside the CAB.

When I run "fwupdmgr get-releases", I get a list of updates and every item in the list has exactly the same "Summary", even if the actual metainfo.xml associated to each of the items had a different one. In the list below, all items in the list reference the "FirstNet carrier", but that Summary only applies to the item with version "T77W968.F1.0.0.3.7.A2.018". All the other items have a different Summary.

Even the older release with version T77W968.F1.0.0.3.6 also references the "FirstNet carrier", which is also not expected, as the Summary in the metainfo.xml file associated to that release (a totally different CAB) didn't reference any carrier at all.

```
Version:                 T77W968.F1.0.0.3.7.DO.009
Name:                    DW5821e
Summary:                 Firmware for the Dell DW5821e LTE modem (FirstNet carrier)
Remote:                  aleksander_es-embargo
URI:                     https://fwupd.org/downloads/FILENAME.cab
Description:             Initial public release in the LVFS.
                         
Checksum:                SHA1(0fa6c4e6d118687cc4e640c8011ca2802e89bc87)

Version:                 T77W968.F1.0.0.3.7.AT.018
Name:                    DW5821e
Summary:                 Firmware for the Dell DW5821e LTE modem (FirstNet carrier)
Remote:                  aleksander_es-embargo
URI:                     https://fwupd.org/downloads/FILENAME.cab
Description:             Initial public release in the LVFS.
                         
Checksum:                SHA1(0fa6c4e6d118687cc4e640c8011ca2802e89bc87)

Version:                 T77W968.F1.0.0.3.7.A2.018
Name:                    DW5821e
Summary:                 Firmware for the Dell DW5821e LTE modem (FirstNet carrier)
Remote:                  aleksander_es-embargo
URI:                     https://fwupd.org/downloads/FILENAME.cab
Description:             Initial public release in the LVFS.
                         
Checksum:                SHA1(0fa6c4e6d118687cc4e640c8011ca2802e89bc87)

Version:                 T77W968.F1.0.0.3.6
Name:                    DW5821e
Summary:                 Firmware for the Dell DW5821e LTE modem (FirstNet carrier)
Remote:                  aleksander_es-embargo
URI:                     https://fwupd.org/downloads/FILENAME2.cab
Description:             Initial public firmware release in LVFS
                         
Checksum:                SHA1(ff4ec33058d08b95f90d7a1e64a71a0353adf807)
```

aleksander0m, 2019-03-03:
I see that the fwupd remote XML file "groups" all the releases under the same component "id":

```
<?xml version='1.0' encoding='UTF-8'?>
<components origin="lvfs" version="0.9">
  <component type="firmware">
    <id>com.dell.dw5821e.firmware</id>
    <name>DW5821e</name>
    <summary>Firmware for the Dell DW5821e LTE modem (FirstNet carrier)</summary>
<releases>
      <release timestamp="1551052800" urgency="high" version="T77W968.F1.0.0.3.7.A2.018">
        ...
      </release>
      <release timestamp="1551052800" urgency="high" version="T77W968.F1.0.0.3.7.AT.018">
        ...
      </release>
      <release timestamp="1551052800" urgency="high" version="T77W968.F1.0.0.3.7.DO.009">
        ...
```
I should probably provide different component ids for the different per-carrier metainfo.xml files? E.g. "com.dell.dw5821e.a2", "com.dell.dw5821e.at", "com.dell.dw5821e.do" and so on?

superm1, 2019-03-03:
They should definitely be different component IDs..

aleksander0m, 2019-03-04:
Yep, the issue is gone when using different component ids. Sorry for the noise :)

closed 2019-03-04 by aleksander0m