fwupd/fwupd #1061: Wrong version reported as downloading with "fwupdmgr update"

aleksander0m, 2019-03-03:
Testing with fwupd at commit 7120667 (March 1st 2019). I have a CAB file uploaded to fwupd that contains multiple metainfo.xml files. Each metainfo.xml file reports a different version number and is associated to a different GUID. Worth noting that all metainfo.xml files reference the same firmware.zip file inside the CAB.

The CAB file in LVFS contains one specific metainfo.xml associated to the "generic" carrier configuration with GUID "USB\VID_413C&PID_81D7&REV_0318&CARRIER_GCF", and the firmware version reported in that metainfo.xml file is "T77W968.F1.0.0.3.7.GC.009".

The same CAB file also contains additional metainfo.xml files associated to other GUIDs, each with a different reported firmware version. E.g. one of the additional metainfo.xml files specifies version "T77W968.F1.0.0.3.7.VZ.010" associated to GUID "USB\VID_413C&PID_81D7&REV_0318&CARRIER_VERIZON".

The device is reporting the following information in "fwupdmgr get-devices":
```
DW5821e Snapdragon X20 LTE
  DeviceId:             d9f8aff596f49b32495ddda6a2c4d92f90fd6b71
  Guid:                 c1d92c58-db79-5ad9-9eef-0ca8e5e49562 <- USB\VID_413C&PID_81D7&REV_0318&CARRIER_GCF
  Guid:                 64da2d58-8d1b-5e5b-b793-f88ba5a25a8f <- USB\VID_413C&PID_81D7&REV_0318
  Guid:                 761d6124-0002-5185-b767-9adf67bf1a5e <- USB\VID_413C&PID_81D7
  Guid:                 795e079d-093b-5503-aa59-35b832480e95 <- USB\VID_413C
  Summary:              Mobile broadband device
  Plugin:               modem_manager
  Flags:                updatable|registered|use-runtime-version
  Vendor:               Dell Inc.
  Version:              T77W968.F1.0.0.3.7.GC.004
  Icon:                 network-modem
  Created:              2019-03-03
```
Looking at the GUIDs above, fwupd should report that for GUID "USB\VID_413C&PID_81D7&REV_0318&CARRIER_GCF" there is an update available (from T77W968.F1.0.0.3.7.GC.004 currently in the device to T77W968.F1.0.0.3.7.GC.009 available in LVFS).

But, when "fwupdmgr update" is run, the operation says it's downloading "T77W968.F1.0.0.3.7.VZ.010" instead of "T77W968.F1.0.0.3.7.GC.009".
```
$ sudo fwupdmgr update
Downloading T77W968.F1.0.0.3.7.VZ.010 for DW5821e Snapdragon X20 LTE...
Fetching firmware https://fwupd.org/downloads/FILENAME.cab
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating DW5821e Snapdragon X20 LTE…
                         [***************************************]
Updating DW5821e Snapdragon X20 LTE…
                         [***************************************]
Restarting device…       [***************************************]
```

In this specific case it's not an issue because both updates are 100% the same one, as the metainfo.xml files all reference the same file inside the CAB, but the output of the command is misleading.

The "get-releases" output shows how "T77W968.F1.0.0.3.7.VZ.010" is the version of the first item reported in the CAB, and all items have the same SHA1:

```
$ sudo fwupdmgr get-releases
DW5821e Snapdragon X20 LTE:
Version:                 T77W968.F1.0.0.3.7.VZ.010
Name:                    DW5821e
Summary:                 Firmware for the Dell DW5821e LTE modem (FirstNet carrier)
Remote:                  aleksander_es-embargo
URI:                     https://fwupd.org/downloads/FILENAME.cab
Description:             Initial public release in the LVFS.
                         
Checksum:                SHA1(0fa6c4e6d118687cc4e640c8011ca2802e89bc87)

Version:                 T77W968.F1.0.0.3.7.VF.007
Name:                    DW5821e
Summary:                 Firmware for the Dell DW5821e LTE modem (FirstNet carrier)
Remote:                  aleksander_es-embargo
URI:                     https://fwupd.org/downloads/FILENAME.cab
Description:             Initial public release in the LVFS.
                         
Checksum:                SHA1(0fa6c4e6d118687cc4e640c8011ca2802e89bc87)

....

Version:                 T77W968.F1.0.0.3.7.GC.009
Name:                    DW5821e
Summary:                 Firmware for the Dell DW5821e LTE modem (FirstNet carrier)
Remote:                  aleksander_es-embargo
URI:                     https://fwupd.org/downloads/FILENAME.cab
Description:             Initial public release in the LVFS.
                         
Checksum:                SHA1(0fa6c4e6d118687cc4e640c8011ca2802e89bc87)
```

superm1, 2019-03-04:
Also like #1060, check if this goes away once you have them all with their own components so they don't get mistakenly associated to the wrong device.

aleksander0m, 2019-03-04:
Yep, the issue is gone when using different component ids. Sorry for the noise :)

closed 2019-03-04 by aleksander0m