fwupd/fwupd #1062: Wrong list of updates for a given GUID

aleksander0m, 2019-03-03:
Testing with fwupd at commit 7120667 (March 1st 2019). I have a CAB file uploaded to fwupd that contains multiple metainfo.xml files. Each metainfo.xml file reports a different version number and is associated to a different GUID. Worth noting that all metainfo.xml files reference the same firmware.zip file inside the CAB.

I have a device exposing a Telefonica-specific GUID, as follows:

```
$ sudo fwupdmgr get-devices
DW5821e Snapdragon X20 LTE
  DeviceId:             d9f8aff596f49b32495ddda6a2c4d92f90fd6b71
  Guid:                 2d8ab76d-b20f-56c0-892d-a8450ec13311 <- USB\VID_413C&PID_81D7&REV_0318&CARRIER_TELEFONICA
  Guid:                 64da2d58-8d1b-5e5b-b793-f88ba5a25a8f <- USB\VID_413C&PID_81D7&REV_0318
  Guid:                 761d6124-0002-5185-b767-9adf67bf1a5e <- USB\VID_413C&PID_81D7
  Guid:                 795e079d-093b-5503-aa59-35b832480e95 <- USB\VID_413C
  Summary:              Mobile broadband device
  Plugin:               modem_manager
  Flags:                updatable|supported|registered|use-runtime-version
  Vendor:               Dell Inc.
  Version:              T77W968.T1.0.0.3.6.TF.005
  Icon:                 network-modem
  Created:              2019-03-03
```

The CAB file contains one metainfo.xml file associated to the Telefonica carrier information:
```
...
  <provides>
    <!-- USB\VID_413C&PID_81D7&REV_0318&CARRIER_TELEFONICA -->
    <firmware type="flashed">2d8ab76d-b20f-56c0-892d-a8450ec13311</firmware>
  </provides>
...
  <releases>
    <release urgency="high" version="T77W968.F1.0.0.3.7.TF.005" date="2019-02-25">
      <checksum target="content" filename="firmware.zip"/>
      <description>
        <p>Initial public release in the LVFS.</p>
      </description>
    </release>
  </releases>
...
```
The same CAB file contains other metainfo.xml files that are associated to other carrier-specific GUIDs and version numbers.

With this setup, if I list updates for the device, I get the full list of updates for all carriers, not just the one associated to the Telefonica-specific GUID exposed by the device:

```
$ sudo fwupdmgr get-updates
No upgrades for DW5821e Snapdragon X20 LTE, current is T77W968.T1.0.0.3.6.TF.005: T77W968.F1.0.0.3.7.A2.018=older, T77W968.F1.0.0.3.7.AT.018=older, T77W968.F1.0.0.3.7.DO.009=older, T77W968.F1.0.0.3.7.GC.009=older, T77W968.F1.0.0.3.7.KD.015=older, T77W968.F1.0.0.3.7.SB.009=older, T77W968.F1.0.0.3.7.SP.009=older, T77W968.F1.0.0.3.7.TE.014=older, T77W968.F1.0.0.3.7.TF.005=older, T77W968.F1.0.0.3.7.VF.007=older, T77W968.F1.0.0.3.7.VZ.010=older, T77W968.F1.0.0.3.6=older

```
(in this case it's showing 'older' versions but it's due to the T>F check when comparing versions, unrelated to this specific issue being reported here).

And if I try to downgrade (due to the older versions reported) I get presented with the full list of per-carrier versions. I would have expected to see only the TF.005 one plus the F1.0.0.3.6 one (a different CAB file, which wasn't associated to any carrier).

```
$ sudo fwupdmgr downgrade 
Choose a release:
1.	T77W968.F1.0.0.3.7.VZ.010 (<p>Initial public release in the LVFS.</p>)
2.	T77W968.F1.0.0.3.7.VF.007 (<p>Initial public release in the LVFS.</p>)
3.	T77W968.F1.0.0.3.7.TF.005 (<p>Initial public release in the LVFS.</p>)
4.	T77W968.F1.0.0.3.7.TE.014 (<p>Initial public release in the LVFS.</p>)
5.	T77W968.F1.0.0.3.7.SP.009 (<p>Initial public release in the LVFS.</p>)
6.	T77W968.F1.0.0.3.7.SB.009 (<p>Initial public release in the LVFS.</p>)
7.	T77W968.F1.0.0.3.7.KD.015 (<p>Initial public release in the LVFS.</p>)
8.	T77W968.F1.0.0.3.7.GC.009 (<p>Initial public release in the LVFS.</p>)
9.	T77W968.F1.0.0.3.7.DO.009 (<p>Initial public release in the LVFS.</p>)
10.	T77W968.F1.0.0.3.7.AT.018 (<p>Initial public release in the LVFS.</p>)
11.	T77W968.F1.0.0.3.7.A2.018 (<p>Initial public release in the LVFS.</p>)
12.	T77W968.F1.0.0.3.6 (<p>Initial public firmware release in LVFS</p>)
```

superm1, 2019-03-04:
I wonder if this is related to #1060.  Can you see if once you give them all their own IDs it helps?

aleksander0m, 2019-03-04:
Yep, the issue is gone when using different component ids. Sorry for the noise :)

closed 2019-03-04 by aleksander0m