fwupd/fwupd #1064: standalone installer failed to install fwupd.shutdown

aleksander0m, 2019-03-03:
Testing with fwupd at commit 7120667 (March 1st 2019), I have created a standalone installer for a CAB file using the snapcraft "edge" channel, which comes with fwupd 1.2.5-26-g7120667.

The creation of the standalone installer is done like this:
```
./contrib/standalone-installer/make.py \
	--snap-channel=edge \
	20190302-dw5821e-T77W968.F1.0.0.3.7.cab \
	20190302-dw5821e-T77W968.F1.0.0.3.7-standalone.py
```

And when I try to test the standalone in a different snap-capable system, I get the following error reported:
```
$ sudo python3 20190302-dw5821e-T77W968.F1.0.0.3.7-standalone.py --verbose install
['snap', 'list', 'fwupd']
['snap', 'ack', '/tmp/fwupd4zmr_eik/fwupd.assert']
['snap', 'install', '--classic', '/tmp/fwupd4zmr_eik/fwupd.snap']
error: cannot perform the following tasks:
- Run install hook of "fwupd" snap if present (run hook "install": install: cannot stat '/snap/fwupd/595//lib/systemd/system-shutdown/fwupd.shutdown': No such file or directory)
Snap installation failed
Snap installation failed
```

aleksander0m, 2019-03-03:
@superm1 looking at `fwupd/snap/hooks/install`, the new `install_if_missing` line for fwupd.shutdown uses the absolute path `/lib...` while all the other lines use relative paths (i.e. not starting with `/`). Maybe that's related? Totally guessing, I have no idea how the snap hooks are supposed to work.

superm1, 2019-03-04:
Hopefully that fixes it, but I'm not able to test it myself tonight.  Let me know if it doesn't and I'll dig more into it this week.

closed 2019-03-04 by superm1

aleksander0m, 2019-03-04:
@superm1 that fixed the issue indeed, thanks!

commit 68847950 (2019-03-04): trivial: snap: fix install location for shutdown script (Fixes: #1064)