fwupd/fwupd #1065: Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.03_B0009]

mattmole, 2019-03-03:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.2.5
checkout info:	1.2.5
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.5

```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
KINGSTON SV300S37A120G
  DeviceId:             602b0a6cc821d155208724f0e22f8d111542b74c
  Guid:                 c8b3f40d-d91e-5809-a1f0-803148660f7a
  Guid:                 8512c90c-53c6-575d-a637-72c2490f902d
  Guid:                 05ce8e85-28d3-53ee-b0c8-41c139e1d8a8
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              521ABBF0
  Icon:                 drive-harddisk
  Created:              2019-03-03

ST9500325AS
  DeviceId:             0a8c36d4c09c803cd6b5861e443fb7a41a20cbe6
  Guid:                 d66280ce-2ba2-5c6f-86f1-24bae3eee4a5
  Guid:                 627b82e7-a341-5c36-ada4-6ebc4be89501
  Guid:                 37cfcffa-5981-5741-99ed-c74126ff7e6f
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              D005DEM1
  Icon:                 drive-harddisk
  Created:              2019-03-03

Unifying Receiver
  DeviceId:             5ac84c9685923de6437a8804bf2b73f38fb5810e
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.07_B0030
  VersionBootloader:    BOT03.03_B0009
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-03-03

```

Please answer the following questions:

- Operating system and version: Ubuntu Mate 18.04
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): Default installed version and snap
- Have you tried rebooting? Yes
- Are you using an NVMe disk? No
- Is secure boot enabled (only for the UEFI plugin)? No

I received the error in the subject when attempting to update my logitech unifying receiver

lundmar, 2019-03-31:
I see the same error details for the Logitech Unifying receiver on a Ubuntu 18.10 system. Same fwupdmgr versions.

the-spyke, 2019-04-01:
I have the same issue on Ubuntu 18.10 with Logitech Unifying receiver:
```
$ fwupdmgr --version
client version:	1.1.2
daemon version:	1.1.2
compile-time dependency versions
	appstream-glib:	0.7.12
	gusb:	0.3.0
	efivar:	34
```

alxpl, 2019-05-21:
Same device as the one above, same problem using fwupd-1.2.8 on f30.

the-spyke, 2019-05-21:
The issue is still reproducible on `fwupd@1.2.9`.

eworm-de, 2019-07-08:
This is a duplicate of #997, no?

kennylevinsen, 2019-07-12:
See https://github.com/hughsie/fwupd/issues/1244.

closed 2019-07-15 by hughsie

tuxmaster5000, 2019-12-12:
I get an similar message, but I is an U0008 device:

> fwupdmgr get-updates
> No releases found for device: Not compatible with org.freedesktop.fwupd version 1.2.11, requires >= 1.3.1, Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.02_B0009]

superm1, 2019-12-12:
Please open a separate issue to track yours.