fwupd/fwupd #1075: flatpak CI failing

superm1, 2019-03-07:
I noticed that flatpak CI started failing at udev.
```
 /usr/bin/mkdir -p '/usr/lib/systemd/system'
 /usr/bin/install -c -m 644 init/udev-control.socket init/udev-kernel.socket init/udev.service init/udev-trigger.service init/udev-settle.service '/usr/lib/systemd/system'
/usr/bin/install: cannot create regular file '/usr/lib/systemd/system/udev-control.socket': Read-only file system
/usr/bin/install: cannot create regular file '/usr/lib/systemd/system/udev-kernel.socket': Read-only file system
/usr/bin/install: cannot create regular file '/usr/lib/systemd/system/udev.service': Read-only file system
/usr/bin/install: cannot create regular file '/usr/lib/systemd/system/udev-trigger.service': Read-only file system
/usr/bin/install: cannot create regular file '/usr/lib/systemd/system/udev-settle.service': Read-only file system
make[3]: *** [Makefile:2534: install-systemdsystemunitDATA] Error 1
make[2]: *** [Makefile:3109: install-am] Error 2
make[1]: *** [Makefile:2624: install-recursive] Error 1
make: *** [Makefile:3103: install] Error 2
Error: module udev: Child process exited with code 2
```

The notable difference between the two is:
**Fail:**

	systemdsystemunitdir:	/usr/lib/systemd/system

**Pass:**

	systemdsystemunitdir:

hughsie, 2019-03-07:
I can't reproduce this locally, even with a totally clean build environment. :/

superm1, 2019-03-07:
Try with docker locally to repro?

superm1, 2019-03-08:
It does repro locally in a docker container too for me.  It looks like it's pulling that information from the systemd pkg-config leading to the failure.  So a big hammer solution might be to remove that pkg-config file in the docker container.

superm1, 2019-03-08:
@hughsie in your local testing do you have a fully updated `org.gnome.Sdk`?  That's where the problem is.
I can see `/var/lib/flatpak/runtime/org.gnome.Sdk/x86_64/3.30/*/files/share/pkgconfig/systemd.pc` is new to it and causing the problem with this shared module.

superm1, 2019-03-08:
Yep and here's the bug for it: https://github.com/flathub/shared-modules/issues/43

hughsie, 2019-03-09:
I think that fixed things. I've modified the ModemManager args for the change and it seems to be working locally. I've kicked off some travis builds to find out.

hughsie, 2019-03-09:
Working!

closed 2019-03-09 by hughsie