fwupd/fwupd #1077: fwupd fails to run when building from source / plugin related?

der-eismann, 2019-03-07:
I wanted to try to bring fwupd to the Solus distribution so I was compiling `fwupd` (and `libxmlb`) from source. However when testing it I get lots of errors, probably related to the plugins. 

There seem to be two cases:
1. A few errors, then fail with "memory access error"
```
$ sudo /usr/lib64/fwupd/fwupd/fwupd
cannot register existing type 'FuPlugin'
cannot add private field to invalid (non-instantiatable) type '<invalid>'
g_once_init_leave: assertion 'result != 0' failed
fu_plugin_set_build_hash: assertion 'FU_IS_PLUGIN (self)' failed
Speicherzugriffsfehler
```
2. Lots of errors, but keeps running
```
$ sudo /usr/lib64/fwupd/fwupd/fwupd
cannot register existing type 'FuPlugin'
cannot add private field to invalid (non-instantiatable) type '<invalid>'
g_once_init_leave: assertion 'result != 0' failed
fu_plugin_set_build_hash: assertion 'FU_IS_PLUGIN (self)' failed
g_ptr_array_add: assertion 'rarray' failed
g_signal_emit_valist: assertion 'signal_id > 0' failed
g_ptr_array_add: assertion 'rarray' failed
g_signal_emit_valist: assertion 'signal_id > 0' failed
colorhug should call fu_plugin_set_build_hash()
cannot register existing type 'FuPlugin'
cannot add private field to invalid (non-instantiatable) type '<invalid>'
g_once_init_leave: assertion 'result != 0' failed
fu_plugin_set_build_hash: assertion 'FU_IS_PLUGIN (self)' failed
g_ptr_array_add: assertion 'rarray' failed
g_signal_emit_valist: assertion 'signal_id > 0' failed
nitrokey should call fu_plugin_set_build_hash()
cannot register existing type 'FuPlugin'
cannot add private field to invalid (non-instantiatable) type '<invalid>'
g_once_init_leave: assertion 'result != 0' failed
fu_plugin_set_build_hash: assertion 'FU_IS_PLUGIN (self)' failed
superio should call fu_plugin_set_build_hash()
cannot register existing type 'FuPlugin'
cannot add private field to invalid (non-instantiatable) type '<invalid>'
g_once_init_leave: assertion 'result != 0' failed
fu_plugin_set_build_hash: assertion 'FU_IS_PLUGIN (self)' failed
```

But wait, there's more!
```
$ sudo /usr/lib64/fwupd/fwupd/fwupd
cannot register existing type 'FuPlugin'
cannot add private field to invalid (non-instantiatable) type '<invalid>'
g_once_init_leave: assertion 'result != 0' failed
fu_plugin_set_build_hash: assertion 'FU_IS_PLUGIN (self)' failed
[1]    3677 segmentation fault  sudo /usr/lib64/fwupd/fwupd/fwupd
```

```
$ sudo /usr/lib64/fwupd/fwupd/fwupd
cannot register existing type 'FuPlugin'
cannot add private field to invalid (non-instantiatable) type '<invalid>'
g_once_init_leave: assertion 'result != 0' failed
fu_plugin_set_build_hash: assertion 'FU_IS_PLUGIN (self)' failed
realloc(): invalid old size
[1]    3604 abort      sudo /usr/lib64/fwupd/fwupd/fwupd
```

I have no idea what this means or why this happens... do you have any idea?

```shell
$ fwupdmgr --version
client version:	1.2.5
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: GDBus.Error:org.freedesktop.DBus.Error.TimedOut: Failed to activate service 'org.freedesktop.fwupd': timed out (service_start_timeout=25000ms)
```

Please answer the following questions:
- Operating system and version: Solus 3.9999
- How did you install fwupd: built it from source
- Have you tried rebooting? yes
- Are you using an NVMe disk? yes
- Is secure boot enabled (only for the UEFI plugin)? no

superm1, 2019-03-07:
I would guess you're compiling with `-Wl,-Bsymbolic-functions`.   Don't do that.
If that's not what's going on, can you please share your full build log?

superm1, 2019-03-07:
See https://github.com/hughsie/fwupd/commit/39ddd968b8e9d452fbf03db907f26274b952cee0 for some more context

der-eismann, 2019-03-07:
Thanks for the quick answer! And yes, that was the problem indeed. By default the Solus build environment exports
```
LDFLAGS='-Wl,--copy-dt-needed-entries -Wl,-O1 -Wl,-z,relro -Wl,-z,now -Wl,-z,max-page-size=0x1000 -Wl,-Bsymbolic-functions -Wl,--sort-common'
```
Without your flag it's working fine. Thanks again!

closed 2019-03-07 by der-eismann