fwupd/fwupd #1083: fwupdmgr system devices not showing updates Lenovo X1 Extreme

hmgibson23, 2019-03-09:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.2.5
compile-time dependency versions
        gusb:   0.3.0
        efivar: 35
daemon version: 1.2.5
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
ThinkPad X1E Thunderbolt Controller
  DeviceId:             719aee6a634c39170d1d71b9795c1998dd7c037f
  Guid:                 b52afd68-e052-543c-92d5-de6c75349bb2
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              23.00
  Icon:                 computer
  Created:              2019-03-09

INTEL SSDPEKKF256G8L
  DeviceId:             c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
  Guid:                 79517f86-8df8-5d6e-a18b-33f0b36a78e9
  Guid:                 68db11e5-b0cf-5bc9-a94e-17e28496e505
  Guid:                 3f235826-4614-56f6-8fd4-b67b346882c9
  Guid:                 cdc79848-b022-5a05-a77a-862f21df0c29
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Intel Corporation
  VendorId:             NVME:0x8086
  Version:              L08P
  Icon:                 drive-harddisk
  Created:              2019-03-09

```

Please answer the following questions:

- Operating system and version:
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): yay -Sy fwupd
- Have you tried rebooting? Yes
- Are you using an NVMe disk? No
- Is secure boot enabled (only for the UEFI plugin)? No

I'm tryinp to apply the 1.19 firmware to my machine and the system firmware is not showing up in the devices. I followed these steps: https://github.com/hughsie/fwupd/issues/856 and still nothing. 

Any help is appreciated! Thanks!

The output from 

```
ThinkPad X1E Thunderbolt Controller
  DeviceId:             719aee6a634c39170d1d71b9795c1998dd7c037f
  Guid:                 b52afd68-e052-543c-92d5-de6c75349bb2
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x00
```

hmgibson23, 2019-03-09:
To add to this, the System Firmware doesn't even show up with:
```
$ fwudmgr get-devices --show-all-devices
ThinkPad X1E Thunderbolt Controller
  DeviceId:             719aee6a634c39170d1d71b9795c1998dd7c037f
  Guid:                 b52afd68-e052-543c-92d5-de6c75349bb2
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              23.00
  Icon:                 computer
  Created:              2019-03-09

Intel AMT [unprovisioned]
  DeviceId:             e2623122c99d58220498aacbfcfdb1baebbae3c5
  Guid:                 2800f812-b7b4-2d4b-aca8-46e0ff65814c
  Summary:              Hardware and firmware technology for remote out-of-band management
  Plugin:               amt
  Flags:                internal|registered
  Vendor:               Intel Corporation
  Version:              12.0.6.1120
  VersionBootloader:    12.0.6.1120
  Icon:                 computer
  Created:              2019-03-09

INTEL SSDPEKKF256G8L
  DeviceId:             c6a0cfba7c7d81e253fce571e1d1e9f6003ae1c7
  Guid:                 79517f86-8df8-5d6e-a18b-33f0b36a78e9
  Guid:                 68db11e5-b0cf-5bc9-a94e-17e28496e505
  Guid:                 3f235826-4614-56f6-8fd4-b67b346882c9
  Guid:                 cdc79848-b022-5a05-a77a-862f21df0c29
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Intel Corporation
  VendorId:             NVME:0x8086
  Version:              L08P
  Icon:                 drive-harddisk
  Created:              2019-03-09

GP107M [GeForce GTX 1050 Ti Mobile]
  DeviceId:             71b677ca0f1bc2c5b804fa1d59e52064ce589293
  Guid:                 dc1e6866-2188-547f-b175-35e07dd7a6af
  Guid:                 04232c57-d805-551d-a1de-1bba9350d71f
  Guid:                 924bbf24-f21c-5c11-939e-e8c56e7c74af
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               NVIDIA Corporation
  VendorId:             PCI:0x10DE
  Version:              a1
  Icon:                 audio-card
  Created:              2019-03-09

UHD Graphics 630 (Mobile)
  DeviceId:             bbbf1ce3d1cf15550c3760b354592040292415bb
  Guid:                 db677673-1102-5481-a11e-6f397e123ced
  Guid:                 afb8c4ec-b9a5-5cd3-aaef-cc976ce37eb8
  Plugin:               udev
  Flags:                internal|registered
  Vendor:               Intel Corporation
  VendorId:             PCI:0x8086
  Icon:                 audio-card
  Created:              2019-03-09
```

hughsie, 2019-03-09:
Are you booting with legacy mode or UEFI mode?

hmgibson23, 2019-03-09:
Legacy. I actually flashed it now but would still be good to know how to fix this. The output is the same after flashing.

hughsie, 2019-03-09:
You can't update devices (or even list them) using UEFI if you're not using UEFI mode to boot. Please reopen if the devices don't appear when running in non-legacy mode. Thanks.

closed 2019-03-09 by hughsie