fwupd/fwupd #1096: X1 Carbon Gen6 Bios update fails to install 1.37 QueryCapsuleCapabilities: Unsupported max: 1124656486 reset:1

klusark, 2019-03-15:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version: 1.2.5
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.2.5
```

Note, the switch `--version` is only present since version 0.9.6. If you use an
earlier version, please use the package manager to find out the package
version. For example, `dpkg -l fwupd`.

```shell
$ fwupdmgr get-devices
X1 Carbon Thunderbolt Controller
  DeviceId:             d34eb43e9705af4baa093abc3ff6f5042f2a94e2
  Guid:                 5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              33.00
  Icon:                 computer
  Created:              2019-03-15

20KH003JAU System Firmware
  DeviceId:             1c53551e7da69d896138fac1ae131c83ad46d923
  Guid:                 a4b51dca-8f97-4310-8821-3330f83c9135
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.34
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2019-03-15
  Modified:             2019-03-15
  UpdateState:          needs-reboot

UEFI Device Firmware
  DeviceId:             29749712e93b58730ecd1b8537a5503ff5b7fb27
  Guid:                 9c9d9769-32fa-4841-b550-ea998e754e99
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.60.3561
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-03-15

UEFI Device Firmware
  DeviceId:             9698faabddf0d7b18925cfbbda95f8b0d0dacc53
  Guid:                 3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.10
  VersionLowest:        0.1.10
  Icon:                 audio-card
  Created:              2019-03-15

Unifying Receiver
  DeviceId:             914a7252a18d094bb9497e65e7fd081bdde2a2b8
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
[... 99 lines cut ...]
$ /usr/lib/fwupd/fwupdate -L
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-a4b51dca-8f97-4310-8821-3330f83c9135-0
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:694:efi_main(): Adding new capsule
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)/\EFI\arch\fw\fwupd-a4b51dca-8f97-4310-8821-3330f83c9135.cap
..
.../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A3B14984012C00)/HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C) and PciRoot(0)/Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A3B14984012C00)/HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C) and Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A3B14984012C00)/HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C) and Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A3B14984012C00)/HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C) and Path(131,23,0x010000005CD2E42B81A3B14984012C00)/HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C) and HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:241:fwup_search_file(): Match up! Returning \EFI\arch\fw\fwupd-a4b51dca-8f97-4310-8821-3330f83c9135.cap
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:531:fwup_add_update_capsule(): Read file; 9417280 bytes
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:532:fwup_add_update_capsule(): updates guid: A4B51DCA-8F97-4310-8821-3330F83C9135
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:533:fwup_add_update_capsule(): File guid: A4B51DCA-8F97-4310-8821-3330F83C9135
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:709:efi_main(): n_updates: 1
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:579:fwup_apply_capsules(): QueryCapsuleCapabilities: Unsupported max: 1124656486 reset:1
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:580:fwup_apply_capsules(): Capsules: 1
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:586:fwup_apply_capsules(): Could not apply capsule update: Not Found
../fwupd-1.2.5/plugins/uefi/efi/fwupdate.c:724:efi_main(): Could not apply capsules: Not Found
```

hughsie, 2019-03-15:
> QueryCapsuleCapabilities: Unsupported

That's the important bit. I think `EFI_UNSUPPORTED` is returned from `QueryCapsuleCapabilities()` only when "The capsule image is not supported on this platform". We're missing a bit of error checking which means we print out junk like `max: 1124656486 reset:1` but that's not the source of this bug.

Whilst I can add the missing error checks, we need to know if:

 * The capsule is valid -- i.e. can you upload \EFI\arch\fw\fwupd-a4b51dca-8f97-4310-8821-3330f83c9135.cap somewhere please -- I suspect it's either not there, incomplete or zero sized.

* The system firmware is an any way locked down. I see you have secure boot turned off as you are not using shim, and the optionargs are missing. Did you tweak something in the firmware settings to disable capsule updates or something?

klusark, 2019-03-15:
I'll upload it on Monday.

Does it come from the .cab file? If so you could download it from the Lenovo website if you wanted it sooner. I tried both a manual download from Lenovo and having fwupd get it for me (although thinking now I might not have tried both with efivars 37)

The log has a capsule size, so doesn't that mean it found something at least?

I went through the uefi settings. There were three options for allowing uefi updates from Windows and without system password. They were all turned on. Would you want me to take pictures of the settings or anything?

hughsie, 2019-03-15:
> Does it come from the .cab file

It does, but I have a suspicion that what you have in your ESP is not what we put there :) e.g. memory corruption or something.

klusark, 2019-03-15:
Right, makes sense. It's just the file from the efi partition you want? No special commands to get it or anything?

hughsie, 2019-03-15:
> It's just the file from the efi partition you want

Right.

klusark, 2019-03-17:
[issue-1096.zip](https://github.com/hughsie/fwupd/files/2975703/issue-1096.zip)
Attached.
The sha1 of the .cap file matches exactly the firmware.bin from Lenovo

hughsie, 2019-03-18:
@superm1 any ideas here?

klusark, 2019-03-19:
Tried the latest version from git since I saw you added a check for that error message. Here's the log.

`../fwupd/plugins/uefi/efi/fwupdate.c:158:fwup_populate_update_table(): Found update fwupd-a4b51dca-8f97-4310-8821-3330f83c9135-0
../fwupd/plugins/uefi/efi/fwupdate.c:712:efi_main(): Adding new capsule
../fwupd/plugins/uefi/efi/fwupdate.c:198:fwup_search_file(): Searching Device Path: HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)/\EFI\arch\fw\fwupd-a4b51dca-8f97-4310-8821-3330f83c9135.cap
..
.../fwupd/plugins/uefi/efi/fwupdate.c:218:fwup_search_file(): Device Path prepared: HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)
../fwupd/plugins/uefi/efi/fwupdate.c:228:fwup_search_file(): Device supporting SFSP: PciRoot(0)/Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A3B14984012C00)/HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C) and PciRoot(0)/Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A3B14984012C00)/HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C) and Pci(0x1C,0x4)/Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A3B14984012C00)/HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C) and Pci(0x0,0x0)/Path(131,23,0x010000005CD2E42B81A3B14984012C00)/HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C) and Path(131,23,0x010000005CD2E42B81A3B14984012C00)/HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)
../fwupd/plugins/uefi/efi/fwupdate.c:233:fwup_search_file(): Comparing: HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C) and HD(Part1,Sig5F29DBE6-A0AB-744B-884A-048541927A0C)
../fwupd/plugins/uefi/efi/fwupdate.c:241:fwup_search_file(): Match up! Returning \EFI\arch\fw\fwupd-a4b51dca-8f97-4310-8821-3330f83c9135.cap
../fwupd/plugins/uefi/efi/fwupdate.c:300:fwup_open_file(): Found device
../fwupd/plugins/uefi/efi/fwupdate.c:307:fwup_open_file(): Found volume
../fwupd/plugins/uefi/efi/fwupdate.c:315:fwup_open_file(): Found file
../fwupd/plugins/uefi/efi/fwupdate.c:545:fwup_add_update_capsule(): Read file; 9417280 bytes
../fwupd/plugins/uefi/efi/fwupdate.c:546:fwup_add_update_capsule(): updates guid: A4B51DCA-8F97-4310-8821-3330F83C9135
../fwupd/plugins/uefi/efi/fwupdate.c:547:fwup_add_update_capsule(): File guid: A4B51DCA-8F97-4310-8821-3330F83C9135
../fwupd/plugins/uefi/efi/fwupdate.c:727:efi_main(): n_updates: 1
../fwupd/plugins/uefi/efi/fwupdate.c:593:fwup_apply_capsules(): Could not query capsule capabilities: Unsupported
../fwupd/plugins/uefi/efi/fwupdate.c:742:efi_main(): Could not apply capsules: Unsupported`

superm1, 2019-03-25:
@hughsie Assuming it's not a BIOS bug, only idea I'd have is a firmware NVRAM corruption issue related to the setting for turning on updates.  Perhaps try to reset BIOS defaults and try it again.