fwupd/fwupd #1099: USB disk for updating firmware (UEFI)

goodwillcoding, 2019-03-17:
Hi, 

I have a few machines which have to boot from legacy mode. I've been trying to figure out of creating alive boot Ubuntu USB drive with EFI partition which would allow an EFI upgrade. Is there a coherent way to do that? Any help would be appreciated.

hughsie, 2019-03-18:
I'm not sure this is going to work very well without an "on disk" ESP. If you do get this working with a LiveUSB disk let us know and maybe we can add it to the official docs. Sorry that's not more helpful.

closed 2019-03-18 by hughsie

superm1, 2019-03-25:
Typically when making a USB disk with `dd` it's an ISO9660 filesystem for the majority of the drive that's read only.  There is something that resembles an ESP at the start of the partition table to most BIOS but I don't believe you'll be able to write the capsule to it.

To accomplish what you want to do you'll need to do some more custom partitioning to the USB disk to make it all FAT32 and then convince fwupd that the ESP is located at something like `/cdrom/boot/efi`.
I don't think this is a good road to go down, but if you do get somewhere with it, sure a wiki page for others to follow with the details sounds good.