fwupd/fwupd #1101: Can't update firmware on Lenovo T480s

tpopela, 2019-03-18:
Can't update the firmware on the T480s. There are updates available, but when I run `fwupdmgr update` the update is scheduled, but when restarting the OS is boot normally (no updater is invoked).

```shell
0 $ fwupdmgr get-updates
20L8S2N80D System Firmware has firmware updates:
GUID:                    ebfe8df8-dee7-4692-a721-cbcf5095c5cf
ID:                      com.lenovo.ThinkPadN22ET.firmware
Update Version:          0.1.29
Update Name:             ThinkPad T480s
Update Summary:          Lenovo ThinkPad T480s System Firmware
Update Remote ID:        lvfs
Update Checksum:         SHA1(646d495a92291b4832225183ace8b88dd8e9ae88)
Update Location:         https://fwupd.org/downloads/7d2e5ce0403a04d9cdb73e5607f94d6b51e4f1e7-Lenovo-ThinkPad-T480s-SystemFirmware-1.29.cab
Update Description:      Lenovo ThinkPad T480s System Firmware Version 1.29

                         This update also adds the following features:

ID:                      com.lenovo.ThinkPadN22ET.firmware
Update Version:          0.1.27
Update Name:             ThinkPad T480s
Update Summary:          Lenovo ThinkPad T480s System Firmware
Update Remote ID:        lvfs
Update Checksum:         SHA1(a074dd91e0988c3460edf0de75d843d808491bea)
Update Location:         https://fwupd.org/downloads/d3a8889c8ec5985311addbb6fe022495b5120ca8-Lenovo-ThinkPad-T480s-SystemFirmware-1.27.cab
Update Description:      Lenovo ThinkPad T480s System Firmware Version 1.27

                         This update also adds the following features:

No upgrades for UEFI Device Firmware, current is 184.55.3510: 184.55.3510=same
```

```shell
tpopela  ~  >  13:04
0 $ fwupdmgr update
Downloading 0.1.29 for 20L8S2N80D System Firmware...
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Updating 20L8S2N80D System Firmware…                             ]
Scheduling…              [***************************************]
No upgrades for UEFI Device Firmware, current is 184.55.3510: 184.55.3510=same

An update requires a reboot to complete. Restart now? [Y|n]: n
```

```shell
$ fwupdmgr --version
client version: 1.2.5
compile-time dependency versions
        gusb:   0.3.0
        efivar: 37
daemon version: 1.2.5
```

```shell
$ fwupdmgr get-devices
Thinkpad T480s Thunderbolt Controller
  DeviceId:             010926bac4e8994820c0064541587cff5064685a
  Guid:                 3b125a20-5ed2-5c0f-b22d-ab0e0b687bbc
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              14.00
  Icon:                 computer
  Created:              2019-03-18

20L8S2N80D System Firmware
  DeviceId:             e7cc234e82d08b6a14e246beb6b5f8142091e570
  Guid:                 ebfe8df8-dee7-4692-a721-cbcf5095c5cf
[... 190 lines cut ...]
├── initramfs-4.20.13-200.fc29.x86_64.img
├── initramfs-4.20.14-200.fc29.x86_64.img
├── initramfs-4.20.15-200.fc29.x86_64.img
├── loader
│   └── entries
├── lost+found
├── memtest86+-5.01
├── System.map-4.20.13-200.fc29.x86_64
├── System.map-4.20.14-200.fc29.x86_64
├── System.map-4.20.15-200.fc29.x86_64
├── vmlinuz-0-rescue-7ccf529d5a9148798a1cb710d47e954a
├── vmlinuz-4.20.13-200.fc29.x86_64
├── vmlinuz-4.20.14-200.fc29.x86_64
└── vmlinuz-4.20.15-200.fc29.x86_64

16 directories, 105 files
```

Please answer the following questions:

- Operating system and version: Fedora 29
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): dnf
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? Yes

hughsie, 2019-03-21:
Can you try doing `/usr/libexec/fwupd/fwupdate --set-debug`, then `fwupdmgr update` then reboot, then on failure `/usr/libexec/fwupd/fwupdate --log` -- thanks!

tpopela, 2019-03-21:
The --log part didn't worked for me Richard..

1 $ sudo /usr/libexec/fwupd/fwupdate --log
failed: Error opening file /sys/firmware/efi/efivars/FWUPDATE_DEBUG_LOG-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory

hughsie, 2019-03-25:
If you press F12 to get the boot menu, can you manually run `fwupdx64.efi`?

tpopela, 2019-03-26:
No, as there is no entry like that. There's only Fedora, then the SSD drive and then LAN. Also just to be sure I see that in the BIOS overview screen, there is:

UEFI Secure boot: off

but in the secure boot settings it's set to enabled. I don't know whether it can cause something so I'm mentioning it.

floriankisser, 2019-04-06:
It's working on my T480s with fedora. Secure boot is not an issue and if you have fedora too (assuming yes from your profile pic) I would recommend to turn it on.

Is Security -> "UEFI BIOS Update Option" -> "Windows UEFI Firmware Update" enabled in your BIOS?

tpopela, 2019-04-09:
We debugged it with Richard on IRC and we didn't found a solution so far.

mscherer, 2019-04-18:
I do have the same symptom on the T470s of my employer. And to answer @floriankisser , yup, that's "enabled" on my laptop.

Peetz0r, 2019-04-19:
I was having the same issue on my ThinkPad Yoga 370, but I worked around it and found and worked around some other issues in the process. These are my steps:

* Attempt `fwupdmgr update` a few times, nothing happened
* Manually run `fwupdx64.efi` from a grub shell using chainloading
* Flashed the new firmware
* Fedora uefi entry disapeared. Fallback option didn't work, it bootlooped.
* Got a spare laptop, made a Fedore live usb, booted to a grub shell from there, could boot via there
* Tried chainloading the fallback uefi image (`EFI/BOOT/BOOTX64.EFI`), got an error message (might post this later)
* Tried manually re-creating the fedora uefi entry. Everytime it got removed after a reboot.
* Did a bios settings reset
* Manually recreating uefi entries now worked, also the default Fedora entry got put back by itself somehow.
* Tried re-installing the firmware update (downloaded the .cab, used `--allow-reinstall`), and everything just worked out of the box this time.

I'm running Fedora 29 on a ThinkPad Yoga 370, started the day with firmware 1.32 and now 1.33, fwupdmgr is version 1.2.6.

Dapid, 2019-06-19:
I have a T480s too, and the upgrade worked without problem. I am running Fedora 30 and fwupmgr 1.2.9.

DavidCWGA, 2019-06-26:
I have a T480s and I had the same problem. I reset all the BIOS settings to factory defaults, and it started to work.

I haven't pinned down which setting breaks it yet, but the defaults seem to be OK.

tpopela, 2019-06-27:
Thank you @DavidCWGA that helped me as well and I'm finally up2date (but I didn't took the photo of my previous settings).

closed 2019-09-23 by superm1

Franr, 2019-11-18:
Was struggling with the same problem, until I figured out that locking the ability of changing the boot order through the BIOS was preventing the firmware updates to boot and start (which make sense!).

andodet, 2020-10-30:
I am facing the same issues with a t480s (on Linux Mint 20). Running the following commands don't apply any updates at reboot. 
```
sudo fwupdmgr refresh
sudo fwupdmgr get-updates
sudo fwupdmgr update
sudo reboot
```
I am wondering what's the best option to correctly apply updates:

- Wait for Ubuntu to update to a newer snapshot (as mentioned [here][1]).
- Apply [updates][2] manually with `fwupdmgr install *.cab`
- Change BIOS settings (not quite clear what's the one affecting boot order

Thanks everyone for surfacing this.

[1]: https://github.com/fwupd/fwupd/issues/1794#issuecomment-599721939
[2]: https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN22RN.firmware

romunov, 2021-09-15:
In case you have a dual boot (windows first) and you reset BIOS you may lose grub from UEFI. At this point you won't be able to boot into Fedora. Good luck!

I solved this problem by 

0) raging why the hell do I have to bother with same problems as in 1998
1) going to the supermarket and buying a USB key
2) making it bootable with supergrub2
3) changed boot sequence in BIOS to boot from the USB key
4) detected boot and selected shim.efi
5) followed instructions from [here](https://superuser.com/questions/596317/how-would-i-reinstall-the-grub-efi-bootloader-on-fedora-linux/596318#596318)

tl;dr

```
sudo efibootmgr -c -L Fedora -d /dev/sda -p 100 -l \\EFI\\fedora\\shim.efi
```
and -d and -p parameters can be found in `lsblk`. Search for `/boot/efi` partition, i.e. `-d /dev/nvme0n1 -p 5`.