fwupd/fwupd #1103: UEFI FW capsule GUID should not be case-sensitive

kimonberlin, 2019-03-20:
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

fwupdmgr 1.0.6 on Ubuntu 18.04

abc@xyz:~/Desktop/Linux$ fwupdmgr get-devices
HP EliteBook 840 G5 System Firmware
  DeviceId:             2480d795089c667b755736e089b3c8d2614dd25a
  Guid:                 854bef26-b844-4f03-8e09-b64b4b1ee21e
(...)

abc@xyz:~/Desktop/Linux$ fwupdmgr get-details Q78_010600.cab
Decompressing…         [***************************************]
Unknown Device
  Guid:                 854BEF26-B844-4F03-8E09-B64B4B1EE21E
(...)

abc@xyz:~/Desktop/Linux$ fwupdmgr install Q78_010600.cab
Decompressing…         [***************************************]
No supported devices found

Changing the GUID to lower case in the <firmware type="flashed"> element of metainfo.xml makes the problem go away.

hughsie, 2019-03-20:
We certainly could do this in fwupd and the LVFS. The problem we have is that you'd then have to depend on a very new version of fwupd which would make it hard to apply updates on products like RHEL. Perhaps this is a documentation / linting issue for the LVFS?

kimonberlin, 2019-03-22:
Yeah, the chicken-and-egg aspect is thorny... by the time we upload to LVFS, we've already tested the capsule locally, so it would be too late to flag a warning. With the current toolchain, our best bet would be to modify our own scripts to ensure lower case when generating metainfo.xml, but that leaves everyone else exposed. Updating fwupd would hopefully minimize the total pain over time. Maybe also document the limitation in the vendor section of LVFS?

superm1, 2019-03-25:
LVFS repacks anyway so it seems sensible to me that it can fix the case at that time to be consistent no?

hughsie, 2019-03-25:
> LVFS repacks anyway

Right, we repack but we don't actually re-write the metainfo.xml file. Fixing it on the LVFS would make it correct in the AppStream metadata, but not actually in the MetaInfo file, so actually installing-after-match-and-download would still fail.

superm1, 2019-03-25:
In that case I agree it's probably best for LVFS to detect and flag this as a warning that this won't work on older fwupd clients and in newer fwupd make it case insensitive.

closed 2019-05-01 by hughsie