fwupd/fwupd #1105: Starting fwupd service temporarily disconnects Dell Thunderbolt Dock

wiktor-k, 2019-03-21:
Hello,

I've got an interesting issue with `fwupd`. When I start the service `systemctl start fwupd` (or automatically on login when it's enabled) and I've got Dell TB16 dock connected the connection will temporarily be broken and re-estabilished several times (just like I'd pull the Thunderbolt cable and re-insert it). After a while it all settles. If the dock is disconnected nothing unusual happens. 

As a workaround `systemctl mask fwupd` will make this problem disappear on next system startup.

For the record I've got the dock and cable authorized using `boltctl` but disabling Thunderbolt security in system firmware doesn't change anything.

I've got all firmware updates installed for both XPS 9350 and the dock (for dock it required running the firmware update under Windows).

```shell
$ fwupdmgr --version
client version:	1.2.5
compile-time dependency versions
	gusb:	0.3.0
	efivar:	35
daemon version:	1.2.5
```

```shell
$ fwupdmgr get-devices
XPS 9350 Thunderbolt Controller
  DeviceId:             948425057b80f0b719b9d5374aa17ed7ff2fc861
  Guid:                 dbbe0ed9-df91-529c-ac86-8976b78f726d
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|supported|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              26.01
  Icon:                 computer
  Created:              2019-03-21

Dell Thunderbolt Cable
  DeviceId:             fbdcc600105a732e21b06f910e9ad78c88ee432a
  ParentDeviceId:       84edf044a6b53037732cfc7591b65a8d36384cc9
  Guid:                 b4fd3cdf-4e3a-5090-a583-45367cfd6421
  Plugin:               thunderbolt
  Flags:                updatable|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              16.00
  Icon:                 audio-card
  Created:              2019-03-21

Dell Thunderbolt Dock
  DeviceId:             a6d6fc68b117742d077081667731649ff05fe565
  ParentDeviceId:       84edf044a6b53037732cfc7591b65a8d36384cc9
  Guid:                 76cc74d4-f062-5b93-a11c-8d2a58a25848
  Plugin:               thunderbolt
  Flags:                updatable|registered
  Vendor:               Dell
  VendorId:             TBT:0x00D4
  Version:              27.00
  Icon:                 audio-card
  Created:              2019-03-21

XPS 13 9350 System Firmware
  DeviceId:             79f2996ff11608674c5d2e8db20e10e0b646f2cb
  Guid:                 33773727-8ee7-4d81-9fa0-57e8d889e1fa
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.10.1
  VersionLowest:        0.1.10.1
  Icon:                 computer
  Created:              2019-03-21
  UpdateState:          success

XPS 13 9350 TPM 2.0
  DeviceId:             eb4a4ec71c680a34c6248dc94ca017f81c772aa8
  Guid:                 b62a2412-5ac4-5350-b16e-7e8f4655d096
  Summary:              Platform TPM device
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              1.3.2.8
  Icon:                 computer
  Created:              2019-03-21

Dell TB16 EC
  DeviceId:             6d24f6d588ad8116a0d7f58d144222318e239ec7
  ParentDeviceId:       84edf044a6b53037732cfc7591b65a8d36384cc9
  Guid:                 33cc8870-b1fc-4ec7-948a-c07496874faf
  Summary:              A Thunderbolt™ 3 docking station
  Plugin:               uefi
  Flags:                updatable|require-ac|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              1.0.0.16
  Icon:                 computer
  Created:              2019-03-21

Dell TB16 Thunderbolt Cable
  DeviceId:             f8194a42320518472a2e67c1d89018c7a5219841
  ParentDeviceId:       84edf044a6b53037732cfc7591b65a8d36384cc9
  Guid:                 6dc832fc-5bb0-4e63-a2ff-02aaba5bc1dc
  Summary:              A Thunderbolt™ 3 docking station
  Plugin:               uefi
  Flags:                updatable|require-ac|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              0.0.3.18
  Icon:                 computer
  Created:              2019-03-21

Dell TB16 Port Controller 1
  DeviceId:             b3fa3dc04e3450450bb17910417562e5b6cf69d0
  ParentDeviceId:       84edf044a6b53037732cfc7591b65a8d36384cc9
  Guid:                 8fe183da-c94e-4804-b319-0f1ba5457a69
  Summary:              A Thunderbolt™ 3 docking station
  Plugin:               uefi
  Flags:                updatable|require-ac|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              0.1.2.17
  Icon:                 computer
  Created:              2019-03-21

Dell TB16 Port Controller 2
  DeviceId:             4bcf1477575b55912993240dd1edd8b0ebb6d491
  ParentDeviceId:       84edf044a6b53037732cfc7591b65a8d36384cc9
  Guid:                 1b52c630-86f6-4aee-9f0c-474dc6be49b6
  Summary:              A Thunderbolt™ 3 docking station
  Plugin:               uefi
  Flags:                updatable|require-ac|registered|needs-reboot
  Vendor:               Dell Inc.
  Version:              0.1.2.50
  Icon:                 computer
  Created:              2019-03-21

SAMSUNG SSD PM830 mSATA 256GB
  DeviceId:             1b44569f3df7eae21d2fc06a369d911542d89bd3
  Guid:                 952d4260-8e60-5d91-b8ce-8fb06b9ef8f0
  Guid:                 2913d726-c964-54b0-91e2-caea13a44f09
  Guid:                 2f7f88cd-f726-5cf0-b4bf-883ee208d1c3
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                updatable|require-ac|registered|needs-reboot
  Version:              CXM13D1Q
  Icon:                 drive-harddisk
  Created:              2019-03-21

PM951 NVMe SAMSUNG 1024GB
  DeviceId:             e11623b2caa18fee292058a5c09ca4e6152f7ecf
  Guid:                 cce2386f-6d68-5f85-9f3f-ebead9288696
  Guid:                 77bb4022-9503-586c-99b7-c8a64502da30
  Guid:                 8fd4ca73-d0ae-52e8-8977-461435c6f4cf
  Guid:                 02d7a481-b251-577f-8079-c3dc4bf6c942
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              BXV77D0Q
  Icon:                 drive-harddisk
  Created:              2019-03-21
```

```shell
$ efibootmgr -v
BootCurrent: 0004
Timeout: 0 seconds
BootOrder: 0004,0000,0006
Boot0000* Windows Boot Manager	HD(1,GPT,5c589a06-6918-4453-911c-9ae8269031ef,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0004* Signed NVMe	PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/NVMe(0x1,00-25-38-45-5A-1B-2A-88)/HD(1,GPT,5c589a06-6918-4453-911c-9ae8269031ef,0x800,0xfa000)/File(\EFI\Arch\linux-signed.efi)
Boot0006* UEFI: PM951 NVMe SAMSUNG 1024GB, Partition 1	HD(1,GPT,5c589a06-6918-4453-911c-9ae8269031ef,0x800,0xfa000)/File(EFI\Microsoft\Boot\bootmgfw.efi)..BO
```

```shell
$ efivar -l | grep fw
(no output)
```

```shell
$ tree /boot
/boot
├── disk.bin.gpg
├── EFI
│   ├── Arch
│   │   ├── fw
│   │   │   └── fwupdx64.efi
│   │   ├── fwupdx64.efi
│   │   ├── linux-signed.efi
│   │   └── linux-signed.efi.bak
│   ├── Boot
│   │   └── bootx64.efi
│   ├── dell
│   │   ├── bios
│   │   │   └── recovery
│   │   │       ├── BIOS_CUR.RCV
│   │   │       └── BIOS_PRE.rcv
│   │   └── logs
│   │       ├── diags_current.xml
│   │       └── diags_previous.xml
(...lots of Microsoft files, omitted, unless it's necessary)
```

Please answer the following questions:

- Operating system and version:Arch Linux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):`pacman -S fwupd`
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled (only for the UEFI plugin)? Yes

superm1, 2019-03-25:
My guess is you only have one external monitor plugged in to the dock, right?
This sounds like it's https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1806323 and/or https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1793965

Basically what's happening is that when the MST controller is powered up without a monitor plugged in it creates an extra fake monitor to probe information.  The kernel graphics driver will flicker while this happens.

You can blacklist the dell plugin in `/etc/fwupd/daemon.conf` to temporarily work around the issue.

I would wonder though what kernel version you've reproduced this issue and if it's not 5.x, if it goes away using brand new kernels at all.

wiktor-k, 2019-03-25:
Hi @superm1, thanks a lot for help!

> My guess is you only have one external monitor plugged in to the dock, right?

I do have one external monitor. Do you say that if I bring another one it will also workaround the issue? Because that's something that I can do (not at the moment but in general).

> I would wonder though what kernel version you've reproduced this issue 

```
$ uname -a
Linux gluon 5.0.3-arch1-1-ARCH #1 SMP PREEMPT Tue Mar 19 13:09:13 UTC 2019 x86_64 GNU/Linux
```

superm1, 2019-03-25:
> I do have one external monitor. Do you say that if I bring another one it will also workaround the issue? Because that's something that I can do (not at the moment but in general).

As long as it's connected to the second MST hub, yes I believe it would also work around the issue.

To further elaborate what's going on and why blacklisting the dell plugin would workaround it;
dell plugin [toggles a pin on both MST hubs](https://github.com/hughsie/fwupd/blob/master/plugins/dell/fu-plugin-dell.c#L758).  When this pin is toggled if something is plugged in it's a NOOP.  If nothing is plugged in, it wakes up MST controllers.  Communication over DP aux channel doesn't work though in this situation so MST controller creates an extra fake display.  If you time it right you can actually see this display in `xrandr` output.

>  5.0.3

OK so still an issue with 5.x series kernels.  I do think this is squarely in the i915 not handling the fake monitor well department.  Would you be able to follow https://01.org/linuxgraphics/documentation/how-report-bugs to get it reported to the i915 maintainers?
In the interim you can apply either of those workarounds but it would be nice to get fixed.

wiktor-k, 2019-03-25:
Okay, I connected the second MST hub (HDMI) to TV, unmasked the service and started it. No flickering at all (it did flicker without a second display). So this is definitely the issue you pointed out.

>  Would you be able to follow https://01.org/linuxgraphics/documentation/how-report-bugs to get it reported to the i915 maintainers?

Yes, I'll do that... once I parse and gather stuff they need.

> In the interim you can apply either of those workarounds but it would be nice to get fixed.

Thanks for your help & workarounds. I'll keep this bug updated in case i915 maintainers respond.

wiktor-k, 2019-04-24:
I attach the upstream ticket for cross-reference: https://bugs.freedesktop.org/show_bug.cgi?id=110506

commit eaa95eaa (2019-08-28): synapticsmst: Partially rewrite the plugin (Fixes #1105)

This removes the 'two-layer' FuDevice and FuSyanpticsmstDevice model, where
a complicated plugin cache was used to add and remove devices to the daemon.

By making FuSynapticsmstDevice derive from FuDevice rather than GObject we can
also use a lot of the helper functionality like the other plugins, for instance
->prepare_firmware().

The `drm_dp_aux_dev` devices do not emit uevents on unplug/re-plug and so all
devices of `drm` class are watched and the actual DP AUX devices rescanned after
a small delay. When the AUX devices emit change [... cut]
commit d350b2ec (2019-08-28): synapticsmst: Partially rewrite the plugin (Fixes #1105)

This removes the 'two-layer' FuDevice and FuSyanpticsmstDevice model, where
a complicated plugin cache was used to add and remove devices to the daemon.

By making FuSynapticsmstDevice derive from FuDevice rather than GObject we can
also use a lot of the helper functionality like the other plugins, for instance
->prepare_firmware().

The `drm_dp_aux_dev` devices do not emit uevents on unplug/re-plug and so all
devices of `drm` class are watched and the actual DP AUX devices rescanned after
a small delay. When the AUX devices emit change [... cut]
closed 2019-08-28 by superm1