fwupd/fwupd #1108: Lenovo firmware active discovery idea.

septs, 2019-03-23:
Based <https://download.lenovo.com/supportdata/index.html>

1. Through product info
e.q: <https://download.lenovo.com/supportdata/product.html?id=LAPTOPS-AND-NETBOOKS/THINKPAD-T-SERIES-LAPTOPS/THINKPAD-T460P>
Data source: <https://download.lenovo.com/SupportData/Product/LAPTOPS-AND-NETBOOKS/THINKPAD-T-SERIES-LAPTOPS/THINKPAD-T460P.json>

2. Get UEFI firmware Bootable ISO Driver ID
e.q: https://download.lenovo.com/supportdata/driver.html?id=DS112082&l=en
Data source: https://download.lenovo.com/supportdata/Driver/DS112082.en.json

3. Convert bootable iso to fwupd package format?

So then we can know that the product  has firmware update by periodically refreshing (for driver info).

hughsie, 2019-03-23:
Unfortunately we have to leave the OEM to the timing of updates. Also, some updates are not done on Linux as they're either not supported or not required, or have different QA requirements. Lenovo are shipping a lot of firmware on the LVFS now, and they are still catching up with older models, so I think we just have to be patient. Thanks for opening this bug with the links, I'll have a look locally and see if anything obvious is missing.

closed 2019-03-23 by hughsie

septs, 2019-03-23:
The maintainer can be noified in this way.

hughsie, 2019-03-23:
> The maintainer can be notified in this way

The maintainer is probably the same person that's writing the .json file...