fwupd/fwupd #1111: Copy translations from Transifex to GitHub more often

scootergrisen, 2019-03-25:
Are the po files only updated from Transifex before a new release?

I have translated fwupd on Transifex.

But when i look at https://l10n.gnome.org/languages/da/freedesktop-org/ui/ it says its not translated.

I guess this is because GNOME gets the po files from GitHub.

I would like the GNOME website to show the translations amount correctly so i can get a better overview of what is left to translate.

So perhaps copy the translations from Transifex to GitHub continuesly even if there is not release.

superm1, 2019-03-25:
It does seem there is an automated way to do it with CI maybe: https://docs.transifex.com/integrations/github

superm1, 2019-03-25:
Although; that's for pushing _to_ transifex.  Not pulling.  This one: https://docs.transifex.com/integrations/github-txgh seems to show more about how to keep it bidirectionally synced.

hughsie, 2019-03-25:
This is semi-deliberate; https://github.com/hughsie/fwupd/blob/master/contrib/fix_translations.py strips out a lot of the junk before storing the strings in source control.

closed 2019-10-02 by hughsie