fwupd/fwupd #1118: Unable to downgrade after upgrading ATA

superm1, 2019-03-27:
Since ATA needs activation I've found that going through the cycle of:

`13000000 -> 13000001`
works fine
but after that upgrade
`13000001-> 13000000`
doesn't work.  

If I clear `pending.db` and try again, the reverse stops working.  The activation flag doesn't get stored anymore.  So this appears to be related to the storage of the history perhaps.

The update shows:
```
modifying device Disk [5dbeb140337f610d54913c0f43d15fdaa0eafb51], only version old 13000001
```

Then calling get-devices to check it again shows:
```
22:12:11:0546 FuEngine             installed version 13000001 matching history 0.198.23873
22:12:11:0546 FuHistory            modifying device Disk [5dbeb140337f610d54913c0f43d15fdaa0eafb51], only version new 13000000
```

hughsie, 2019-03-28:
> upgrade doesn't work

Do you mean the activation doesn't get called?

hughsie, 2019-03-28:
Out of interest, does this commit fix things? ^^^

superm1, 2019-03-28:
Yes I mean activation doesn't get called because needs activation flag is not staying. That fix looks promising. I'll try later.

closed 2019-03-28 by superm1