fwupd/fwupd #1121: fwupd "stuck" in a Dell Latitude 7424 Rugged Extreme

aleksander0m, 2019-03-28:
The exact version in use is a git snapshot from 20190302, so fwupd 1.2.5 plus including the MM plugin support. The setup is running Ubuntu 18.04 with kernel 4.19 manually installed on top. The setup has an external DW5821e modem also plugged in, correctly detected by MM at this point.

The problem here is that as soon as fwupd starts, e.g. manually running `$ sudo /usr/lib/fwupd/fwupd --verbose`, the process reaches a point where it gets totally stuck, and it's unusable. Trying to kill the process, even with SIGKILL, doesn't seem to have any effect. fwupdmgr also gets stuck waiting for the daemon to reply I assume.

The full --verbose syslog when the daemon is run is here:
```shell
16:24:23:0107 FuDebug              Verbose debugging enabled (on console 1)
16:24:23:0107 FuConfig             loading config values from /etc/fwupd/daemon.conf
16:24:23:0110 Fwupd                looking for os-release at /etc/os-release
16:24:23:0110 FuConfig             loading /usr/share/fwupd/metainfo
16:24:23:0110 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
16:24:23:0110 XbSilo               file: ce6bbfdb-ebfa-541a-8b28-52b3bb8f690c, current:ce6bbfdb-ebfa-541a-8b28-52b3bb8f690c, cached: (null)
16:24:23:0110 XbSilo               loading silo with file contents
16:24:23:0110 FuConfig             using config path of /etc/fwupd
16:24:23:0110 FuConfig             loading config from /etc/fwupd/remotes.d/fwupd.conf
16:24:23:0111 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs.conf
16:24:23:0111 FuConfig             loading config from /etc/fwupd/remotes.d/vendor.conf
16:24:23:0111 FuConfig             loading config from /etc/fwupd/remotes.d/vendor-directory.conf
16:24:23:0111 FuConfig             loading config from /etc/fwupd/remotes.d/lvfs-testing.conf
16:24:23:0111 FuConfig             using config path of /usr/lib/fwupd
16:24:23:0111 FuConfig             path /usr/lib/fwupd/remotes.d does not exist
16:24:23:0111 FuConfig             ordering lvfs=fwupd+1
16:24:23:0111 FuConfig             ordering lvfs-testing=lvfs+1
16:24:23:0112 FuIdle               setting timeout to 7200s
16:24:23:0112 FuHwids              smbios property Manufacturer=Dell Inc.
16:24:23:0112 FuHwids              smbios property EnclosureKind=a
16:24:23:0112 FuHwids              smbios property Family=Latitude
16:24:23:0112 FuHwids              smbios property ProductName=Latitude 7424 Rugged Extreme
16:24:23:0112 FuHwids              smbios property ProductSku=087B
16:24:23:0112 FuHwids              smbios property BiosVendor=Dell Inc.
16:24:23:0112 FuHwids              smbios property BiosVersion=1.0.0
16:24:23:0112 FuHwids              smbios property BiosMajorRelease=01
16:24:23:0112 FuHwids              smbios property BiosMinorRelease=00
16:24:23:0112 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
16:24:23:0112 FuHwids              smbios property BaseboardProduct=
16:24:23:0112 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/altos.quirk
16:24:23:0112 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/ata.quirk
16:24:23:0112 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/colorhug.quirk
16:24:23:0112 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/csr-aiaiai.quirk
16:24:23:0113 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dell-dock.quirk
16:24:23:0113 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dell.quirk
16:24:23:0113 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/dfu.quirk
16:24:23:0114 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/ebitdo.quirk
16:24:23:0114 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/fastboot.quirk
16:24:23:0114 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/flashrom.quirk
16:24:23:0115 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/nitrokey.quirk
16:24:23:0115 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/nvme.quirk
16:24:23:0115 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/rts54hid.quirk
16:24:23:0115 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/rts54hub.quirk
16:24:23:0115 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/steelseries.quirk
16:24:23:0115 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/superio.quirk
16:24:23:0115 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/synapticsmst.quirk
16:24:23:0115 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/uefi.quirk
16:24:23:0115 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/unifying.quirk
16:24:23:0115 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/wacom-raw.quirk
16:24:23:0116 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d/wacom-usb.quirk
16:24:23:0116 FuQuirks             now 183 quirk entries
16:24:23:0116 FuQuirks             no /var/lib/fwupd/quirks.d, skipping
16:24:23:0116 FuEngine             remote lvfs-testing not enabled, so skipping
16:24:23:0116 FuEngine             remote vendor not enabled, so skipping
16:24:23:0116 FuEngine             remote vendor-directory not enabled, so skipping
16:24:23:0116 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
16:24:23:0116 XbSilo               0.02ms      parse blob
16:24:23:0116 XbSilo               0.06ms      loaded file
16:24:23:0116 XbSilo               file: 066297ab-c391-501c-94bc-1377a36431d6, current:066297ab-c391-501c-94bc-1377a36431d6, cached: (null)
16:24:23:0116 XbSilo               loading silo with file contents
16:24:23:0116 XbSilo               0.01ms      parse blob
16:24:23:0116 XbSilo               0.18ms      query on / with `components/component` limit=0 -> 233 results
16:24:23:0116 FuEngine             233 components now in silo
16:24:23:0117 XbSilo               0.27ms      query on / with `components/component/provides/firmware` limit=0 -> 322 results
[... 472 lines cut ...]
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       └── zfs.mod
├── initrd.img-4.15.0-20-generic
├── initrd.img-4.15.0-46-generic
├── initrd.img-4.19.0
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── retpoline-4.15.0-20-generic
├── System.map-4.15.0-20-generic
├── System.map-4.15.0-46-generic
├── System.map-4.19.0
├── vmlinuz-4.15.0-20-generic
├── vmlinuz-4.15.0-46-generic
└── vmlinuz-4.19.0
```

Please answer the following questions:

- Operating system and version:
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc): apt-get from a personal PPA.
- Have you tried rebooting? yeah
- Are you using an NVMe disk? don't think so
- Is secure boot enabled (only for the UEFI plugin)? don't think so, this is a remote setup I have access to only via SSH.

aleksander0m, 2019-03-28:
From the above log, the only error that seems to be reported is this one:
```
16:24:23:0176 FuPluginUefi         Failed to get PCR0s: Child process exited with code 1
```

superm1, 2019-03-28:
A few shots in the dark:
* The 7424 is fairly new - is this pre-production HW by chance?  If so, do you know which HW build it is?
* Anything in the kernel log when this happens?
* Since the last thing is left on was synapticsmst, can you try to run with synapticsmst blacklisted to see if it helps?

hughsie, 2019-03-28:
Does `strace` point fingers?

aleksander0m, 2019-03-28:
I have very limited ssh access to this system, and rebooting it ends up breaking ssh. The stale fwupd processes I have around are in uninterruptible sleep state, so cannot be manually killed.

No idea about the 7424 HW build, will ask.

The kernel log has indeed some kernel crash reported:
```
[16862.519947] [drm:generic_reg_wait [amdgpu]] *ERROR* REG_WAIT timeout 10us * 160 tries - submit_channel_request line:246
[16862.519983] WARNING: CPU: 5 PID: 3015 at drivers/gpu/drm/amd/amdgpu/../display/dc/dc_helper.c:254 generic_reg_wait+0xe2/0x160 [amdgpu]
[16862.519984] Modules linked in: rfcomm snd_hda_codec_realtek snd_hda_codec_generic cmac bnep arc4 nls_iso8859_1 amdkfd amd_iommu_v2 amdgpu i915 dell_rbtn iwlmvm snd_soc_skl wmi_bmof dell_wmi intel_rapl x86_pkg_temp_thermal dell_laptop intel_powerclamp snd_soc_skl_ipc dell_smbios coretemp snd_soc_sst_ipc dell_wmi_descriptor kvm_intel dcdbas snd_soc_sst_dsp snd_hda_ext_core kvmgt dell_smm_hwmon snd_soc_acpi_intel_match mac80211 vfio_mdev crct10dif_pclmul mdev vfio_iommu_type1 crc32_pclmul snd_soc_acpi vfio ghash_clmulni_intel snd_soc_core kvm pcbc snd_compress ac97_bus aesni_intel snd_hda_codec_hdmi snd_pcm_dmaengine chash gpu_sched aes_x86_64 snd_hda_intel crypto_simd snd_hda_codec cryptd ttm glue_helper pcmcia irqbypass snd_hda_core drm_kms_helper intel_cstate intel_rapl_perf snd_hwdep uvcvideo snd_pcm
[16862.520008]  drm cdc_ether iwlwifi cdc_mbim videobuf2_vmalloc snd_seq_midi videobuf2_memops cdc_ncm snd_seq_midi_event videobuf2_v4l2 snd_rawmidi videobuf2_common videodev input_leds serio_raw btusb qmi_wwan btrtl r8152 snd_seq btbcm media cdc_wdm btintel usbnet option mii usb_wwan bluetooth usbserial cdc_acm hid_multitouch joydev snd_seq_device snd_timer cfg80211 snd yenta_socket pcmcia_rsrc idma64 pcmcia_core soundcore virt_dma i2c_algo_bit fb_sys_fops syscopyarea mei_me sysfillrect ecdh_generic sysimgblt mei ucsi_acpi intel_lpss_pci processor_thermal_device typec_ucsi intel_lpss intel_soc_dts_iosf intel_pch_thermal typec wmi soc_button_array mac_hid intel_hid int3400_thermal video int3403_thermal dell_smo8800 sparse_keymap acpi_thermal_rel int340x_thermal_zone acpi_pad sch_fq_codel parport_pc
[16862.520033]  ppdev lp parport ip_tables x_tables autofs4 hid_generic usbhid uas usb_storage e1000e ahci libahci i2c_hid hid
[16862.520040] CPU: 5 PID: 3015 Comm: fwupd Not tainted 4.19.0 #1
[16862.520040] Hardware name: Dell Inc. Latitude 7424 Rugged Extreme/, BIOS 1.0.0 09/12/2018
[16862.520062] RIP: 0010:generic_reg_wait+0xe2/0x160 [amdgpu]
[16862.520063] Code: ab 44 8b 45 20 48 8b 4d 18 44 89 e6 8b 55 10 48 c7 c7 f0 bf 12 c1 44 89 4d d4 e8 79 1d 65 ff 41 83 7d 18 01 44 8b 4d d4 74 02 <0f> 0b 48 83 c4 18 44 89 c8 5b 41 5c 41 5d 41 5e 41 5f 5d c3 41 0f
[16862.520064] RSP: 0018:ffffbdff831ffbe0 EFLAGS: 00010297
[16862.520065] RAX: 0000000000000000 RBX: 00000000000000a1 RCX: 0000000000000000
[16862.520065] RDX: 0000000000000000 RSI: ffff965fee356428 RDI: ffff965fee356428
[16862.520066] RBP: ffffbdff831ffc20 R08: 00000000000004d5 R09: 00000000ffffffff
[16862.520066] R10: fffff464915dbd00 R11: ffffffff981923ed R12: 000000000000000a
[16862.520067] R13: ffff965fdea5d480 R14: 0000000000000001 R15: 0000000000000000
[16862.520068] FS:  00007f17444485c0(0000) GS:ffff965fee340000(0000) knlGS:0000000000000000
[16862.520068] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[16862.520069] CR2: 00007f1732987150 CR3: 000000045f4fe006 CR4: 00000000003606e0
[16862.520070] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[16862.520070] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[16862.520070] Call Trace:
[16862.520097]  submit_channel_request+0x394/0x480 [amdgpu]
[16862.520118]  dc_link_aux_transfer+0xca/0x160 [amdgpu]
[16862.520142]  dm_dp_aux_transfer+0x59/0x120 [amdgpu]
[16862.520146]  drm_dp_dpcd_access+0x75/0x110 [drm_kms_helper]
[16862.520148]  drm_dp_dpcd_read+0x33/0xc0 [drm_kms_helper]
[16862.520152]  auxdev_read_iter+0x10f/0x1d0 [drm_kms_helper]
[16862.520154]  __vfs_read+0x106/0x180
[16862.520155]  vfs_read+0x8e/0x130
[16862.520156]  ksys_read+0x55/0xc0
[16862.520157]  __x64_sys_read+0x1a/0x20
[16862.520159]  do_syscall_64+0x5a/0x120
[16862.520161]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[16862.520162] RIP: 0033:0x7f1741b3a384
[16862.520163] Code: 84 00 00 00 00 00 41 54 55 49 89 d4 53 48 89 f5 89 fb 48 83 ec 10 e8 8b fc ff ff 4c 89 e2 41 89 c0 48 89 ee 89 df 31 c0 0f 05 <48> 3d 00 f0 ff ff 77 38 44 89 c7 48 89 44 24 08 e8 c7 fc ff ff 48
[16862.520163] RSP: 002b:00007fffadd980f0 EFLAGS: 00000246 ORIG_RAX: 0000000000000000
[16862.520164] RAX: ffffffffffffffda RBX: 0000000000000013 RCX: 00007f1741b3a384
[16862.520165] RDX: 0000000000000001 RSI: 00007fffadd98154 RDI: 0000000000000013
[16862.520165] RBP: 00007fffadd98154 R08: 0000000000000000 R09: 000055fd7e9c7010
[16862.520166] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000001
[16862.520166] R13: 00007fffadd981c0 R14: 00007fffadd98154 R15: 000055fd7eae35b0
[16862.520167] ---[ end trace 8e8b280a234718a5 ]---
[16862.520176] BUG: unable to handle kernel NULL pointer dereference at 0000000000000008
[16862.520177] PGD 0 P4D 0 
[16862.520179] Oops: 0000 [#1] SMP PTI
[16862.520180] CPU: 5 PID: 3015 Comm: fwupd Tainted: G        W         4.19.0 #1
[16862.520181] Hardware name: Dell Inc. Latitude 7424 Rugged Extreme/, BIOS 1.0.0 09/12/2018
[16862.520205] RIP: 0010:dal_ddc_close+0xd/0x30 [amdgpu]
[... cut ...]
[16862.520386] R10: fffff464915dbd00 R11: 0000000000000001 R12: 0000000000000000
[16862.520387] R13: ffffbdff831ffdd0 R14: ffffbdff831ffd3d R15: 0000000000000000
[16862.520388] FS:  00007f17444485c0(0000) GS:ffff965fee340000(0000) knlGS:0000000000000000
[16862.520389] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[16862.520390] CR2: 0000000000000008 CR3: 000000045f4fe006 CR4: 00000000003606e0
[16862.520391] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[16862.520392] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
```

superm1, 2019-03-28:
Interesting, this is quite similar to the issue that was reported in https://github.com/hughsie/fwupd/pull/1117 and @vicamo was seeing on some other pre-release hardware with `amdgpu`.

In that case it was determined that newer hardware build fixed the problem though.

I believe for now you should be able to blacklist `synapticsmst` to get around this issue, but we are definitely not to the bottom of it yet since initially it pointed at amdgpu in 4.18.x fixing it and you're running 4.19.0.

superm1, 2019-03-28:
I'm starting to think maybe we just blacklist amdgpu/synapticsmst plugin until we're at the bottom of this.

superm1, 2019-03-28:
This is the trace seen on the other system:

```
kernel NULL pointer dereference in amdgpu driver right after the freeze happens:

[  129.549237] [drm:generic_reg_wait [amdgpu]] *ERROR* REG_WAIT timeout 10us * 160 tries - submit_channel_request line:246
[  129.549285] WARNING: CPU: 9 PID: 4273 at /var/lib/dkms/amdgpu/18.50-727488/build/amd/amdgpu/../display/dc/dc_helper.c:249 generic_reg_wait+0xe2/0x150 [amdgpu]
[  129.549286] Modules linked in: ccm bnep btusb uvcvideo btrtl videobuf2_vmalloc btbcm videobuf2_memops btintel videobuf2_v4l2 videobuf2_core bluetooth videodev media ecdh_generic snd_hda_codec_realtek snd_hda_codec_generic snd_hda_codec_hdmi joydev hid_multitouch arc4 dell_rbtn nls_iso8859_1 intel_rapl x86_pkg_temp_thermal intel_powerclamp iwlmvm(OE) coretemp dell_laptop mac80211(OE) kvm_intel snd_hda_intel kvm dell_smm_hwmon snd_hda_codec irqbypass snd_hda_core crct10dif_pclmul snd_hwdep crc32_pclmul ghash_clmulni_intel pcbc snd_pcm snd_seq_midi snd_seq_midi_event iwlwifi(OE) snd_rawmidi aesni_intel aes_x86_64 crypto_simd glue_helper dell_wmi cryptd snd_seq cfg80211(OE) dell_smbios dcdbas snd_seq_device input_leds snd_timer intel_cstate rtsx_pci_ms idma64 virt_dma intel_rapl_perf wmi_bmof serio_raw
[  129.549305]  dell_wmi_descriptor intel_wmi_thunderbolt snd memstick compat(OE) mei_me intel_lpss_pci soundcore mei intel_lpss processor_thermal_device ucsi_acpi intel_soc_dts_iosf typec_ucsi intel_pch_thermal typec int3403_thermal int340x_thermal_zone dell_smo8800 intel_hid int3400_thermal mac_hid sparse_keymap acpi_thermal_rel acpi_pad dptf_power sch_fq_codel parport_pc ppdev lp parport ip_tables x_tables autofs4 btrfs zstd_compress raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq libcrc32c raid1 raid0 multipath linear dm_mirror dm_region_hash dm_log amdgpu(OE) rtsx_pci_sdmmc amdchash(OE) amd_sched(OE) amd_iommu_v2 amdttm(OE) i915 psmouse e1000e ptp i2c_algo_bit amdkcl(OE) pps_core drm_kms_helper rtsx_pci syscopyarea sysfillrect thunderbolt sysimgblt ahci fb_sys_fops
[  129.549325]  libahci drm wmi i2c_hid hid video
[  129.549328] CPU: 9 PID: 4273 Comm: fwupd Tainted: G           OE    4.15.0-1034-oem #39-Ubuntu
[  129.549329] Hardware name: <redacted>
[  129.549353] RIP: 0010:generic_reg_wait+0xe2/0x150 [amdgpu]
[  129.549353] RSP: 0018:ffffb7e0870b7be8 EFLAGS: 00010297
[  129.549354] RAX: 00000000000000f6 RBX: 00000000000000a1 RCX: 0000000000000000
[  129.549355] RDX: 0000000000000000 RSI: ffff8a9a7bc56498 RDI: ffff8a9a7bc56498
[  129.549355] RBP: ffffb7e0870b7c28 R08: 0000000000000001 R09: 000000000000052c
[  129.549356] R10: 00000000ffffffff R11: 0000000000000000 R12: 000000000000000a
[  129.549356] R13: ffff8a9a68e8ed80 R14: 0000000000000001 R15: 0000000000000000
[  129.549357] FS:  00007f55b9e62940(0000) GS:ffff8a9a7bc40000(0000) knlGS:0000000000000000
[  129.549357] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[  129.549358] CR2: 000000c820911000 CR3: 0000000875fbc005 CR4: 00000000003606e0
[  129.549358] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[  129.549359] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[  129.549359] Call Trace:
[  129.549387]  submit_channel_request+0x394/0x480 [amdgpu]
[  129.549410]  dc_link_aux_transfer+0xca/0x150 [amdgpu]
[  129.549436]  dm_dp_aux_transfer+0x59/0x120 [amdgpu]
[  129.549439]  drm_dp_dpcd_access+0x75/0x120 [drm_kms_helper]
[  129.549441]  drm_dp_dpcd_read+0x31/0x70 [drm_kms_helper]
[  129.549444]  auxdev_read_iter+0x10f/0x1d0 [drm_kms_helper]
[  129.549446]  ? __seccomp_filter+0x49/0x590
[  129.549448]  new_sync_read+0xe4/0x130
[  129.549449]  __vfs_read+0x29/0x40
[  129.549450]  vfs_read+0x8e/0x130
[  129.549451]  SyS_read+0x55/0xc0
[  129.549452]  do_syscall_64+0x73/0x130
[  129.549454]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
[  129.549454] RIP: 0033:0x7f55b7732384
[  129.549455] RSP: 002b:00007ffd650ddb60 EFLAGS: 00000246 ORIG_RAX: 0000000000000000
[  129.549456] RAX: ffffffffffffffda RBX: 0000000000000014 RCX: 00007f55b7732384
[  129.549456] RDX: 0000000000000001 RSI: 00007ffd650ddbc4 RDI: 0000000000000014
[  129.549457] RBP: 00007ffd650ddbc4 R08: 0000000000000000 R09: 000055f90446b010
[  129.549457] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000001
[  129.549457] R13: 000055f9045527e0 R14: 000055f9045527e0 R15: 000055f90452ee30
[  129.549458] Code: 44 8b 45 10 44 89 e1 48 c7 c2 48 fa 98 c0 48 c7 c7 b3 a2 99 c0 44 89 55 d4 50 e8 da f9 9d ff 41 83 7d 18 01 44 8b 55 d4 58 74 02 <0f> 0b 48 8d 65 d8 44 89 d0 5b 41 5c 41 5d 41 5e 41 5f 5d c3 41 
[  129.549474] ---[ end trace 509f42b7871dd8da ]---
[  129.549478] BUG: unable to handle kernel NULL pointer dereference at 0000000000000008
[  129.549505] IP: dal_ddc_close+0xd/0x30 [amdgpu]
[  129.549506] PGD 0 P4D 0 
[  129.549508] Oops: 0000 [#1] SMP NOPTI
[  129.549509] Modules linked in: ccm bnep btusb uvcvideo btrtl videobuf2_vmalloc btbcm videobuf2_memops btintel videobuf2_v4l2 videobuf2_core bluetooth videodev media ecdh_generic snd_hda_codec_realtek snd_hda_codec_generic snd_hda_codec_hdmi joydev hid_multitouch arc4 dell_rbtn nls_iso8859_1 intel_rapl x86_pkg_temp_thermal intel_powerclamp iwlmvm(OE) coretemp dell_laptop mac80211(OE) kvm_intel snd_hda_intel kvm dell_smm_hwmon snd_hda_codec irqbypass snd_hda_core crct10dif_pclmul snd_hwdep crc32_pclmul ghash_clmulni_intel pcbc snd_pcm snd_seq_midi snd_seq_midi_event iwlwifi(OE) snd_rawmidi aesni_intel aes_x86_64 crypto_simd glue_helper dell_w [... cut ...]
[  129.549677] R10: 0000000000000000 R11: 0000000000000246 R12: 0000000000000001
[  129.549678] R13: 000055f9045527e0 R14: 000055f9045527e0 R15: 000055f90452ee30
[  129.549679] Code: 45 f1 00 e8 06 f5 ff ff 48 8b 55 f8 65 48 33 14 25 28 00 00 00 75 02 c9 c3 e8 10 ee 7e ec 0f 1f 44 00 00 55 48 89 e5 53 48 89 fb <48> 8b 7f 08 e8 da f5 ff ff 48 8b 3b e8 d2 f5 ff ff 5b 5d c3 66 
[  129.549719] RIP: dal_ddc_close+0xd/0x30 [amdgpu] RSP: ffffb7e0870b7c60
[  129.549720] CR2: 0000000000000008
[  129.549721] ---[ end trace 509f42b7871dd8db ]---
```

closed 2019-03-29 by superm1

superm1, 2019-08-13:
Just to close the loop on the upstreaming of the bug report for this it was reported here:
https://bugs.freedesktop.org/show_bug.cgi?id=110604

superm1, 2019-08-13:
And it has been reported in Ubuntu here too: https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1826691

commit 57816a79 (2019-03-29): synapticsmst: blacklist plugin when using amdgpu (Fixes: #1121)

An undiagnosed kernel issue is leading to a kernel crash involving
DP aux reads with amdgpu involved.

It's been reproduced in kernel 4.15 with a DKMS version of amdgpu as
well as kernel 4.19 with amdgpu within the kernel.

```
[16862.519947] [drm:generic_reg_wait [amdgpu]] *ERROR* REG_WAIT timeout 10us * 160 tries - submit_channel_request line:246
[16862.519983] WARNING: CPU: 5 PID: 3015 at drivers/gpu/drm/amd/amdgpu/../display/dc/dc_helper.c:254 generic_reg_wait+0xe2/0x160 [amdgpu]
[16862.519984] Modules linked in: rfcomm snd_h [... cut]