fwupd/fwupd #1123: com.logitech.Unifying.RQR12.metainfo.xml could not be loaded

bassch, 2019-03-29:
Something went wrong when Ubuntu offered to update the firmware on the Logitech Unifying Receiver today. The Ubuntu Software Center error message was not very verbose, so I tried the update from the command line with the following result:
```shell
$ fwupdmgr -v update                                                              
Downloading RQR12.08_B0030 for Unifying Receiver...
(fwupdmgr:3411): Fu-DEBUG: 11:49:09.324: creating path /home/******/.cache/fwupdmgr
(fwupdmgr:3411): Fu-DEBUG: 11:49:09.325: skpping download as file already exists
Updating RQR12.08_B0030 on Unifying Receiver...
(fwupdmgr:3411): Fwupd-DEBUG: 11:49:09.325: Emitting ::status-changed() [decompressing]
Decompressing…         [-                                      ]com.logitech.Unifying.RQR12.metainfo.xml could not be loaded: could not parse MetaInfo XML: Error on line 51 char 10: <p> already set 'A few of Logitech's devices:' and tried to replace with '
          used to send select buttons in an unencrypted way, and in an effort
          to protect against this vulnerability, Logitech removed the
          feature.  Although Logitech does not recommend it, these features
          may be re-activated by keeping/downgrading the receiver to an older
          firmware.
        '
```
---
To help us pinpoint your issue, please insert the output of the
following commands when ran on the system with the issue:

```shell
$ fwupdmgr --version
client version:	1.0.9
daemon version:	1.0.9
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34

```

```shell
$ fwupdmgr get-devices
Unifying Receiver
  DeviceId:             cd05276c240d8069f6fac4064192d8552429f99b
  Guid:                 77d843f7-682c-57e8-8e29-584f5b4f52a1
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR12.07_B0029
  VersionBootloader:    BOT01.02_B0015
  Icon:                 preferences-desktop-keyboard
  Created:              2019-03-29

Synaptics VMM3332 inside Dell WD15/TB16 wired Dock
  DeviceId:             d83854e8c81f5d47b0b99c8a3b42ae5127c75eb8
  Guid:                 78c73924-821a-5e6f-af82-86df6f67b182
  Guid:                 02b77ccb-03ab-521f-b98e-ef05395e9fa2
  Guid:                 a6f00ca7-54b7-5d51-9d6e-543d8df85376
  Summary:              Multi-Stream Transport Device
  Plugin:               synapticsmst
  Flags:                updatable|registered
  Vendor:               Synaptics
  Version:              3.12.002
  Icon:                 computer
  Created:              2019-03-29

XPS 13 9370 System Firmware
  DeviceId:             8a21cacfb0a8d2b30c5ee9290eb71db021619f8b
  Guid:                 7ceaf7a8-0611-4480-9e30-64d8de420c7c
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.8.1
  VersionLowest:        0.1.8.1
  Icon:                 computer
[... 347 lines cut ...]
├── vmlinuz-4.15.0-45-generic
└── vmlinuz-4.15.0-46-generic

13 directories, 298 files
```

Please answer the following questions:

- Operating system and version:
```shell
$ cat /etc/lsb-release 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=18.04
DISTRIB_CODENAME=bionic
DISTRIB_DESCRIPTION="Ubuntu 18.04.2 LTS"

```
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):  
`apt-get`
- Have you tried rebooting?  
yes
- Are you using an NVMe disk?  
no
- Is secure boot enabled (only for the UEFI plugin)?  
I think so

hughsie, 2019-03-29:
This is a bug in the XML set in the update description. I've pulled the update from testing and will notify Logitech now, thanks for reporting this bug! I'll update it when I get more info.

xnlcasad, 2019-04-01:
@hughsie Do we have the same problem in the following xmls ?
- MPK03.02_B0009
- RQK62.01_B0014
- MPK01.03_B0024
- MPK04.01_B0010

hughsie, 2019-04-03:
> Do we have the same problem in the following xmls

They look fine to me.

hughsie, 2019-04-11:
This is fixed now.

closed 2019-04-11 by hughsie