fwupd/fwupd #1124: Unable to see daemon debugging output from a snap autoinstaller

superm1, 2019-03-29:
Currently launching a snap autoinstaller based archive the `--verbose` flag only gets carried over to the client since the daemon starts from dbus activation.

I was thinking about this, and had an idea that may _generally_ help debugging.

What if when the client connects to the daemon offer an option that will adjust daemon verbosity at that time?  So if the client is launched with `--verbose` the daemon switches to `--verbose` at that time too?

The other option is to modify snap auto installer to also modify the systemd unit to add `--verbose` when the auto installer is launched with verbose.

hughsie, 2019-03-29:
Hmm, lots to think about there. I assume the daemon would just start dumping stuff to the journal rather than proxy it back to the client. The tricky bit is working out how to "unset" the --verbose flag when the daemon has quit.

superm1, 2019-03-29:
Yes dumping to the journal was my expectation.  But when someone has a problem, you tell them to do something along these lines to get you the debugging:

# systemctl stop fwupd
# fwupdmgr action --verbose
# journalctl -u fwupd.service

> The tricky bit is working out how to "unset" the --verbose flag when the daemon has quit.

You mean when the _client_ has quit right?

hughsie, 2019-03-29:
I'm a bit hesitant allowing users to magically cause a daemon running as root to dump loads of stuff in the journal -- it seems like it's a privileged operation to see the inner workings of the daemon.

superm1, 2019-03-29:
That's true.  What about if this behavior only happened when the client was launched with `sudo` via a PK check?  Similar to how we only show serial number when the client was run with permissions.

hughsie, 2019-03-29:
> when the client was launched with sudo

I think that's a really good idea.

hughsie, 2019-04-30:
Good to close now?

commit db780927 (2019-04-08): trivial: standalone-installer: Have verbose toggle daemon too (Fixes: #1124)

Use the fwupd 1.2.7 feature for daemon verbose logging for one
command.
commit 16bb9561 (2019-04-09): trivial: standalone-installer: Have verbose toggle daemon too (Fixes: #1124)

Use the fwupd 1.2.7 feature for daemon verbose logging for one
command.
commit 6627f659 (2019-04-10): trivial: standalone-installer: Have verbose toggle daemon too (Fixes: #1124)

Use the fwupd 1.2.7 feature for daemon verbose logging for one
command.
closed 2019-08-07 by superm1