fwupd/fwupd #1125: activation doesn't run at shutdown for snap

superm1, 2019-03-29:
I've gotten a report from a storage vendor that the snap is not running activation properly.

I started a PR to sort it out and found 4 problems:

* The path to fwupdtool is wrong in the shutdown script for snap
* The path to pending.db is wrong for snap (and also very difficult to predict)
* the shutdown script is not executable due to a mistake in the install hook
* The snap is unmounted when the shutdown script runs (presumably by `umount.target`?)

The first 3 are easy enough to fix with sed/awk, but the 4th I don't see line of sight to a fix.  
Which has me wondering again if systemd-shutdown is actually the right time to be calling this.  The same thing could potentially happen if you partitions segmented such that `/var` or `/usr` are not accessible when systemd unmounted them.

Should this maybe instead be a really late systemd target that is run `Before=shutdown.target`?

superm1, 2019-03-29:
One more thing to mention; at least in my testing with calling standby immediate followed by activation hasn't made anything blow up for ATA while applications are active.  So introducing in the shutdown process as a unit "might" be OK instead of after disk is read-only.

hughsie, 2019-03-29:
> hasn't made anything blow up for ATA

Did the Dell storage team raise an eyebrow? My gut feeling is we should use the shutdown script we have now, but I'd be perfectly okay with a snap-only service that does something else.

superm1, 2019-03-29:
> Did the Dell storage team raise an eyebrow? 
At least not a visible one, but I'll keep discussing.

> snap-only service that does something else.
Alright I'll march towards that for now.

commit 524221b9 (2019-04-02): snap: Add a custom systemd unit to call activation on shutdown (Fixes: #1125)

The systemd shutdown script gets called after /snap/fwupd/* gets
unmounted meaning it can't be used to do the activation.

Explicitly check that the symlink for /snap/fwupd/current is mounted
when calling the script.
commit a29ca197 (2019-04-02): snap: Add a custom systemd unit to call activation on shutdown (Fixes: #1125)

The systemd shutdown script gets called after /snap/fwupd/* gets
unmounted meaning it can't be used to do the activation.

Explicitly check that the symlink for /snap/fwupd/current is mounted
when calling the script.
closed 2019-04-02 by superm1