fwupd/fwupd #1130: activation flag removal doesn't save to history database

superm1, 2019-04-02:
I've noticed that even after a succesful reboot + activation that the daemon isn't really clearing the activation flag from the history database.  It seems that the state isn't updating either.

```
05:44:58:0882 FuEngine             installed version 13000000 matching history 0.198.23872
05:44:58:0882 FuHistory            modifying device Disk [5dbeb140337f610d54913c0f43d15fdaa0eafb51], only version new 13000001
```

But then running get-history the activation flag is still there.
```
Disk
  DeviceId:             5dbeb140337f610d54913c0f43d15fdaa0eafb51
  Guid:                 cdfcae10-d678-5e2b-9742-d35891619fd2
  Plugin:               ata
  Flags:                internal|updatable|require-ac|needs-reboot|no-auto-instance-ids|needs-activation
  Version:              13000000
  Created:              2019-03-31
  UpdateState:          needs-reboot
  
  [Release]
  Version:              0.198.23873
  Checksum:             SHA1(39eadff89783818b44b09f1deff5e1034f53590d)
  Flags:                none
  CpuArchitecture:      x86_64
  RuntimeVersion(org.freedesktop.fwupd): 1.2.7
  DistroId:             ubuntu
  CompileVersion(org.freedesktop.fwupd): 1.2.7
  CompileVersion(com.redhat.fwupdate): 12
  CompileVersion(org.freedesktop.gusb): 0.2.11
  RuntimeVersion(org.freedesktop.appstream-glib): 0.7.14
  KernelVersion:        4.15.0-29-generic
  DistroVersion:        18.04
  CompileVersion(com.redhat.efivar): 34
  RuntimeVersion(com.dell.libsmbios): 2.4
  BootTime:             1553898087
  RuntimeVersion(com.redhat.fwupdate): 12

Disk
  DeviceId:             5dbeb140337f610d54913c0f43d15fdaa0eafb51
  Guid:                 cdfcae10-d678-5e2b-9742-d35891619fd2
  Plugin:               ata
  Flags:                internal|updatable|require-ac|needs-reboot|no-auto-instance-ids|needs-activation
  Version:              13000001
  Created:              2019-03-31
  UpdateState:          needs-reboot
  
  [Release]
  Version:              0.198.23872
  Checksum:             SHA1(96a508d36892d45bd702a20161ceb82b3c6f13f2)
  Flags:                none
  CpuArchitecture:      x86_64
  RuntimeVersion(org.freedesktop.fwupd): 1.2.7
  DistroId:             ubuntu
  CompileVersion(org.freedesktop.fwupd): 1.2.7
  CompileVersion(com.redhat.fwupdate): 12
  CompileVersion(org.freedesktop.gusb): 0.2.11
  RuntimeVersion(org.freedesktop.appstream-glib): 0.7.14
  KernelVersion:        4.15.0-29-generic
  DistroVersion:        18.04
  CompileVersion(com.redhat.efivar): 34
  RuntimeVersion(com.dell.libsmbios): 2.4
  BootTime:             1554062270
  RuntimeVersion(com.redhat.fwupdate): 12
```

hughsie, 2019-04-02:
What version _to_ and _from_ are you using? If you're just reinstalling the same version the daemon can't know if the flash was a success or not.

superm1, 2019-04-02:
Going from 13000001 to 13000000 and 13000000 back to 13000001

superm1, 2019-04-02:
I think I see what's going on checking `pending.db`.

`version_new` is marked as `0.198.23873` but the new FW version was `13000001`.  So it's the conversion messing things up in the history.

commit c98bf8b8 (2019-04-03): fu-history: When matching `version_new` use the release not the device (Fixes: #1130)
commit acc51d11 (2019-04-03): fu-history: When matching `version_new` use the release not the device (Fixes: #1130)
PR #1132 (2019-04-09): fu-engine: Clear activation flag on old firmware not new (Fixes: #1130)
For me this seems to fix #1130 and match the right device.

Repro steps:
1. Clear pending.db
2. Start with ATA firmware X.
3. Flash ATA firmware Y.
4. Reboot machine (activate runs on shutdown)
5. Verify running on ATA firmware Y.
6. Verify that `fwupdmgr activate` return `No firmware to activate`.

Type of pull request:
- [ ] New plugin (Please include [new plugin checklist](https://github.com/hughsie/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
closed 2019-04-09 by hughsie