fwupd/fwupd #1139: Can't update logitech unifying receiver: cannot handle firmware requirement 'not-child'

kacpi2442, 2019-04-07:
```shell
$ sudo fwupdmgr get-updates
cannot handle firmware requirement 'not-child'
```

```shell
$ fwupdmgr --version
client version: 1.2.2
compile-time dependency versions
        gusb:   0.3.0
        efivar: 35
daemon version: 1.2.6
```

```shell
$ fwupdmgr get-devices
Unifying Receiver
  DeviceId:             41b3929e3920d086c5ecae8db4b55d57cf6a2f2a
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.05_B0029
  VersionBootloader:    BOT03.01_B0008
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-04-07
```

Please answer the following questions:

- Operating system and version:
Manjaro linux
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
I have installed it using pacman

I simply cannot install latest RQR24.06_B0030 firmware

agustinventura, 2019-04-07:
Hi, I'm trying to fix a bricked Logitech MK520 mouse and keyboard combo. Followed instructions here: https://github.com/hughsie/fwupd/issues/655 but I'm getting the same 'not-child' error here.

- fwupdmgr version:
`
sudo fwupdmgr --version
client version:	1.2.7
checkout info:	1.2.6-27-g02ac92c
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.7
`
- fwupdmger get-devices:
`
fwupdmgr get-devices 
Unifying Receiver
  DeviceId:             54c916aa848583256a9b6959446d0890818cbe18
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 a160f16a-613b-5627-a224-ace42013f9a6
  Guid:                 9d131a0c-a606-580f-8eda-80587250b8d6
  Guid:                 941f4133-39e5-56a0-9df6-902547436c5c
  Guid:                 afa5518d-7c55-51cb-8e1c-bf3b11b67dfa
  Guid:                 cee9c5bc-aa12-51eb-a220-d0ef588d7cb2
  Guid:                 571229d5-caed-5cbc-91c1-5c216601a194
  Summary:              A miniaturised USB wireless receiver (bootloader)
  Plugin:               unifying
  Flags:                updatable|supported|registered|is-bootloader
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR00.00_B0000
  VersionBootloader:    BOT01.02_B0014
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-04-07
`

- Operating system and version:
  Ubuntu 18.04
- Installation via snap, tried with classic and edge channel (currently on edge)

kacpi2442, 2019-04-07:
I have actually managed to install RQR24.06_B0030 by myself.
Here is what i did:
```shell
$ git clone https://github.com/Logitech/fw_updates.git
$ cd fw_updates/RQR24/RQR24.06/lvfs
$ nano com.logitech.Unifying.RQR24.metainfo.xml
```
Comment out lines 62-67:
```
<!--    <firmware compare="regex" version="RQK62.00_*">not-child</firmware>
    <firmware compare="regex" version="RQK63.0[01]_*">not-child</firmware>
    <firmware compare="regex" version="MPK01.0[0-2]_*">not-child</firmware>
    <firmware compare="regex" version="MPK03.0[0-1]_*">not-child</firmware>
    <firmware compare="regex" version="MPK04.00_*">not-child</firmware> -->
```
```shell
$ make
$ sudo fwupdmgr install Logitech-Unifying-RQR24.06_B0030.cab
```
It works flawlessly.

kacpi2442, 2019-04-07:
@agustinventura 
Try this, maybe it will work for you too:
```shell
$ git clone https://github.com/Logitech/fw_updates.git
$ cd fw_updates/RQR12/RQR12.08/lvfs
$ nano com.logitech.Unifying.RQR12.metainfo.xml
```
Comment out lines 62-67:
```
<!--    <firmware compare="regex" version="RQK62.00_*">not-child</firmware>
    <firmware compare="regex" version="RQK63.0[01]_*">not-child</firmware>
    <firmware compare="regex" version="MPK01.0[0-2]_*">not-child</firmware>
    <firmware compare="regex" version="MPK03.0[0-1]_*">not-child</firmware>
    <firmware compare="regex" version="MPK04.00_*">not-child</firmware> -->
```
```shell
$ make
$ sudo fwupdmgr install Logitech-Unifying-RQR12.08_B0030.cab
```
(I have only changed `RQR24.06` to `RQR12.08`)

LinuxOnTheDesktop, 2019-04-07:
I have the same problem, on Linux Mint 19.1 x64 Cinnamon. Details as follows.

```
$  fwupdmgr --version
client version:	1.1.1
daemon version:	1.0.9
checkout info:	1.1.0-19-g7769304c
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	efivar:	34
$  fwupdmgr refresh
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading…           [***************************************]
Fetching signature https://cdn.fwupd.org/downloads/firmware.xml.gz.asc

$  fwupdmgr get-updates
cannot handle firmware requirement not-child
No upgrades for UEFI Device Firmware, current is 184.60.3561: 184.55.3510=older, 184.60.3561=same
No upgrades for 20KHCTO1WW System Firmware, current is 0.1.37: 0.1.27=older, 0.1.28=older, 0.1.30=older, 0.1.31=older, 0.1.34=older, 0.1.36=older, 0.1.37=same
 nicholas  ~  $  sudo fwupdmgr get-updates
[sudo] password for [user]: 
cannot handle firmware requirement not-child
No upgrades for UEFI Device Firmware, current is 184.60.3561: 184.55.3510=older, 184.60.3561=same
No upgrades for 20KHCTO1WW System Firmware, current is 0.1.37: 0.1.27=older, 0.1.28=older, 0.1.30=older, 0.1.31=older, 0.1.34=older, 0.1.36=older, 0.1.37=same
```

agustinventura, 2019-04-07:
@kacpi2442 it works!!! You made my day, thanks!!!

LinuxOnTheDesktop, 2019-04-07:
It worked for me too, which is good and I am grateful.

Less good is the continuing error message from `fwupdmgr`.

romanstingler, 2019-04-08:
with this method I bricked my receiver 

```
~/fw_updates/RQR24/RQR24.07/lvfs                                                     [8:53:12] 
> $ sudo fwupdmgr install Logitech-Unifying-RQR24.07_B0030.cab                                              [±master ●●]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Installing on Unifying Receiver…
Idle…                    [***************************************]
Writing…                 [************************************** ]failed to write ram bufer @0x00: invalid command response of 01, expected e0
                                                                                                                         
blackout@romanworkstation ~/fw_updates/RQR24/RQR24.07/lvfs                                                     [8:53:29] 
> $ sudo fwupdmgr get-devices                                                                               [±master ●●]
Unifying Receiver
  DeviceId:             970e85c8400ed95ec5c02b86e52a0ca4ce6fc051
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Guid:                 de81631f-2117-51f8-bcb1-4f874a293b29 <- USB\VID_046D&PID_AAAC&REV_0301
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1 <- USB\VID_046D&PID_AAAC
  Guid:                 941f4133-39e5-56a0-9df6-902547436c5c <- USB\VID_046D
  Guid:                 afa5518d-7c55-51cb-8e1c-bf3b11b67dfa <- USB\CLASS_03&SUBCLASS_00&PROT_00
  Guid:                 cee9c5bc-aa12-51eb-a220-d0ef588d7cb2 <- USB\CLASS_03&SUBCLASS_00
  Guid:                 571229d5-caed-5cbc-91c1-5c216601a194 <- USB\CLASS_03
  Guid:                 21e75d9a-5ce6-5da2-b7ab-910c7f3f6836
  Summary:              A miniaturised USB wireless receiver (bootloader)
  Plugin:               unifying
  Flags:                updatable|supported|registered|is-bootloader
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.00_B0000
  VersionBootloader:    BOT03.01_B0008
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-04-08

```

before flashing I had RQR24.03_xxxxx
is there anything I can do??

hughsie, 2019-04-08:
I'm working on this now.

closed 2019-04-09 by superm1

taupehat, 2019-04-10:
Confirmed both problem and fix in Fedora 28.

LinuxOnTheDesktop, 2019-04-25:
Any chance of pushing the fix to my machine? I have Mint 19.1.

hughsie, 2019-04-25:
You need to file a bug with Mint and ask them to update fwupd. There's nothing I can do.

superm1, 2019-04-25:
If mint doesn't want to do it, you can possibly use the snap instead: https://github.com/hughsie/fwupd/wiki/fwupd-snap

LinuxOnTheDesktop, 2019-04-27:
I installed the snap, and now things seem rather messed up:

```
$  fwupdmgr get-topology
○
├─ X1 Carbon Thunderbolt Controller          eded407096031a81370bc25ae01650697a7c72b6
├─ 20KHCTO1WW System Firmware                1c53551e7da69d896138fac1ae131c83ad46d923
├─ UEFI Device Firmware                      29749712e93b58730ecd1b8537a5503ff5b7fb27
├─ UEFI Device Firmware                      9698faabddf0d7b18925cfbbda95f8b0d0dacc53
├─ LENSE20512GMSP34MEAT2TA                   04e17fcf7d3de91da49a163ffe4907855c3648be
└─ Unifying Receiver                         fb1a5fd6e7d8cbf5d5a2ba7df68aee106ce41027
 ~  $  fwupdmgr update
ignoring 20KHCTO1WW System Firmware [1c53551e7da69d896138fac1ae131c83ad46d923] as not updatable
ignoring UEFI Device Firmware [29749712e93b58730ecd1b8537a5503ff5b7fb27] as not updatable
No upgrades for Unifying Receiver, current is RQR24.06_B0030: RQR24.06_B0030=same, RQR24.05_B0029=older
 ~  $  fwupdmgr get-updates
ignoring 20KHCTO1WW System Firmware [1c53551e7da69d896138fac1ae131c83ad46d923] as not updatable
ignoring UEFI Device Firmware [29749712e93b58730ecd1b8537a5503ff5b7fb27] as not updatable
No upgrades for Unifying Receiver, current is RQR24.06_B0030: RQR24.06_B0030=same, RQR24.05_B0029=older
```
I did not get these 'ignoring' messages before.

superm1, 2019-04-27:
Can you please share fwupdmgr --version with the snap?

You installed from standard channel or edge channel?

Also please share get-devices output.

[EXTERNAL EMAIL]

LinuxOnTheDesktop, 2019-04-28:
Thanks.

I used this command to instatt the snap version of fwup: `snap install fwupd --classic`. As to version:

```
$  fwupdmgr --version
client version:	1.1.1
daemon version:	1.2.8
checkout info:	1.1.0-19-g7769304c
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	efivar:	34

```

superm1, 2019-04-28:
You're not running the fwupdmgr from the snap.
You must still have one in /use/bin or /usr/local/bin.

You can try fwupd.fwupdmgr or remove your other fwupd installation.

LinuxOnTheDesktop, 2019-04-28:
Thanks. After removing remnants of my old installation, now I see this:

```
 $  fwupdmgr refresh
2019/04/28 18:25:32.455915 cmd_run.go:876: WARNING: cannot create user data directory: failed to verify SELinux context of /home/<username>/snap: exec: "matchpathcon": executable file not found in $PATH
Fetching metadata https://cdn.fwupd.org/downloads/firmware.xml.gz
Downloading?             [***************************************] Less than one minute remaining?
```

superm1, 2019-04-29:
You can try to install `selinux-utils` to see if that helps.

LinuxOnTheDesktop, 2019-04-29:
Yes, I did try that. No dice. I could not get the flatpak version of fwupd to work either. At this point I give up. Mint is due an update in June and perhaps the Mint team will update fwupd before that anyway. Either way, I think I will just wait.

Bass850, 2019-12-19:
> I have actually managed to install RQR24.06_B0030 by myself.
> Here is what i did:
> 
> ```shell
> $ git clone https://github.com/Logitech/fw_updates.git
> $ cd fw_updates/RQR24/RQR24.06/lvfs
> $ nano com.logitech.Unifying.RQR24.metainfo.xml
> ```
> 
> Comment out lines 62-67:
> 
> ```
> 
> ```
> 
> ```shell
> $ make
> $ sudo fwupdmgr install Logitech-Unifying-RQR24.06_B0030.cab
> ```
> 
> It works flawlessly.

This solution worked perfectly for  me on 19.04 ater trying everything else i could find. Thanks