fwupd/fwupd #1141: Failed to alter database: duplicate column name: checksum_device

ncraike, 2019-04-08:
I'm seeing an error from `fwupd` on a new install of Fedora 29.

I first see a few errors from the Fedora software center: when I try to check for updates, I get an error notification in the GUI reading "Error calling StartServiceByName for org.freedesktop.fwupd".

If I check `journalctl -u fwupd`, I see these lines:
```
Apr 08 09:48:51 localhost.localdomain systemd[1]: Starting Firmware update daemon...
Apr 08 09:48:51 localhost.localdomain fwupd[14615]: Failed to load engine: Failed to alter database: duplicate column name: checksum_device
Apr 08 09:48:51 localhost.localdomain systemd[1]: fwupd.service: Main process exited, code=exited, status=1/FAILURE
Apr 08 09:48:51 localhost.localdomain systemd[1]: fwupd.service: Failed with result 'exit-code'.
Apr 08 09:48:51 localhost.localdomain systemd[1]: Failed to start Firmware update daemon.
```

They repeat if I try `sudo systemctl restart fwupd`.

----

This is a very fresh Fedora 29 install as of Friday 5th April 2019 (Australian time). All I've done is installed the OS, then opened the Software GUI and restarted to install updates when prompted a couple times.

However, the computer itself has had some previous installs of Windows 10 and other Linux distros (eg Manjaro), which means its partitioning table is a little weird (see `efibootmgr` output below).

----

Debugging commands:

The `fwupdmgr --version` command actually fails to complete. It pauses for a while before outputting the "Failed to connect to daemon" message.
```shell
$ fwupdmgr --version
client version:	1.2.6
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Timeout was reached
```

Also:

```shell
$ dnf info fwupd
Last metadata expiration check: 0:04:18 ago on Mon 08 Apr 2019 10:22:57 AEST.
Installed Packages
Name         : fwupd
Version      : 1.2.6
Release      : 1.fc29
Architecture : x86_64
Size         : 10 M
Source       : fwupd-1.2.6-1.fc29.src.rpm
Repository   : @System
From repo    : updates
Summary      : Firmware update daemon
URL          : https://github.com/hughsie/fwupd
License      : LGPLv2+
Description  : fwupd is a daemon to allow session software to update device firmware.

```

```shell
$ efibootmgr -v
BootCurrent: 0003
Timeout: 0 seconds
BootOrder: 0003,0001,0000
Boot0000  Windows Boot Manager	HD(1,GPT,f6162b6a-7f3a-4938-96d9-a19ea5c61bb9,0x800,0xfa000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)WINDOWS.........x...B.C.D.O.B.J.E.C.T.=.{.9.d.e.a.8.6.2.c.-.5.c.d.d.-.4.e.7.0.-.a.c.c.1.-.f.3.2.b.3.4.4.d.4.7.9.5.}....................
Boot0001* Manjaro	HD(1,GPT,f6162b6a-7f3a-4938-96d9-a19ea5c61bb9,0x800,0xfa000)/File(\EFI\Manjaro\grubx64.efi)
Boot0003* Fedora	HD(1,GPT,f6162b6a-7f3a-4938-96d9-a19ea5c61bb9,0x800,0xfa000)/File(\EFI\fedora\shimx64.efi)
```

As far as I can tell, I'm not using the UEFI plugin. The `efivar` command is not installed.

----

> - Operating system and version:

Fedora 29

```
$ uname -a
Linux localhost.localdomain 5.0.5-200.fc29.x86_64 #1 SMP Wed Mar 27 20:58:04 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```

> - How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):

Installed by the Fedora 29 installer, as far as I can tell.

> - Have you tried rebooting?

Yes.

> - Are you using an NVMe disk?

I don't know.

> - Is secure boot enabled (only for the UEFI plugin)?

I don't know. I don't think I'm using the UEFI plugin.

hughsie, 2019-04-08:
> and other Linux distros

Did you reuse /var from a previous install? Either way you can just do `sudo rm /var/lib/fwupd/pending.db` and then restart fwupd. Newer fwupd daemons detect this kind of problem and do just that themselves: https://github.com/hughsie/fwupd/commit/dbeba37def91c0bfb55507522449ab6f70a001e5

closed 2019-04-08 by hughsie

ncraike, 2019-04-09:
> Did you reuse /var from a previous install?
Nope, I made a small empty partition at the end of the disk, put a tar.gz of `/home/` from the previous install there, and repartitioned the previous install for Fedora's use.

> Either way you can just do `sudo rm /var/lib/fwupd/pending.db` and then restart fwupd. 
I'll give this a go, thanks.

call-25153, 2019-04-11:
In case this helps others, I wanted to add that I had the exact same problem.  Clean install of Fedora 29 to SATA SSD (new disk), used auto-partitioning and enabled encryption.  I'm not using secure boot or UEFI. Only non-standard item was installing current nVidia drivers from RPMfusion repo.  Error occurred after first round of system updates as managed by Fedora.  Specifically, problem started after upgrade from (Fedora) version fwupd-1.1.2-1.fc29.x86_64 to fwupd-1.2.6-1.fc29.x86_64.

Deleting file `/var/lib/fwupd/pending.db` fixed the issue.

Thanks to @ncraike for reporting the issue, as finding this bug report let me fix my problem immediately.  Also thanks to @hughsie for the development efforts and bug fix instructions.