fwupd/fwupd #1142: Thunderbolt update completed even if system is in battery mode and has less than 10% battery

RJ-LeJaLTD, 2019-04-08:
Hello, I tried to update my Thunderbolt port firmware using `fwupdmgr --allow-older --allow-reinstall install <cab file>`, the firmware update was successful however the battery level of system was less than 10% and running in battery mode or AC mode. I would like to ask on how can we prohibit firmware update when system is in AC mode. 

```shell
client version:	1.0.9
daemon version:	1.0.9
compile-time dependency versions
	appstream-glib:	0.7.7
	gusb:	0.2.11
	fwupdate:	10
	efivar:	34
```

```shell
$ fwupdmgr get-devices
ThinkPad X280 Thunderbolt Controller
  DeviceId:             a7c3aa8da0219d7d072404ea9cd3dd13f2818c74
  Guid:                 4808eca4-fd4a-50e6-9e8d-bfd813f063da
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              16.00
  Icon:                 computer
  Created:              2019-04-08

UEFI Device Firmware
  DeviceId:             c0fa355f884de9e5d3912e5d697c80039e44e861
  Guid:                 84bf8333-8724-471e-b60b-5759bdbd9785
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.50.3425
  VersionLowest:        0.0.1
  Icon:                 audio-card
  Created:              2019-04-08

UEFI Device Firmware
  DeviceId:             ca368aebcf7da847029e9f2520ec55fb7a036b31
  Guid:                 3f4a527b-6588-45b8-b2d3-dc61189b63cb
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.4
  VersionLowest:        0.1.4
  Icon:                 audio-card
  Created:              2019-04-08

20KFCTO1WW System Firmware
  DeviceId:             5a566863d357fb728a620cdf235632fb9bc99f5f
  Guid:                 508f7539-1ad6-48b9-8680-38377535009d
  Guid:                 230c8b18-8d9b-53ec-838b-6cfc0383493a
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.14
  VersionLowest:        0.0.1
  Icon:                 computer
  Created:              2019-04-08
```
*:･ﾟ✧*:･ﾟ✧*:･ﾟ✧*:･ﾟ✧*:･ﾟ✧*:･ﾟ✧
- Operating system and version:
   Ubuntu 18.04.2 LTS
- How did you install fwupd (ex: `from source`, `pacman`, `apt-get`, etc):
   apt install fwupd
- Have you tried rebooting?
   No
- Are you using an NVMe disk?
   Yes
- Is secure boot enabled (only for the UEFI plugin)?

hughsie, 2019-04-08:
We already added this in 1.2.4 -- if you update using the snap version we will refuse to update on <30% battery life.

cheejpn, 2019-04-08:
Hello Hughsie, I'm working with RJ in Lenovo ThinkPad team. Thanks for your comment.... Usually we only support FW update when AC adapter connected even battery capacity is enough. Is it policy of FWUPD in linux to support battery mode in case 30% capacity?

hughsie, 2019-04-08:
> Is it policy of FWUPD in linux to support battery mode in case 30% capacity

Yes, one failure mode is the user starting the update and then removing the AC adapter; 30% should always give us enough power to complete the update.

RJ-LeJaLTD, 2019-04-09:
hello @hughsie , I tried to update again the Thunderbolt firmware while the battery life is lower than 30% in fedora29 using the following command. 

`fwupdmgr --allow-reinstall --allow-older install <cab file>`

with this version of fwupd 

`
fwupdmgr --version
client version:	1.2.6
compile-time dependency versions
	gusb:	0.3.0
	efivar:	37
daemon version:	1.2.6
`
the firmware update was completed without error or warning. 
May we request support on how can we prohibit firmware update?

superm1, 2019-04-09:
It looks to me that the Thunderbolt plugin doesn't actually set a policy around AC.  I don't see anything in `FWUPD_DEVICE_FLAG_REQUIRE_AC`

commit cc3d5605 (2019-04-09): thunderbolt: Set require-ac for Thunderbolt devices (Fixes: #1142)
closed 2019-04-09 by hughsie